SBFspot/SBFspot #550: Build fails with 3.9.5

roelofjansen, 2022-06-08:
My docker build fails as from v3.9.5

```
 > [linux/amd64 builder 6/6] RUN cd SBFspotUploadDaemon && make sqlite && make install_sqlite && cd ..:
#30 4.191  extern int quiet;
#30 4.191             ^~~~~
#30 4.192 In file included from ../SBFspotUploadCommon/../SBFspot/SBFspot.h:40:0,
#30 4.192                  from ../SBFspotUploadCommon/PVOutput.h:38,
#30 4.192                  from ../SBFspotUploadCommon/PVOutput_x.cpp:36:
#30 4.192 ../SBFspotUploadCommon/../SBFspot/EventData.h:42:13: note: previous declaration as 'bool quiet'
#30 4.192  extern bool quiet;
#30 4.192              ^~~~~
#30 4.937 makefile:106: recipe for target 'sqlite/obj/../SBFspotUploadCommon/PVOutput_x.o' failed
#30 4.937 make: *** [sqlite/obj/../SBFspotUploadCommon/PVOutput_x.o] Error 1
```

roelofjansen, 2022-06-08:
Please see my github project; https://github.com/solarexplorer/sbfspot-sqlite

The Dockerfile builds the sbfspot-sqlite image. You can check it out locally and build it using;

`docker buildx build -t solarexplorer/sbfspot-sqlite --load .
`

sbf-, 2022-06-09:
Sources in release 3.9.5 were updated yesterday, so probably you have to redownload them.
Your issue was fixed [here](https://github.com/SBFspot/SBFspot/commit/53cc5393d633e1040ea2c421545340232bf41f0a)