SBFspot/SBFspot #552: Unable to upgrade to 3.9.5 - 'GLIBCXX_3.4.26' not found

camhef, 2022-07-07:
I have just tried to update to the latest release (3.9.5) and that has failed as shown below.  Here's the job output:
```
curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash

SBFspot Configuration Tool V1.2.1
Running on Raspberry Pi 4 Model B Rev 1.4
Checking whiptail... Installed.
architecture=armhf
Raspbian GNU/Linux 10 (buster)
username=pi
userhome=/home/pi
Minimum SBFspot version supported by this tool is 3.5.0
Request to install version 3.9.5
Searching for existing configuration(s)...
Reading settings from /usr/local/bin/sbfspot.3/SBFspot.cfg
/var/tmp/sbfspot.3/cfg.tmp: line 12: fg: no job control
Reading settings from /usr/local/bin/sbfspot.3/SBFspotUpload.cfg
Saving defaults to /var/tmp/sbfspot.3/sbfspot.defaults... Done
Performing some checks before installation... Done
Downloading https://github.com/sbfspot/sbfspot/releases/download/V3.9.5/sbfspot-sqlite-arm-linux-buster.tar.gz
Copying files...
SBFspot.db exists
Checking database...
Schema Version=1
Saving existing SBFspot.cfg... Done
Configuring SBFspot... Done
Running SBFspot...
/usr/local/bin/sbfspot.3/SBFspot: /lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /usr/local/bin/sbfspot.3/SBFspot)

Installation aborted
```
I did manage to upgrade from the previously installed version (3.5.0!) to 3.9.4 sucessfully using command:
`curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash -s 3.9.5`

Does anyone know what I may have done wrong, or what I can do get release 3.9.5 installed.

Thanks.

sbf-, 2022-07-08:
You will need to build SBFspot manually following the wiki
https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite

deagel, 2022-07-08:
Hi,
i have this issue too with mariadb:

SBFspot Configuration Tool V1.2.1
Running on Raspberry Pi 4 Model B Rev 1.1
Checking whiptail... Installed.
architecture=armhf
Raspbian GNU/Linux 10 (buster)
username=sbfspot
userhome=/home/sbfspot
Minimum SBFspot version supported by this tool is 3.5.0
Request to install latest version (3.9.5)
Searching for existing configuration(s)...
Reading settings from /usr/local/bin/sbfspot.3/SBFspot.cfg
/var/tmp/sbfspot.3/cfg.tmp: Zeile 10: fg: Keine Job Steuerung in dieser Shell.
Reading settings from /usr/local/bin/sbfspot.3/SBFspotUpload.cfg
Saving defaults to /var/tmp/sbfspot.3/sbfspot.defaults... Done
Performing some checks before installation... Done
Downloading https://github.com/sbfspot/sbfspot/releases/download/V3.9.5/sbfspot-mariadb-arm-linux-buster.tar.gz
Copying files...
Checking database...
Schema Version=1
Saving existing SBFspot.cfg... Done
Configuring SBFspot... Done
Running SBFspot...
/usr/local/bin/sbfspot.3/SBFspot: /usr/lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /usr/local/bin/sbfspot.3/SBFspot)

Installation aborted

sbf-, 2022-07-08:
I guess your system is not up to date...
Try this:
sudo apt update
sudo apt-get install --only-upgrade libstdc++6

deagel, 2022-07-08:
hi,
no:
libstdc++6 is already the newest version (8.3.0-6+rpi1).

camhef, 2022-07-09:
> You will need to build SBFspot manually following the wiki https://github.com/SBFspot/SBFspot/wiki/Installation-Linux-SQLite

Thanks for the reply. I plan to move my SBFspot instance to a new Raspberry Pi next week. If 3.9.5 installs cleanly then great, otherwise I'll nstall from source.

deagel, 2022-07-11:
> I guess your system is not up to date... Try this: [(source)](https://stackoverflow.com/a/66465559/1509300)
> 
> ```
> sudo apt update
> sudo apt-get install --only-upgrade libstdc++6
> ```

Hi,
it was not an old libstdc++6 it was the old buster.

it look like a prerequisite of 3.9.5 is raspbian 11 (bullseye) but I miss this dependency or did not found it in preparation for this update.  

It is possible to add a small check in the update process?

Thanks
deagel

wokkeltje13, 2022-07-17:
Because I was not able to upgrade my 3.9.4 to 3.9.5 on Buster (RPi Zero W 1.1) I took a brand new SD card and setup a new Bullseye install using 3.9.5. Everything looked OK until I noticed the following in HA. After the yellow line
[image]
When testing using `/usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv` I noticed that SBFspot could not contact my SMA after the default of 10 tests, I ran this multiple times and 1 on maybe 10 it could (so 1 on 100 tests) It could contact my SMA.
When I put back my old SD card with Buster and 3.9.4, I could contact my SMA everytime after about 5 of the default 10 tests.
My question: Could this be related to 3.9.5 or to Bulseye (rPi is the same device on the same location, so no difference in signal strength)

obduser, 2022-08-08:
Same Problem with "PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"

I installed 3.9.4 again. Will there be a solution for "Buster" ?

sbf-, 2022-08-19:
@obduser
@deagel 
I recompiled the V3.9.5 sources for "Buster"
Seems to be OK now

closed 2022-08-19 by sbf-