SBFspot/SBFspot #553: Difference between App-Data and EToday

hevilp, 2022-07-08:
Hi,

Appdata says around 12,7 kwH and the sqlite-data says 20560.

``startTime: 08/07/2022 00:00:00
Reading events: 2022-Jul-01
logoffMultigateDevices()
logoffSMAInverter()
Fri Jul  8 12:13:31 2022: INFO: Done.
dietpi@DietPi:/usr/local/bin/sbfspot.3$ nano daydata 
dietpi@DietPi:/usr/local/bin/sbfspot.3$ ./SBFspot -?
SBFspot V3.9.5
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
SBFspot [-options]
 -scan               Scan for bluetooth enabled SMA inverters.
 -d#                 Set debug level: 0-5 (0=none, default=2)
 -v#                 Set verbose output level: 0-5 (0=none, default=2)
 -ad#                Set #days for archived daydata: 0-300
                     0=disabled, 1=today (default), ...
 -am#                Set #months for archived monthdata: 0-300
                     0=disabled, 1=current month (default), ...
 -ae#                Set #months for archived events: 0-300
                     0=disabled, 1=current month (default), ...
 -cfgX.Y             Set alternative config file to X.Y (multiple inverters)
 -finq               Force Inquiry (Inquire inverter also during the night)
 -q                  Quiet (No output)
 -nocsv              Disables CSV export (Overrules CSV_Export in config)
 -nosql              Disables SQL export
 -sp0                Disables Spot.csv export
 -installer          Login as installer
 -password:xxxx      Installer password
 -loadlive           Use predefined settings for manual upload to pvoutput.org
 -startdate:YYYYMMDD Set start date for historic data retrieval
 -settime            Sync inverter time with host time
 -mqtt               Publish spot data to MQTT broker

Libraries used:
        SQLite V3.34.1
        BOOST V1.74.0
dietpi@DietPi:/usr/local/bin/sbfspot.3$ ./daydata 
dietpi@DietPi:/usr/local/bin/sbfspot.3$ tail -f /var/log/sbfspot.3/PV_20220708.log 
Inverter Sleep Time  : 08/07/2022 12:26:52
MQTT: Publishing (sbfspot_{serial}) "Timestamp": "08/07/2022 12:26:54","SunRise": "08/07/2022 04:55:00","SunSet": "08/07/2022 21:27:00","InvSerial": 3009019519,"InvName": "SN: 3009019519","InvTime": "08/07/2022 12:26:53","InvStatus": "Ok","InvTemperature": 0.000,"InvGridRelay": "Closed","EToday": 20.560,"ETotal": 10039.566,"PACTot": 5320.000,"UDC1": 615.450,"UDC2": 444.580,"IDC1": 5.108,"IDC2": 5.179,"PDC1": 3144.000,"PDC2": 2302.000``

sbf-, 2022-07-08:
Which app?
Like Sunnyexplorer, SBFspot gets its data directly from inverter, so this should be correct.

hevilp, 2022-07-08:
SMA Energymeter - the App from SMA

hevilp, 2022-07-08:
I saw that App is not correct - maybe they correct it and need a few hours - thanks

closed 2022-07-08 by hevilp