SBFspot/SBFspot #554: Fresh install: SBFspotUpload installation failed: Different number of SID/Serial

broke23, 2022-07-10:
Hello again,

sbfspot at my parents' home went bust, corrupted card, so after +2years I had the chance to reinstall to latest version of SBF and raspian...

latest release installed on pi, boots, set up of SBFspot always results in **different number of SID/Serial...**

my parents have 3 inverters, all are seen by bluetooth, and are selected by default:

```
[*] 00:80:25:1F:71:E2  SMA002d 10 MIDDEN SN: 2120093240 SN2120093240
[*] 00:80:25:1D:5B:9C  SMA002d 10 VOOR SN: 2120102129 SN2120102129
[*] 00:80:25:1D:5B:96  SMA002d 12 ACHTER SN: 2120102179 SN2120102179 
```
the ID's at pvoutput: 49273, 49274 and [49275 ](https://pvoutput.org/intraday.jsp?sid=49275)(sums up on 49259 to have global overview instead of per inverter)

what I tried: 49259: not ok
49273 till 49275: not ok

tried the old ways:

SN2120102129:49273,SN2120093240:49274,SN2120102179:49275
2120102129:49273,2120093240:49274,2120102179:49275

when I abort: 

```
Different number of SID/Serial
Found 1 SIDS
Found 0 S/N

```

the api is ok, manually checked after copy paste ... 

What am I missing here you think ?

broke23, 2022-07-10:
https://pvoutput.org/service/r2/getsystem.jsp?key=APIKEY&sid=SYSTEMID

replaced api and system id corresponding ours, 3x success...

broke23, 2022-07-10:
sunny explorer works AFTER having changed default passwords... edited that in config, still no match, can't get past SID/serial part ...

sbf-, 2022-07-10:
It´s a known bug
https://github.com/SBFspot/sbfspot-config/issues/46

broke23, 2022-07-10:
Thanks, pff, you can see I'm not used to github, I was looking through the closed tickets to find anything, didn't look in the obvious category