SBFspot/SBFspot #559: SBFspotUploadDaemon log to stdout?

ghen2, 2022-08-02:
Feature request: could SBFspotUploadDaemon get an option to write its logs to stdout, instead of a file in `LogDir` ? (eg. by "`LogDir=-`")

This would play nice(r) with process managers like Linux systemd or OpenWRT procd, to capture and centralize the logs, rather than having to manage yet another custom logfile location.

sbf-, 2022-08-03:
There is already limited use of `std::clog`
The problem I encounter is the availability of the `systemd` log after some time:
```
systemctl status SBFspotUpload.service
Active: active (running) since Tue 2022-07-19 15:17:16 CEST; 2 weeks 0 days ago

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.
```

I have basic Linux knowledge, so maybe there is a solution?

ghen2, 2022-08-03:
It only prints this on startup:

```
SBFspotUploadDaemon Version 3.0.1
Loglevel=2
LogDir=/tmp/
```
This is what systemd/journald has rotated away since you started the daemon.

The regular "Uploading datapoint..." output is logged to date-stamped files under LogDir, I'd like to (get the option to) capture these on stdout instead as well, writing nothing to the filesystem at all (on an embedded device).

obduser, 2022-08-21:
How can I deactivate the writing of the log file by SBFspotUploadDaemon?

unfortunately this doesn't work:

/usr/local/bin/sbfspot.3/daydata --> change the line log=/var/log/sbfspot.3/PlantName_$(date '+%Y%m%d').log to log=/dev/null`

sbf-, 2022-08-21:
This feature will be implemented in the next release

closed 2022-08-27 by sbf-

ghen2, 2022-08-27:
Thanks!

Two small nits;
* it's logging the version banner twice (first three lines were already logged to stdout before this change)
* timestamps are usually not included on stdout, the service manager takes care of that

```
SBFspotUploadDaemon Version 3.0.2
Loglevel=2
LogDir=
[23:15:29.169] INFO: SBFspotUploadDaemon Version 3.0.2
[23:15:29.194] INFO: Starting Daemon...
```

ghen2, 2022-08-27:
And thirdly, could you write regular info logs to stdout (std::cout) instead of stderr?

ghen2, 2022-08-27:
ie. this works better for me:
```
        // On linux systems when logdir is empty, log to console instead of file
        std::cout << errlevelText[level] << ": " << txt << '\n';
        return rc;
```

sbf-, 2022-09-02:
See commit [62be5d5](https://github.com/SBFspot/SBFspot/commit/62be5d5fc8a45c8877cf1b790990f09c52767068)
I guess it's better to remove the "log to file" completely in one of the next releases...

habuild, 2022-09-03:
Unable to write to logfile [null/SBFspotUpload20220904.log]
logdir has to be hashed out in .cfg? not just empty?

commit 81fb5fa8 (2022-08-27): Fix #559 SBFspotUploadDaemon log to stdout?