SBFspot/SBFspot #561: Different number of SID/Serial

echoztrip, 2022-08-23:
Hi there!
Sorry if this is the wrong place to put this problem/question? I'm trying to get this running on a RPi2 with a Bluetooth adaptor.
If I leave the pvoutput details out, it installs fine, however when I put them in, I get the "Different number of SID/Serial" message.
I only have a single inverter registered with pvoutput 
[image]
I found this older post, but it didn't seem to apply to my error? https://forum.pvoutput.org/t/issue-with-sbfspot-config/1056/31

Here are some of the outputs if it helps (and let me know if there is anything more helpful?)

/usr/local/bin/sbfspot.3/SBFspot -v -ad0 -am0 -ae0 -sp0 -nocsv -nosql -finq
SBFspot V3.9.5
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -ad0 -am0 -ae0 -sp0 -nocsv -nosql -finq
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Tue Aug 23 16:18:40 2022: INFO: Starting...
sunrise: 06:26
sunset : 17:29
Connecting to 00:80:25:1D:DB:DA (1/10)
Initialising...
SUSyID: 125 - SessionID: 832863961
SMA netID=01
Serial Nr: 7E5DBBF5 (2120072181)
BT Signal=70.6%
Logon OK
Local Host Time: 23/08/2022 16:18:41
Plant Time     : 23/08/2022 16:18:41 (0 sec)
TZ offset      : 36000 sec - DST: On
Last Time Set  : 23/08/2022 16:05:04
SUSyID: 131 - SN: 2120072181
Device Name:      SN: 2120072181
Device Class:     Solar Inverters
Device Type:      SB 2000HF-30
Software Version: 02.10.18.R
Packet status: 21
SUSyID: 131 - SN: 2120072181
Device Status:      Ok
SUSyID: 131 - SN: 2120072181
Device Temperature: 18.4°C
SUSyID: 131 - SN: 2120072181
GridRelay Status:      Closed
SUSyID: 131 - SN: 2120072181
Energy Production:
        EToday: 1.517kWh
        ETotal: 16749.250kWh
        Operation Time: 41262.39h
        Feed-In Time  : 38819.26h
SUSyID: 131 - SN: 2120072181
DC Spot Data:
        MPPT 1 Pdc:   0.098kW - Udc: 368.11V - Idc:  0.268A
        MPPT 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
        Calculated Total Pdc:   0.098kW
AC Spot Data:
        Phase 1 Pac :   0.069kW - Uac: 236.28V - Iac:  0.293A
        Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   0.069kW - Calculated Pac:   0.069kW
        Efficiency  :   70.41%
SUSyID: 131 - SN: 2120072181
Grid Freq. : 49.96Hz
SUSyID: 131 - SN: 2120072181
Current Inverter Time: 23/08/2022 16:18:37
Inverter Wake-Up Time: 23/08/2022 07:53:34
Inverter Sleep Time  : 23/08/2022 16:18:41
Tue Aug 23 16:18:43 2022: INFO: Done.

And this:

SBFspot Configuration Tool V1.2.1
Running on Raspberry Pi 2 Model B Rev 1.1
Checking whiptail... Installed.
architecture=armhf
Raspbian GNU/Linux 11 (bullseye)
username=pi
userhome=/home/pi
Minimum SBFspot version supported by this tool is 3.5.0
Request to install latest version (3.9.5)
Searching for existing configuration(s)...
Reading settings from /usr/local/bin/sbfspot.3/SBFspot.cfg
/var/tmp/sbfspot.3/cfg.tmp: line 12: fg: no job control
Saving defaults to /var/tmp/sbfspot.3/sbfspot.defaults... Done
Performing some checks before installation... Done
Downloading https://github.com/sbfspot/sbfspot/releases/download/V3.9.5/sbfspot-sqlite-arm-linux-bullseye.tar.gz
Copying files...
SBFspot.db exists
Checking database...
Schema Version=1
Saving existing SBFspot.cfg... Done
Configuring SBFspot... Done
Running SBFspot...
Creating cron jobs...

## SBFspot
*/5 6-22 * * * /usr/local/bin/sbfspot.3/daydata
55 05 * * * /usr/local/bin/sbfspot.3/monthdata
NetID=01
PVO enabled, configuring upload daemon...
Device Serial(s):
Not replacing daemon (unchanged)
Different number of SID/Serial
Found 1 SIDS
Found 0 S/N
Saving defaults to /var/tmp/sbfspot.3/sbfspot.defaults... Done
Performing some checks before installation... Done
Downloading https://github.com/sbfspot/sbfspot/releases/download/V3.9.5/sbfspot-sqlite-arm-linux-bullseye.tar.gz
Copying files...
SBFspot.db exists
Checking database...
Schema Version=1
Saving existing SBFspot.cfg... Done
Configuring SBFspot... Done
Running SBFspot...
Creating cron jobs...

echoztrip, 2022-08-23:
Aaaaand I think I managed to get it working. I added this to my config (but substituted with my own values)
PVoutput_SID=SERIAL:SID
PVoutput_Key=KEY
It must be struggling to get my serial from my inverter as part of the script?

sbf-, 2022-08-24:
> It must be struggling to get my serial from my inverter as part of the script?

Yes, it's a known bug since V3.9.5 (3.9.4 is OK)
See [sbfspot-config #46](https://github.com/SBFspot/sbfspot-config/issues/46)