SBFspot/SBFspot #564: SBFSpot sometimes sends data to PVOutput with the Voltage not filled in

ZuinigeRijder, 2022-08-26:
See also this problem submitted at PVOUTPUT:
https://forum.pvoutput.org/t/pvoutput-getstatus-jsp-regularly-shows-nan-for-voltage-but-intraday-jsp-not/5663/2

SBFSpot sometimes sends data of my SMA inverter to PVOutput with the Voltage not filled in (see examples in bold below). Today 20220826 I saw this already a few times in the SBFSpot logging.

[09:42:34.874] INFO: Uploading datapoint: 20220826,09:40,44749664,384,36.8,234.38 => OK (200)
**[09:48:34.782] INFO: Uploading datapoint: 20220826,09:45,44749699,420 => OK (200)**
[09:53:34.893] INFO: Uploading datapoint: 20220826,09:50,44749739,480,36.9,234.06 => OK (200)
[09:58:34.783] INFO: Uploading datapoint: 20220826,09:55,44749776,444,37,234.3 => OK (200)
[10:04:38.194] INFO: Uploading datapoint: 20220826,10:00,44749798,264,37.1,233.43 => OK (200)
[10:09:35.199] INFO: Uploading datapoint: 20220826,10:05,44749814,192,37.2,231.35 => OK (200)
[10:15:37.990] INFO: Uploading 2 datapoints, starting with 20220826,10:10,44749827,156,37.3,232.01 => OK (200)
[10:20:34.257] INFO: Uploading datapoint: 20220826,10:20,44749858,204,37.4,232.18 => OK (200)
[10:26:38.386] INFO: Uploading datapoint: 20220826,10:25,44749881,276,37.4,232.5 => OK (200)
[10:31:34.213] INFO: Uploading datapoint: 20220826,10:30,44749907,312,37.4,232 => OK (200)
[10:37:38.767] INFO: Uploading datapoint: 20220826,10:35,44749942,420,37.5,232.37 => OK (200)
[10:42:34.662] INFO: Uploading datapoint: 20220826,10:40,44749985,516,37.6,232.4 => OK (200)
**[10:48:37.983] INFO: Uploading datapoint: 20220826,10:45,44750030,540 => OK (200)**
[10:53:34.174] INFO: Uploading datapoint: 20220826,10:50,44750071,492,37.7,232.45 => OK (200)
[10:59:33.795] INFO: Uploading datapoint: 20220826,10:55,44750107,432,37.8,231.92 => OK (200)
[11:04:35.974] INFO: Uploading datapoint: 20220826,11:00,44750141,408,37.9,231.49 => OK (200)

What can be the cause of this? Looks like it occurs regularly, e.g. once per hour. Workarounds?

Using SBFspot V3.9.4 on a Raspberry Pi Model B+ with OS 11 (bullseye)

sbf-, 2022-08-27:
Have a look in the SBFspot logging and in the vwSpotdata view for that day. I'm almost sure there will be no voltage data available for  these times. When voltage is null, PVoutput takes the previous (last known) value, shown below:

<html>
<body>

Date | Time | Energy | Efficiency | Power | Average | Normalised | Temperature | Voltage | Energy Used | Power Used
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
26/08/22 | 09:45 | 0.525kWh | 0.107kWh/kW | 420W | 420W | 0.086kW/kW | 20.6C | 234.4V | 0.000kWh | -
26/08/22 | 09:40 | 0.490kWh | 0.100kWh/kW | 384W | 384W | 0.078kW/kW | 20.6C | 234.4V | 0.000kWh | -

<br class="Apple-interchange-newline">
</body>
</html>

<html>
<body>

Date | Time | Energy | Efficiency | Power | Average | Normalised | Temperature | Voltage | Energy Used | Power Used
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
26/08/22 | 10:45 | 0.856kWh | 0.175kWh/kW | 540W | 540W | 0.110kW/kW | 20.6C | 232.4V | 0.000kWh | -
26/08/22 | 10:40 | 0.811kWh | 0.166kWh/kW | 516W | 516W | 0.105kW/kW | 20.6C | 232.4V | 0.000kWh | -

<br class="Apple-interchange-newline">
</body>
</html>

ZuinigeRijder, 2022-08-27:
> Have a look in the SBFspot logging and in the vwSpotdata view for that day. I'm almost sure there will be no voltage data available for these times. When voltage is null, PVoutput takes the previous (last known) value, shown below:
> Date 	Time 	Energy 	Efficiency 	Power 	Average 	Normalised 	Temperature 	Voltage 	Energy Used 	Power Used
> 26/08/22 	09:45 	0.525kWh 	0.107kWh/kW 	420W 	420W 	0.086kW/kW 	20.6C 	234.4V 	0.000kWh 	-
> 26/08/22 	09:40 	0.490kWh 	0.100kWh/kW 	384W 	384W 	0.078kW/kW 	20.6C 	234.4V 	0.000kWh 	-
> 
> Date 	Time 	Energy 	Efficiency 	Power 	Average 	Normalised 	Temperature 	Voltage 	Energy Used 	Power Used
> 26/08/22 	10:45 	0.856kWh 	0.175kWh/kW 	540W 	540W 	0.110kW/kW 	20.6C 	232.4V 	0.000kWh 	-
> 26/08/22 	10:40 	0.811kWh 	0.166kWh/kW 	516W 	516W 	0.105kW/kW 	20.6C 	232.4V 	0.000kWh 	-

I had a look at SBFSpot logging and in vwSpotdata.

> ********************
> * ArchiveDayData() *
> ********************
> startTime: 26/08/2022 00:00:00
> Reading events: 2022-Aug-01
> Fri Aug 26 09:42:09 2022: INFO: Done.
> SBFspot V3.9.4
> Yet another tool to read power production of SMA solar inverters
> (c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
> Compiled for Linux (LE) 32 bit with SQLite support
> Commandline Args: -v2 -ad1 -am0 -ae1 -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
> Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
> Fri Aug 26 09:47:32 2022: INFO: Starting...
> sunrise: 06:47
> sunset : 20:40
> Connecting to 00:80:25:A4:76:9D (1/10)
> Initialising...
> SUSyID: 125 - SessionID: 831863570
> SMA netID=01
> Serial Nr: 7EF614FB (2130056443)
> BT Signal=70.6%
> Logon OK
> Local Host Time: 26/08/2022 09:47:34
> Plant Time     : 26/08/2022 09:47:51 (+17 sec)
> TZ offset      : 3600 sec - DST: On
> Last Time Set  : 30/07/2022 14:04:41
> SUSyID: 138 - SN: 2130056443
> Device Name:      SN: 2130056443
> Device Class:     Solar Inverters
> Device Type:      SB 4000TL-21
> Software Version: 02.10.16.R
> Serial number:    2130056443
> Packet status: 21
> SUSyID: 138 - SN: 2130056443
> Device Status:      Ok
> SUSyID: 138 - SN: 2130056443
> Device Temperature: 36.9Â°C
> SUSyID: 138 - SN: 2130056443
> GridRelay Status:      Closed
> SUSyID: 138 - SN: 2130056443
> Energy Production:
> 	EToday: 0.546kWh
> 	ETotal: 44749.721kWh
> 	Operation Time: 42981.48h
> 	Feed-In Time  : 41516.42h
> SUSyID: 138 - SN: 2130056443
> DC Spot Data:
> 	MPPT 1 Pdc:   0.261kW - Udc: 286.04V - Idc:  0.917A
> 	MPPT 2 Pdc:   0.248kW - Udc: 277.19V - Idc:  0.897A
> 	Calculated Total Pdc:   0.509kW
> AC Spot Data:
> 	Phase 1 Pac :   0.494kW - Uac: 234.06V - Iac:  2.111A
> 	Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
> 	Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
> 	Total Pac   :   0.494kW - Calculated Pac:   0.494kW
> 	Efficiency  :   97.05%
> SUSyID: 138 - SN: 2130056443
> Grid Freq. : 49.96Hz
> SUSyID: 138 - SN: 2130056443
[... 22 lines cut ...]
+---------------------+---------------------+----------------+--------------+------------+------+------+-------+-------+--------+--------+------+------+------+-------+------+------+--------+------+------+--------+--------+------------+--------+----------+-----------+---------------+------------+-----------+--------+-----------+-------------+

```

```
sqlite> select * from vwSpotdata where TimeStamp LIKE "2022-08-26 10:%" order by TimeStamp;
+---------------------+---------------------+----------------+--------------+------------+------+------+-------+-------+--------+--------+------+------+------+-------+------+------+--------+------+------+--------+--------+------------+--------+----------+-----------+---------------+------------+-----------+--------+-----------+-------------+
|      TimeStamp      |     Nearest5min     |      Name      |     Type     |   Serial   | Pdc1 | Pdc2 | Idc1  | Idc2  |  Udc1  |  Udc2  | Pac1 | Pac2 | Pac3 | Iac1  | Iac2 | Iac3 |  Uac1  | Uac2 | Uac3 | PdcTot | PacTot | Efficiency | EToday |  ETotal  | Frequency | OperatingTime | FeedInTime | BT_Signal | Status | GridRelay | Temperature |
+---------------------+---------------------+----------------+--------------+------------+------+------+-------+-------+--------+--------+------+------+------+-------+------+------+--------+------+------+--------+--------+------------+--------+----------+-----------+---------------+------------+-----------+--------+-----------+-------------+
| 2022-08-26 10:03:54 | 2022-08-26 10:05:00 | SN: 2130056443 | SB 4000TL-21 | 2130056443 | 108  | 102  | 0.392 | 0.395 | 277.65 | 260.41 | 185  | 0    | 0    | 0.8   | 0.0  | 0.0  | 231.35 | 0.0  | 0.0  | 210    | 185    | 88.1       | 637    | 44749811 | 49.96     | 42981.7       | 41516.7    | 71.0      | OK     | Closed    | 37.2        |
| 2022-08-26 10:09:24 | 2022-08-26 10:10:00 | SN: 2130056443 | SB 4000TL-21 | 2130056443 | 99   | 94   | 0.358 | 0.357 | 277.65 | 264.61 | 165  | 0    | 0    | 0.712 | 0.0  | 0.0  | 232.01 | 0.0  | 0.0  | 193    | 165    | 85.5       | 652    | 44749827 | 49.98     | 42981.8       | 41516.8    | 71.0      | OK     | Closed    | 37.3        |
| 2022-08-26 10:14:44 | 2022-08-26 10:15:00 | SN: 2130056443 | SB 4000TL-21 | 2130056443 | 110  | 105  | 0.4   | 0.392 | 277.65 | 268.8  | 190  | 0    | 0    | 0.822 | 0.0  | 0.0  | 231.49 | 0.0  | 0.0  | 215    | 190    | 88.4       | 667    | 44749841 | 49.98     | 42981.9       | 41516.9    | 71.0      | OK     | Closed    | 37.4        |
[... cut ...]  | Closed    | 37.9        |
+---------------------+---------------------+----------------+--------------+------------+------+------+-------+-------+--------+--------+------+------+------+-------+------+------+--------+------+------+--------+--------+------------+--------+----------+-----------+---------------+------------+-----------+--------+-----------+-------------+
```

But why does it then send the following:
[09:48:34.782] INFO: Uploading datapoint: 20220826,09:45,44749699,420 => OK (200)

sbf-, 2022-08-27:
> Would that be the root cause?

Yes

> Should always the daydata be triggered at exact 5 minutes interval?

More or less at 5min interval, but in your case it runs sometimes 1 or 2 mins later. So it looks like there is an issue with the cron job

> But why does it then send the following:
[09:48:34.782] INFO: Uploading datapoint: 20220826,09:45,44749699,420 => OK (200)

When there is no data in the spotdata table to match with the daydata, SBFspot has nothing to send.

This is my data:
```
sqlite> select timestamp,nearest5min from vwSpotdata where TimeStamp LIKE "2022-08-26 10:%" order by TimeStamp;
2022-08-26 10:00:03|2022-08-26 10:00:00
2022-08-26 10:05:04|2022-08-26 10:05:00
2022-08-26 10:10:04|2022-08-26 10:10:00
2022-08-26 10:15:04|2022-08-26 10:15:00
2022-08-26 10:20:03|2022-08-26 10:20:00
2022-08-26 10:25:03|2022-08-26 10:25:00
2022-08-26 10:30:04|2022-08-26 10:30:00
2022-08-26 10:35:03|2022-08-26 10:35:00
2022-08-26 10:40:04|2022-08-26 10:40:00
2022-08-26 10:45:04|2022-08-26 10:45:00
2022-08-26 10:50:03|2022-08-26 10:50:00
2022-08-26 10:55:03|2022-08-26 10:55:00
```

ZuinigeRijder, 2022-08-28:
Still strange that data IS send:

> [09:48:34.782] INFO: Uploading datapoint: 20220826,09:45,44749699,420 => OK (200)

Because my external USB bluetooth sometimes stops working, I do a reset. That is why it does not do at at almost exact 5 minutes intervals. I will adapt my script, so the daydata is called at almost exact 5 minutes interval. Therefore this issue can be closed for now. Thanks for the work and help.

closed 2022-08-28 by ZuinigeRijder

sbf-, 2022-08-29:
> Still strange that data IS send:
> 
> > [09:48:34.782] INFO: Uploading datapoint: 20220826,09:45,44749699,420 => OK (200)

This is by design...
The inverter keeps 5min data for about 62 days (stored in sbfspot.daydata table) and momentary data like ac/dc voltage etc is not kept by the inverter but is stored in sbfspot.spotdata table.
When you are missing the spotdata on a certain 5min interval there is null data for voltage (left join in vwPvoData)
Now, if we don't send the power/generated energy when there is no voltage your PVO graph will be messed up by the gaps.