SBFspot/SBFspot #567: Not writing information to database

ziogref, 2022-09-09:
I have a SMA Sunnyboy 6kw inverter connected to the network using speedwire. (I also have a SMA Home Manager 2.0, but I havent tried talking to that yet, this is just about the inverter)

I am following the install process on a Ubuntu 22.04 Server and I get to the "Testing" section. I run `/usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv`

and the output I get is 

`administrator@pidgey:~/smadata$ /usr/local/bin/sbfspot.3/SBFspot -v -finq -nocsv
SBFspot V3.9.6
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -v -finq -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri Sep  9 14:54:35 2022: INFO: Starting...
sunrise: 19:04
sunset : 06:33
Connecting to Local Network...
Initialising...
SUSyID: 125 - SessionID: 948292952
Logon OK
Segmentation fault (core dumped)
`
I then open the database as instructed I dont get any outputs. This is copied from my terminal 
`administrator@pidgey:~/smadata$ sqlite3 SBFspot.db
SQLite version 3.37.2 2022-01-06 13:25:41
Enter ".help" for usage hints.
sqlite> select * from vwspotdata;
sqlite> select * from inverters;
sqlite> .quit
`
I'm not familiar with databases, but I assume no response means that the table is empty.

Can you please assist?

sbf-, 2022-09-09:
`Segmentation fault (core dumped)` in the output means the program crashed somewhere.
It looks like this happens right after the logon. The HM2 might be the cause of this.
Try with `/usr/local/bin/sbfspot.3/SBFspot -v5 -d5 -finq -nocsv -nosql` for detailed logging

ziogref, 2022-09-09:
same issue

```
administrator@pidgey:~$ /usr/local/bin/sbfspot.3/SBFspot -v5 -d5 -finq -nocsv -nosql
SBFspot V3.9.6
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -v5 -d5 -finq -nocsv -nosql
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'

Configuration settings:
Password=<undisclosed>
Plantname=Jardinelle
OutputPath=/home/administrator/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=147.43
Longitude=-42.89
Timezone=Australia/Hobart
Locale=en-US
CalculateMissingSpotValues=1
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
SQL_Database=/home/administrator/smadata/SBFspot.db
End of Config

Fri Sep  9 20:12:00 2022: INFO: Starting...
sunrise: 19:04
sunset : 06:33
Connecting to Local Network...
Initialising...
SUSyID: 125 - SessionID: 804802572
logonSMAInverter()
Logon OK
Segmentation fault (core dumped)
```

sbf-, 2022-09-09:
You stripped the debug log and part of the config settings? That's not the -d5 -v5 log

ziogref, 2022-09-09:
> You stripped the debug log and part of the config settings? That's not the -d5 -v5 log

I ran exactly what you sent. This was my 2nd attempt at an install so I followed the guide very closely changing only what was necessary. Here is my config

```
################################################################################
#                     ____  ____  _____                _   
#                    / ___|| __ )|  ___|__ _ __   ___ | |_ 
#                    \___ \|  _ \| |_ / __| '_ \ / _ \| __|
#                     ___) | |_) |  _|\__ \ |_) | (_) | |_ 
#                    |____/|____/|_|  |___/ .__/ \___/ \__|
#                                         |_|
#
#  SBFspot.cfg - Configuration file for SBFspot.exe
#  SBFspot - Yet another tool to read power production of SMA solar inverters
#  (c)2012-2022, SBF
#
#  DISCLAIMER:
#  A user of SBFspot software acknowledges that he or she is receiving this
#  software on an "as is" basis and the user is not relying on the accuracy 
#  or functionality of the software for any purpose. The user further
#  acknowledges that any use of this software will be at his own risk 
#  and the copyright owner accepts no responsibility whatsoever arising from
#  the use or application of the software. 
#
#       SMA and Speedwire are registered trademarks of SMA Solar Technology AG
################################################################################

# SMA Inverter's Bluetooth address
# Windows: sbfspot -scan
# Linux  : hcitool scan
# IMPORTANT FOR SPEEDWIRE USERS: COMMENT OUT BTADDRESS (PUT # IN FRONT)
#BTAddress=00:00:00:00:00:00

# If multiple local BT adapters are available, select the one to use
#LocalBTAddress=00:00:00:00:00:00

# SMA Inverter's Speedwire IP address
# If IP_Address is not set or is 0.0.0.0 SBFspot will try to detect the speedwire inverter by broadcast
# If IP_Address is set to a valid IP, SBFspot will try to connect directly to that IP without broadcast detection
# Multiple IP addresses can be provided (comma separated)
#IP_Address=0.0.0.0

# User password (default 0000)
Password=JHh5Gahqyfn87?h?

# MIS_Enabled (Multi Inverter Support: Default=0 Disabled)
# +------------+-------+-------------+
# | #Inverters | NetID | MIS_Enabled |
# +------------+-------+-------------+
# |      1     |   1   | Don't Care  |
# +------------+-------+-------------+
# |      1     |   >1  |      0      |
# +------------+-------+-------------+
# |      >1    |   >1  |      1      |
# +------------+-------+-------------+
MIS_Enabled=0

# Plantname
Plantname=Jardinelle

# OutputPath (Place to store CSV files)
# 
# Windows: C:\Users\Public\SMAdata\%Y
# Linux  : /home/pi/smadata/%Y
# %Y %m and %d will be expanded to Year Month and Day
OutputPath=/home/administrator/smadata/%Y

# OutputPathEvents (Place to store CSV files for events)
# If omitted, OutputPath is used
[... 185 lines cut ...]
# IDC1/IDC2/IDC     DcMsAmp                 DC current input string 1/2/All
# PDCTot            DcMsWatt                DC power all strings
# OperTm            MeteringTotOpTms        Operating time
# FeedTm            MeteringTotFeedTms      Feed-in time
# PAC1              GridMsWphsA             Power L1
# PAC2              GridMsWphsB             Power L2
# PAC3              GridMsWphsC             Power L3
# UAC1              GridMsPhVphsA           Grid voltage phase L1
# UAC2              GridMsPhVphsB           Grid voltage phase L2
# UAC3              GridMsPhVphsC           Grid voltage phase L3
# IAC1              GridMsAphsA_1           Grid current phase L1
# IAC2              GridMsAphsB_1           Grid current phase L2
# IAC3              GridMsAphsC_1           Grid current phase L3
# GridFreq          GridMsHz                Grid frequency
# BTSignal          BTSignal                Bluetooth signal strength
# BatTmpVal         BatTmpVal               Battery temperature
# BatVol            BatVol                  Battery voltage
# BatAmp            BatAmp                  Battery current
# BatChaStt         BatChaStt               Current battery charge status
# InvWakeupTm                               Inverter Wakeup date/time
# InvSleepTm                                Inverter Sleep date/time
# MeteringWOut      MeteringGridMsTotWOut   Power sent towards the Grid (W)
# MeteringWIn       MeteringGridMsTotWIn    Power drawn from the Grid (W)
# MeteringWTot                              Power sent/drawn (negative when sending towards the net)
```

sbf-, 2022-09-09:
> I ran exactly what you sent

Sorry, I see what happens... You have no IP address in your config (it is commented out with #)
`#IP_Address=0.0.0.0`

Remove the # and if known, fill in the IP address of the inverter (0.0.0.0 will probably not work with your HM2)

ziogref, 2022-09-09:
> > I ran exactly what you sent
> 
> Sorry, I see what happens... You have no IP address in your config (it is commented out with #) `#IP_Address=0.0.0.0`
> 
> Remove the # and if known, fill in the IP address of the inverter (0.0.0.0 will probably not work with your HM2)

Wow, im such a dumbass. Thanks mate, I will continue the config, it seems to be working as Im getting responses now

sbf-, 2022-09-09:
Actually it's a bug... I'll fix it to avoid this

commit 74864ad9 (2022-09-13): Fix #567 Segmentation fault when connection address is missing
closed 2022-09-13 by sbf-