SBFspot/SBFspot #576: CRITICAL: Failed to initialise Speedwire connection.

tobiasbender, 2022-09-17:
I receive sporadically a CRITICAL: Failed to initialise Speedwire connection. The device is pingable, still I get the error. 

Could you explain me what I can do to find the root cause of the issue?

Here the logs:
```
SBFspot V3.9.6
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -v -ad1 -am0 -ae1 -mqtt -cfg/usr/local/bin/sbfspot.3/SBFspot.cfg
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Sat Sep 17 18:15:01 2022: INFO: Starting...
sunrise: 07:05
sunset : 19:33
Connecting to Local Network...
Initialising...
SUSyID: 125 - SessionID: 853668308
Device IP address: 192.168.0.181 from config
ERROR: Connection to inverter failed!
Is 192.168.0.181 a correct IP?
Sat Sep 17 18:15:03 2022: CRITICAL: Failed to initialise Speedwire connection.
Sat Sep 17 18:15:03 2022: INFO: Done.
```

and here the corresponding wireshark log:
`
Capturing on 'wlan0'
    1 0.000000000 192.168.0.206 → 192.168.0.181 UDP 100 9522 → 9522 Len=58
Frame 1: 100 bytes on wire (800 bits), 100 bytes captured (800 bits) on interface wlan0, id 0
    Interface id: 0 (wlan0)
        Interface name: wlan0
    Encapsulation type: Ethernet (1)
    Arrival Time: Sep 17, 2022 18:15:01.765437041 CEST
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1663431301.765437041 seconds
    [Time delta from previous captured frame: 0.000000000 seconds]
    [Time delta from previous displayed frame: 0.000000000 seconds]
    [Time since reference or first frame: 0.000000000 seconds]
    Frame Number: 1
    Frame Length: 100 bytes (800 bits)
    Capture Length: 100 bytes (800 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:ip:udp:data]
Ethernet II, Src: Raspberr_27:66:ac (e4:5f:01:27:66:ac), Dst: SmaRegel_9f:bc:12 (00:40:ad:9f:bc:12)
    Destination: SmaRegel_9f:bc:12 (00:40:ad:9f:bc:12)
        Address: SmaRegel_9f:bc:12 (00:40:ad:9f:bc:12)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Source: Raspberr_27:66:ac (e4:5f:01:27:66:ac)
        Address: Raspberr_27:66:ac (e4:5f:01:27:66:ac)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: IPv4 (0x0800)
Internet Protocol Version 4, Src: 192.168.0.206, Dst: 192.168.0.181
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
        0000 00.. = Differentiated Services Codepoint: Default (0)
        .... ..00 = Explicit Congestion Notification: Not ECN-Capable Transport (0)
    Total Length: 86
    Identification: 0xa37e (41854)
    Flags: 0x40, Don't fragment
        0... .... = Reserved bit: Not set
        .1.. .... = Don't fragment: Set
        ..0. .... = More fragments: Not set
    Fragment Offset: 0
    Time to Live: 64
    Protocol: UDP (17)
    Header Checksum: 0x1445 [validation disabled]
    [Header checksum status: Unverified]
    Source Address: 192.168.0.206
    Destination Address: 192.168.0.181
User Datagram Protocol, Src Port: 9522, Dst Port: 9522
    Source Port: 9522
    Destination Port: 9522
    Length: 66
    Checksum: 0xfc5c [unverified]
    [Checksum Status: Unverified]
    [Stream index: 0]
    [Timestamps]
        [Time since first frame: 0.000000000 seconds]
        [Time since previous frame: 0.000000000 seconds]
    UDP payload (58 bytes)
Data (58 bytes)

0000  53 4d 41 00 00 04 02 a0 00 00 00 01 00 26 00 10   SMA..........&..
0010  60 65 09 a0 ff ff ff ff ff ff 00 00 7d 00 d4 f1   `e..........}...
0020  e1 32 00 00 00 00 00 00 01 80 00 02 00 00 00 00   .2..............
0030  00 00 00 00 00 00 00 00 00 00                     ..........
    Data: 534d4100000402a00000000100260010606509a0ffffffffffff00007d00d4f1e1320000…
    [Length: 58]

    2 5.141361128 Raspberr_27:66:ac → SmaRegel_9f:bc:12 ARP 42 Who has 192.168.0.181? Tell 192.168.0.206
Frame 2: 42 bytes on wire (336 bits), 42 bytes captured (336 bits) on interface wlan0, id 0
    Interface id: 0 (wlan0)
        Interface name: wlan0
    Encapsulation type: Ethernet (1)
    Arrival Time: Sep 17, 2022 18:15:06.906798169 CEST
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1663431306.906798169 seconds
    [Time delta from previous captured frame: 5.141361128 seconds]
    [Time delta from previous displayed frame: 5.141361128 seconds]
    [Time since reference or first frame: 5.141361128 seconds]
    Frame Number: 2
    Frame Length: 42 bytes (336 bits)
    Capture Length: 42 bytes (336 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:arp]
Ethernet II, Src: Raspberr_27:66:ac (e4:5f:01:27:66:ac), Dst: SmaRegel_9f:bc:12 (00:40:ad:9f:bc:12)
    Destination: SmaRegel_9f:bc:12 (00:40:ad:9f:bc:12)
        Address: SmaRegel_9f:bc:12 (00:40:ad:9f:bc:12)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Source: Raspberr_27:66:ac (e4:5f:01:27:66:ac)
        Address: Raspberr_27:66:ac (e4:5f:01:27:66:ac)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: ARP (0x0806)
Address Resolution Protocol (request)
    Hardware type: Ethernet (1)
    Protocol type: IPv4 (0x0800)
    Hardware size: 6
    Protocol size: 4
    Opcode: request (1)
    Sender MAC address: Raspberr_27:66:ac (e4:5f:01:27:66:ac)
    Sender IP address: 192.168.0.206
    Target MAC address: 00:00:00_00:00:00 (00:00:00:00:00:00)
    Target IP address: 192.168.0.181

    3 5.146411042 SmaRegel_9f:bc:12 → Raspberr_27:66:ac ARP 60 192.168.0.181 is at 00:40:ad:9f:bc:12
Frame 3: 60 bytes on wire (480 bits), 60 bytes captured (480 bits) on interface wlan0, id 0
    Interface id: 0 (wlan0)
        Interface name: wlan0
    Encapsulation type: Ethernet (1)
    Arrival Time: Sep 17, 2022 18:15:06.911848083 CEST
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1663431306.911848083 seconds
    [Time delta from previous captured frame: 0.005049914 seconds]
    [Time delta from previous displayed frame: 0.005049914 seconds]
    [Time since reference or first frame: 5.146411042 seconds]
    Frame Number: 3
    Frame Length: 60 bytes (480 bits)
    Capture Length: 60 bytes (480 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:arp]
Ethernet II, Src: SmaRegel_9f:bc:12 (00:40:ad:9f:bc:12), Dst: Raspberr_27:66:ac (e4:5f:01:27:66:ac)
    Destination: Raspberr_27:66:ac (e4:5f:01:27:66:ac)
        Address: Raspberr_27:66:ac (e4:5f:01:27:66:ac)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Source: SmaRegel_9f:bc:12 (00:40:ad:9f:bc:12)
        Address: SmaRegel_9f:bc:12 (00:40:ad:9f:bc:12)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: ARP (0x0806)
    Padding: 000000000000000000000000000000000000
Address Resolution Protocol (reply)
    Hardware type: Ethernet (1)
    Protocol type: IPv4 (0x0800)
    Hardware size: 6
    Protocol size: 4
    Opcode: reply (2)
    Sender MAC address: SmaRegel_9f:bc:12 (00:40:ad:9f:bc:12)
    Sender IP address: 192.168.0.181
    Target MAC address: Raspberr_27:66:ac (e4:5f:01:27:66:ac)
    Target IP address: 192.168.0.206
`

sbf-, 2022-09-18:
Well, I guess the wlan connection is not optimal

tobiasbender, 2022-09-27:
I have to admit: I cabled the connection and did not have a connection issue since then. (surprisingly, since I have AP an RPI in 2m distance. ) In case I get other issues next year in summer, I will reopen. Sorry for the noise.

closed 2022-09-27 by tobiasbender