SBFspot/SBFspot #578: Corrupted/Future data from Inverter

SBFspot, 2022-09-23:
### Discussed in https://github.com/SBFspot/SBFspot/discussions/577

<div type='discussions-op-text'>

<sup>Originally posted by **alefhar** September 22, 2022</sup>
Since a few days my inverter seems to be acting up and has old data stored for the current day. The data up to the current 5-minute interval gets overwritten but data that has a timestamp in the future is stale. This causes `./SBFspot -ad1` to read the correct data up to the current time and the corrupted, future data. Everything is then written to the database and uploaded to PVOutput. The values in the database aren't overwritten when new data becomes available.
I have an example here: [pasebin](https://pastebin.com/kU6PXiCv)
There is a very high power value and after that the total energy is lower and power is always zero:
```
21/09/2022 11:40:00 : 6441.234kWh - 456.000W
21/09/2022 11:45:00 : 5672.121kWh - 61489146903135816.000W
21/09/2022 11:50:00 : 5672.121kWh - 0.000W
```
Currently, I've changed my cron jobs to archive day data only once in the evening.

Maybe SBFspot could not accept data with timestamps in the future?</div>

diedvdyk, 2022-10-11:
I am seeing this bug on my inverter too. My inverter haves sometimes data in the future.
So I support the question : Maybe SBFspot could not accept data with timestamps in the future?

Driekes, 2023-06-16:
I have this issue and it appeared with all invertors I had. One got replaced, but the latest one now starts to show data of the past (it could be that this data was stored in the past for 'today', e.g. I had data recorded between 0:00 and 08:00, with data aprox 2 weeks ago.)

It is very strange behaviour and I suspect just some bug/mistake in the software of SMA. However, it would still be great if SBFspot can somehow see/handle big jumps in data and discard them and/or put them in a 'possible rubbish' column.

commit d6c259ce (2024-01-23): Refactored ArchiveDayData()
Fixed #578 Corrupted/Future data from Inverter
Fixed #635 The mystery of the sole 2.5 min interval datapoint... and it's corrupt