SBFspot/SBFspot #580: Frequency and logging of SBFspotUploadDaemon

arminus, 2022-10-21:
After I have crashed an SD card in a Raspi recently due to excessive writes by telegraf, I've become a bit sensitive regarding who writes how much and how often on any device with an SD card.

Basically, I don't need live data on pvoutput.org, nor do I need the logs. But so far, I could not find any documentation wether it's possible to configure the frequency of SBFspotUploadDaemon's uploads, nor did I manage to send its log files to /dev/null by setting LogDir.

Is there any way to configure these aspects?

Otherwise, I guess I'll just write a script which starts SBFspotUploadDaemon after sundown and kills it some 10 minutes later again...

Other than that, thanks a lot for SBFspot, runs flawlessly since years :-)

sbf-, 2022-10-22:
V3.9.6 daemon logs to stdout instead of file (set logdir to empty value or just remove it from config file)
Upload frequency is not configurable.

FutureCow, 2022-11-15:
I've added this to my crontab:

0 23 * * * /usr/bin/timeout 300s /usr/local/bin/sbfspot.3/./SBFspotUploadDaemon