SBFspot/SBFspot #618: Out of range value for column 'TotalYield' at row 1

TerrorSource, 2023-02-03:
I'm using SBFspot in Docker on my Synology (using https://github.com/nakla/sbfspot for docker)
Letting SBFspot writing the data in MariaDB

When i check the logs i see the following:
[2023-02-03 12:42:12.404] Error: [month_data]mysql_stmt_execute() returned : Out of range value for column 'TotalYield' at row 1

How can i fix this?

TerrorSource, 2023-02-03:
This did not fix the issue:
ALTER TABLE `MonthData` CHANGE `TotalYield` `TotalYield` BIGINT(16) NULL DEFAULT NULL;
ALTER TABLE `DayData` CHANGE `TotalYield` `TotalYield` BIGINT(16) NULL DEFAULT NULL;

sbf-, 2023-02-03:
Probably the value is invalid (too big)
Try with verbose logging (-v5) to check

TerrorSource, 2023-02-03:
@sbf- I run -v5 and noticed something weird:

2023-02-03T14:25:27.724559202Z | stdout | [2023-02-03 14:25:27.723] Error: [month_data]mysql_stmt_execute() returned : Out of range value for column 'DayYield' at row 1
-- | -- | --
2023-02-03T14:25:27.723363107Z | stdout | ======
2023-02-03T14:25:27.723322042Z | stdout | 28/02/2021 : 4294967.157kWh - 0.045kWh
2023-02-03T14:25:27.723279987Z | stdout | 27/02/2021 : 4294967.112kWh - -0.078kWh
2023-02-03T14:25:27.723226914Z | stdout | 26/02/2021 : 4294967.190kWh - -0.074kWh
2023-02-03T14:25:27.723181913Z | stdout | 25/02/2021 : 4294967.264kWh - 0.162kWh
2023-02-03T14:25:27.723129484Z | stdout | 24/02/2021 : 4294967.102kWh - -0.109kWh
2023-02-03T14:25:27.723083999Z | stdout | 23/02/2021 : 4294967.211kWh - 0.110kWh
2023-02-03T14:25:27.723041884Z | stdout | 22/02/2021 : 4294967.101kWh - 0.049kWh
2023-02-03T14:25:27.722998633Z | stdout | 21/02/2021 : 4294967.052kWh - -0.053kWh
2023-02-03T14:25:27.722941424Z | stdout | 20/02/2021 : 4294967.105kWh - -0.089kWh
2023-02-03T14:25:27.722892200Z | stdout | 19/02/2021 : 4294967.194kWh - 0.154kWh
2023-02-03T14:25:27.722840829Z | stdout | 18/02/2021 : 4294967.040kWh - -0.090kWh
2023-02-03T14:25:27.722798718Z | stdout | 17/02/2021 : 4294967.130kWh - 0.003kWh
2023-02-03T14:25:27.722756986Z | stdout | 16/02/2021 : 4294967.127kWh - -0.124kWh
2023-02-03T14:25:27.722712551Z | stdout | 15/02/2021 : 4294967.251kWh - 0.080kWh
2023-02-03T14:25:27.722669852Z | stdout | 14/02/2021 : 4294967.171kWh - -0.121kWh
2023-02-03T14:25:27.722619206Z | stdout | 13/02/2021 : 4294967.292kWh - 0.235kWh
2023-02-03T14:25:27.722577987Z | stdout | 12/02/2021 : 4294967.057kWh - -0.071kWh
2023-02-03T14:25:27.722534628Z | stdout | 11/02/2021 : 4294967.128kWh - 0.064kWh
2023-02-03T14:25:27.722444537Z | stdout | 10/02/2021 : 4294967.064kWh - -0.102kWh
2023-02-03T14:25:27.722401923Z | stdout | 09/02/2021 : 4294967.166kWh - 0.054kWh
2023-02-03T14:25:27.722351642Z | stdout | 08/02/2021 : 4294967.112kWh - -0.123kWh
2023-02-03T14:25:27.722304796Z | stdout | 07/02/2021 : 4294967.235kWh - 0.146kWh
2023-02-03T14:25:27.722184504Z | stdout | 06/02/2021 : 4294967.089kWh - 0.028kWh
2023-02-03T14:25:27.722142596Z | stdout | 05/02/2021 : 4294967.061kWh - -0.001kWh
2023-02-03T14:25:27.722100198Z | stdout | 04/02/2021 : 4294967.062kWh - -0.155kWh
2023-02-03T14:25:27.722057452Z | stdout | 03/02/2021 : 4294967.217kWh - -0.050kWh
2023-02-03T14:25:27.722014913Z | stdout | 02/02/2021 : 4294967.267kWh - 0.050kWh
2023-02-03T14:25:27.721963644Z | stdout | 01/02/2021 : 4294967.217kWh - -0.050kWh
2023-02-03T14:25:27.721774205Z | stdout | SUSyID: 138 - SN: 2130306449
2023-02-03T14:25:27.656761895Z | stdout | startTime: 01/02/2021 12:00:00
2023-02-03T14:25:27.656711538Z | stdout | **********************
2023-02-03T14:25:27.656670150Z | stdout | * ArchiveMonthData() *
2023-02-03T14:25:27.656626495Z | stdout | **********************
2023-02-03T14:25:27.656560306Z | stdout | [2023-02-03 14:25:27.655] Error: [month_data]mysql_stmt_execute() returned : Out of range value for column 'DayYield' at row 1

sbf-, 2023-02-03:
Indeed strange those negative values... Never seen this before.
Inverter is SB TL-21 series: SB3000/3600/4000/5000 I guess?
Can you provide a debug log?
`SBFspot -d5 -v5 -finq -nocsv -nosql >debug.log`

TerrorSource, 2024-04-15:
> Indeed strange those negative values... Never seen this before. Inverter is SB TL-21 series: SB3000/3600/4000/5000 I guess? Can you provide a debug log? `SBFspot -d5 -v5 -finq -nocsv -nosql >debug.log`

i deleted the sbfspot docker from my NAS and trying to run it now again from inside HomeAssistant (Docker). Running into the same issue.

SBFspot V3.9.7
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -d5 -v5 -finq -nocsv -nosql
Reading config '/usr/bin/sbfspot/SBFspot.cfg'

Configuration settings:
IP_Address=192.168.1.15
Password=<undisclosed>
Plantname=MyPlant
OutputPath=/data/sbfspot/%Y
OutputPathEvents=/data/sbfspot/%Y/events
Latitude=51.832
Longitude=4.31
Timezone=Europe/Amsterdam
Locale=en-US
CalculateMissingSpotValues=1
DateTimeFormat=%Y-%m-%dT%H:%M:%S%z
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S%
SynchTime=0
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
SQL_Database=SBFspot
SQL_Hostname=192.168.1.9
SQL_Port=3306
SQL_Username=sbfspot
SQL_Password=<undisclosed>
End of Config

Mon Apr 15 21:23:44 2024: INFO: Starting...
sunrise: 06:46
sunset : 20:39
Connecting to Local Network...
Initialising...
SUSyID: 125 - SessionID: 968619825
Device IP address: 192.168.1.15 from config
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 26 00 10 60 65 09 A0 
00000020: FF FF FF FF FF FF 00 00 7D 00 
00000030: 31 F7 BB 39 00 00 00 00 00 00 
00000040: 01 80 00 02 00 00 00 00 00 00 
00000050: 00 00 00 00 00 00 00 00 
58 Bytes sent to [192.168.1.15]
MAX_CommBuf is now 98 bytes
98 bytes received from [192.168.1.15]
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 4E 00 10 60 65 13 90 
00000020: 7D 00 31 F7 BB 39 00 00 8A 00 
00000030: 91 E5 F9 7E 00 00 00 00 00 00 
00000040: 01 80 01 02 00 00 00 00 00 00 
00000050: 00 00 00 00 00 03 00 00 00 FF 
00000060: 00 00 3C 74 00 20 01 00 8A 00 
[... 1925 lines cut ...]
<<<=================================>>>
Events for device 138:2130306449
2024-04-09T03:18:10+0200 Webconnect error: No reply to SIP option packs
2024-04-09T02:14:08+0200 Webconnect error: No reply to SIP option packs
2024-04-08T18:50:42+0200 Time adjusted / new time
2024-04-08T18:50:53+0200 Time adjusted / old time
2024-04-05T04:05:03+0200 Webconnect error: no connection
2024-04-05T03:05:03+0200 SMACOM A: communication status goes to "Ok"
2024-04-05T03:05:03+0200 SMACOM A: connection speed goes to "100 Mbit/s"
2024-04-05T03:05:00+0200 SMACOM A: communication status goes to "No connection"
2024-04-05T03:04:53+0200 SMACOM A: communication status goes to "Ok"
2024-04-05T03:04:53+0200 SMACOM A: connection speed goes to "100 Mbit/s"
2024-04-05T03:03:34+0200 Webconnect error: no connection
2024-04-05T03:03:34+0200 SMACOM A: communication status goes to "No connection"
logoffMultigateDevices()
logoffSMAInverter()
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 22 00 10 60 65 08 A0 
00000020: FF FF FF FF FF FF 00 03 7D 00 
00000030: 31 F7 BB 39 00 03 00 00 00 00 
00000040: 16 80 0E 01 FD FF FF FF FF FF 
00000050: 00 00 00 00 
54 Bytes sent to [192.168.1.15]
Mon Apr 15 21:23:45 2024: INFO: Done.

TerrorSource, 2024-04-16:
@sbf- is this something you can work with?

sbf-, 2024-04-16:
> @sbf- is this something you can work with?

Sorry, but I don't see any issues in this log. Everything looks OK here.

TerrorSource, 2024-04-16:
> > @sbf- is this something you can work with?
> 
> Sorry, but I don't see any issues in this log. Everything looks OK here.

is there anything i can retreive via SBFspot or config's to sort this out?

If you want we can do a remote session so you can try whatever you want with my Docker/config :)

sbf-, 2024-04-16:
Execute the same debug command without the `-nosql` so I can see the SQL error

`SBFspot -d5 -v5 -finq -nocsv >debug.txt`

And attach the debug.log to this thread rather than posting the huge text (less scrolling :-)

TerrorSource, 2024-04-16:
> Execute the same debug command without the `-nosql` so I can see the SQL error
> 
> `SBFspot -d5 -v5 -finq -nocsv >debug.txt`
> 
> And attach the debug.log to this thread rather than posting the huge text (less scrolling :-)

Debug.txt file as requested
[debug.txt](https://github.com/SBFspot/SBFspot/files/15001640/debug.txt)

sbf-, 2024-04-19:
Where is the initial error (`Out of range value for column 'TotalYield' at row 1`) in the debug file?

I really can't see the issue here

TerrorSource, 2024-04-19:
> Where is the initial error (`Out of range value for column 'TotalYield' at row 1`) in the debug file?
> 
> I really can't see the issue here

i exactly did what you asked me to do. Create logs with -nosql and without. 

What you are saying is that without -nosql it works as expected?

sbf-, 2024-04-19:
> What you are saying is that without `-nosql` it works as expected?

With -nosql it's normal you don't see the `Error: [month_data]mysql_stmt_execute() returned` because the SQL functions are not called.

In your OP you wrote "When i check the logs i see the following" but in the logs you provide now there is no trace of this error.

So, what exactly goes wrong in HA?

closed 2024-06-15 by sbf-