SBFspot/SBFspot #622: logon via BT Failed 'CRITICAL: Logon failed. Check 'USER' Password'

914tom, 2023-02-20:
Hi,
I try to logon on my SUNNY Boy 5000TL via BT an get the Error 
' Mon Feb 20 17:31:23 2023: CRITICAL: Logon failed. Check 'USER' Password'
It is still the standard password (0000) set. 
Windows Sonny Explorer tell me to change it, but does this since 10 Years.
This is the verbose Dump, hope you can help me out, 
btw, LINUX Mint 21, HP Thin Client.

thank you, greetings
tom

SBFspot V3.9.6
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -v -finq -nocsv -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'

Configuration settings:
BTAddress=00:80:25:A4:8B:97
MIS_Enabled=0
BTConnectRetries=10
Password=<undisclosed>
Plantname=Solaranlage
OutputPath=/home/tom/smadata/%Y
OutputPathEvents=/home/tom/smadata/%Y/Events
Latitude= 
Longitude=
Timezone=Europe/Brussels
Locale=de-DE
CalculateMissingSpotValues=1
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
SQL_Database=/home/tom/smadata/SBFspot.db
End of Config

Mon Feb 20 17:31:21 2023: INFO: Starting...
sunrise: 07:17
sunset : 17:41
Connecting to 00:80:25:A4:8B:97 (1/10)
Initialising...
SUSyID: 125 - SessionID: 921025066
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 97 8B A4 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 04 
00000020: 70 00 01 00 00 00 00 01 00 00 
00000030: 00 
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 97 8B A4 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 04 
00000020: 70 00 01 00 00 00 00 01 00 00 
00000030: 00 
<<<=================================>>>
MAX_pcktBuf is now 31 bytes
SMA netID=01
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 00 00 00 00 00 00 
00000010: 97 8B A4 25 80 00 02 00 00 04 
00000020: 70 00 01 00 00 00 00 01 00 00 
00000030: 00 
31 Bytes sent
getPacket(5)
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 97 8B A4 25 80 00 
00000010: 00 00 00 00 00 00 0A 00 97 8B 
00000020: A4 25 80 00 01 77 15 01 5B 02 
00000030: 00 
cmd=10
Received 18 bytes
Received 2 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 14 00 6A 97 8B A4 25 80 00 
00000010: 00 00 00 00 00 00 0C 00 02 00 
cmd=12
Received 18 bytes
Received 16 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 22 00 5C 97 8B A4 25 80 00 
00000010: 00 00 00 00 00 00 05 00 97 8B 
00000020: A4 25 80 00 01 01 77 15 01 5B 
00000030: 02 00 02 01 
cmd=5
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 22 00 5C 97 8B A4 25 80 00 
00000010: 00 00 00 00 00 00 05 00 97 8B 
00000020: A4 25 80 00 01 01 77 15 01 5B 
00000030: 02 00 02 01 
<<<=================================>>>
MAX_pcktBuf is now 34 bytes
Local BT address: 00:02:5B:01:15:77
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 3F 00 41 77 15 01 5B 02 00 
00000010: FF FF FF FF FF FF 01 00 7E FF 
00000020: 03 60 65 09 A0 FF FF FF FF FF 
00000030: FF 00 00 7D 5D 00 2A BA E5 36 
00000040: 00 00 00 00 00 00 02 80 00 02 
00000050: 00 00 00 00 00 00 00 00 00 00 
00000060: 0A 65 7E 
63 Bytes sent
getPacket(1)
Received 18 bytes
MAX_CommBuf is now 88 bytes
Received 88 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 6A 00 14 97 8B A4 25 80 00 
00000010: 77 15 01 5B 02 00 01 00 7E FF 
00000020: 03 60 65 7D 33 90 7D 5D 00 2A 
00000030: BA E5 36 00 00 8A 00 16 31 F6 
00000040: 7D 5E 00 00 00 00 00 00 02 80 
00000050: 01 02 00 00 00 00 00 00 00 00 
00000060: 00 00 00 03 00 00 00 FF 00 00 
00000070: A0 6F 00 20 01 00 8A 00 16 31 
00000080: F6 7D 5E 00 00 0A 00 0C 00 00 
00000090: 00 00 00 00 00 03 00 00 00 01 
00000100: 01 00 00 87 E4 7E 
cmd=1
PacketLength=106
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E FF 03 60 65 13 90 7D 00 2A 
00000010: BA E5 36 00 00 8A 00 16 31 F6 
00000020: 7E 00 00 00 00 00 00 02 80 01 
00000030: 02 00 00 00 00 00 00 00 00 00 
00000040: 00 00 03 00 00 00 FF 00 00 A0 
00000050: 6F 00 20 01 00 8A 00 16 31 F6 
00000060: 7E 00 00 0A 00 0C 00 00 00 00 
00000070: 00 00 00 03 00 00 00 01 01 00 
00000080: 00 87 E4 7E 
<<<=================================>>>
MAX_pcktBuf is now 84 bytes
Serial Nr: 7EF63116 (2130063638)
logoffSMAInverter()
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 3B 00 45 77 15 01 5B 02 00 
00000010: FF FF FF FF FF FF 01 00 7E FF 
00000020: 03 60 65 08 A0 FF FF FF FF FF 
00000030: FF 00 03 7D 5D 00 2A BA E5 36 
00000040: 00 03 00 00 00 00 03 80 0E 01 
00000050: FD FF FF FF FF FF AE 98 7E 
59 Bytes sent
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 14 00 6A 77 15 01 5B 02 00 
00000010: 97 8B A4 25 80 00 03 00 05 00 
20 Bytes sent
getPacket(4)
Received 18 bytes
Received 6 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 18 00 66 97 8B A4 25 80 00 
00000010: 77 15 01 5B 02 00 04 00 05 00 
00000020: 00 00 AD 00 
cmd=4
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 18 00 66 97 8B A4 25 80 00 
00000010: 77 15 01 5B 02 00 04 00 05 00 
00000020: 00 00 AD 00 
<<<=================================>>>
BT Signal=67.8%
logonSMAInverter()
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 53 00 2D 77 15 01 5B 02 00 
00000010: FF FF FF FF FF FF 01 00 7E FF 
00000020: 03 60 65 0E A0 FF FF FF FF FF 
00000030: FF 00 01 7D 5D 00 2A BA E5 36 
00000040: 00 01 00 00 00 00 05 80 0C 04 
00000050: FD FF 07 00 00 00 84 03 00 00 
00000060: 5B A0 F3 63 00 00 00 00 B8 B8 
00000070: B8 B8 88 88 88 88 88 88 88 88 
00000080: A3 0D 7E 
83 Bytes sent
getPacket(1)
Received 18 bytes
Received 66 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 54 00 2A 97 8B A4 25 80 00 
00000010: 77 15 01 5B 02 00 01 00 7E FF 
00000020: 03 60 65 0E D0 7D 5D 00 2A BA 
00000030: E5 36 00 01 8A 00 16 31 F6 7D 
00000040: 5E 00 01 00 01 00 00 05 80 0D 
00000050: 04 FD FF 07 00 00 00 84 03 00 
00000060: 00 5B A0 F3 63 00 00 00 00 B8 
00000070: B8 B8 B8 88 88 88 88 88 88 88 
00000080: 88 48 58 7E 
cmd=1
PacketLength=84
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E FF 03 60 65 0E D0 7D 00 2A 
00000010: BA E5 36 00 01 8A 00 16 31 F6 
00000020: 7E 00 01 00 01 00 00 05 80 0D 
00000030: 04 FD FF 07 00 00 00 84 03 00 
00000040: 00 5B A0 F3 63 00 00 00 00 B8 
00000050: B8 B8 B8 88 88 88 88 88 88 88 
00000060: 88 48 58 7E 
<<<=================================>>>
Mon Feb 20 17:31:23 2023: CRITICAL: Logon failed. Check 'USER' Password
Mon Feb 20 17:31:23 2023: INFO: Done.

sbf-, 2023-02-21:
Are you sure no other devices/SW are connected to your inverter?
Running SunnyExplorer and SBFspot at the same time e.g.

914tom, 2023-02-21:
Thank you for your reply... 
At the time I establish the connection the Bluetooth light on the Sunny Boy is off. 
So yes, I'm pretty sure no other connection is established.

sbf-, 2023-02-27:
I double checked with my own inverter and I'm pretty sure your password is wrong. The returncode 0x0100 at position 23 is "wrong password"
```
--------: 00 01 02 03 04 05 06 07 08 09
00000020: 7E 00 01 00 01 00 00 05 80 0D
-------------------^^^^^
```

914tom, 2023-02-27:
Sorry, blame it on me, my Fault.
I misinterpreted Windows sunny Explorer. If I try to log in with a faulty password it rejected the login. 
If I log in with the standard 0000 it seems to log in, but it only provides Basi infos, Serial number, Machine State etc. but no further Data like Power output etc. It seems there is a Problem with the login like you mentioned. 
I contacted SMA, hope the Service can help me out. As soon as I log in the Bluelight came on and everything seems to be fine.

closed 2023-02-27 by sbf-