SBFspot/SBFspot #626: speedwire 0.0.0.0. Autodetect doesn't work any more after inverter update to 4.0.7.R

limwakra, 2023-03-22:
After an autoupdate from the inverter last night to firmware version 4.0.7.R. The autodetect ip doesn't seem to work anymore. 

Got the following after trying to fetch to data of the missing day:
`Connecting to Local Network...
Initialising...
SUSyID: 125 - SessionID: 974398691
Valid response from SMA device 192.168.1.127
Valid response from SMA device 192.168.1.126
Valid response from SMA device 0.0.10.128
Inverter replied: 192.168.1.127 -> 408:3009012627 ## ethernet connection
Inverter replied: 192.168.1.126 -> 408:3009012627 ## wireless connection
ERROR: Connection to inverter failed!
Is 192.168.1.126 a correct IP?
ERROR: Connection to inverter failed!
Is 0.0.10.128 a correct IP?
Wed Mar 22 19:32:37 2023: CRITICAL: Failed to initialise Speedwire connection.
Wed Mar 22 19:32:37 2023: INFO: Done.
`

sbf-, 2023-03-23:
I need a debuglog (-d5 -v5) to say something meaningful...
If you have fixed IP's for your inverters it's better to put these in the config. You'll notice faster connection speed while initializing.

sbf-, 2023-03-23:
> Attached the log you requested

Thanks, but I meant a log with IP=0.0.0.0 to see where it goes wrong ;-)

limwakra, 2023-03-23:
No problem. Got an error after running sbfspot config when changing speedwire to 0.0.0.0. Attached bot ssh logs.
[log_sbfspot.txt](https://github.com/SBFspot/SBFspot/files/11056609/log_sbfspot.txt)
[log after running sbfspot-config.txt](https://github.com/SBFspot/SBFspot/files/11056610/log.after.running.sbfspot-config.txt)

halfagascan, 2023-03-24:
I am also experiencing this issue.
SMA inverter  SUNNY BOY 5.0-US Firmware version: 3.0.27.R
I have a debian 11 server SBFspot running, and outputing csv and sqlite to 123solar, on occasion, from cli, fails with speedwire connect fails.
A new clean install on a PI 3 results in sporadic failures, a connection can be made once out of four or five tries, if it fails, I keep trying until I succeed
[1debug_sbfspot.txt](https://github.com/SBFspot/SBFspot/files/11063283/1debug_sbfspot.txt)
[debug_sbfspot.txt](https://github.com/SBFspot/SBFspot/files/11063285/debug_sbfspot.txt)
.
Attached are two log files, generated Fri Mar 24 08:12:31 2023, and the other Fri Mar 24 08:16:22 2023, the file 1debug_sbfspot is the failure, the other is successful, time difference of four minutes. 
New to outputting to mysql, there are failures to write to the db, reading CreateMySQLUser.sql, this line:
CREATE USER 'SBFspotUser'@'%' IDENTIFIED BY 'SBFspotPassword'
is that my user name on the Pi?
Thanks

halfagascan, 2023-03-25:
Still dark here, so I thought I would test.
Both the debian 11 and raspberry pi3 were able to connect, four out of four times, no errors, using
/usr/local/bin/sbfspot.3/SBFspot -v -finq
You may disregard the questions about writing to the mysqldb, I have it working as it should.
Thanks

sbf-, 2023-03-25:
@halfagascan This is another issue... TS has problems when IP in config is 0.0.0.0
Is it failing only during connection? Once it gets connected all data comes through?

halfagascan, 2023-03-25:
crap, failed to post.
on debian 11 2 tries 2 success /usr/local/bin/sbfspot.3/SBFspot -v
on pi3 3 tries 1 success /usr/local/bin/sbfspot.3/SBFspot -v
Sun is up and its generating
[sbfspot_debug.txt](https://github.com/SBFspot/SBFspot/files/11069267/sbfspot_debug.txt)
sbfspot_debug is from pi3

wimleers, 2023-04-13:
I got bitten by this firmware update too in my software, but in a very different way, because I'm listening to Speedwire multicasts — see https://github.com/wimleers/homebridge-sma-home-manager/issues/4#issuecomment-1468669369. I have no idea how to help with this though 😅

J-A-U, 2023-07-30:
Seems to me i run into the same problem. Aaaand: The title of the Issue is wrong...

Tripower X12, Firmware 03.02.08.R
[smalog.txt](https://github.com/SBFspot/SBFspot/files/12208838/smalog.txt)

Just like @limwakra the inverter got found, but the actual connection timed out.

sbf-, 2023-07-30:
@J-A-U Unfortunately the newer inverters like your Tripower X12 don't talk SMAData2+ and will not work with Sunnyexplorer nor SBFspot - See discussion #634

sbf-, 2025-03-10:
Unable to reproduce

closed 2025-03-10 by sbf-