SBFspot/SBFspot #630: Battery related values all zero (Hybrid Inverter)

sbf-, 2023-03-31:
### Discussed in https://github.com/SBFspot/SBFspot/discussions/628

<div type='discussions-op-text'>

<sup>Originally posted by **pickup7** March 30, 2023</sup>
Dear all, 
I am running SBFspot as a stable application since May 2022. Just upgraded to the latest version. What I never could achieve until now was to extract and MQTT transfer  the values for the Battery (BYD model 7,7 V) related values:
"**BatVol**": 0.000,"**BatAmp**": 0.000,"BatChaStt": 0.000
They unfortunalety remain zero although reasonable values are present and show up when I login to the Web based GUI of the SMA inverter. There it currently just reads:
- Aktueller Batterieladezustand  93 % 
- Batterietemperatur 17,0 °C 
- Batteriespannung 327,80 V  
- Batteriestrom  0,000 A  
- Betriebsstatus der Batterie Batterie Standby
A complete MQTT JSON-formatted message reads:
sbfspot {"Timestamp": "2023-03-30 14:34:02","SunRise": "2023-03-30 07:12:00","SunSet": "2023-03-30 19:58:00","InvTime": "2023-03-30 14:34:05","InvStatus": "Ok","InvTemperature": 0.000,"EToday": 5.862,"PACTot": 0.000,"MeteringWOut": 0.000,"MeteringWIn": 150.000,"MeteringWTot": 150.000,"BatVol": 0.000,"BatAmp": 0.000,"BatChaStt": 0.000}

Since at least the BaChaStt is of rather great importance to plan a my multi-level water heating in my scheduling software (depending on the Byttery Charge Status in percent) I would very much appreciate any hint or suggestion on why the values are not extracted / MQTT-forwarded to the data sink.

Thank you very much indeed.
Ulrich</div>

sbf-, 2023-03-31:
@pickup7
I made a fix to support hybrid inverters. Can you test it?
The master branch is updated to 3.9.7

pickup7, 2023-04-01:
G'day SBF,
after testing 3.9.7 this morning.
perfect job.. 
just curious: was it a matter of offsets or sthg different ?
MQTT: Publishing (sbfspot) "Timestamp": "2023-04-01 09:04:41","SunRise": "2023-04-01 07:08:00","SunSet": "2023-04-01 20:01:00","InvTime": "2023-04-01 09:04:43","InvStatus": "Ok","InvTemperature": 0.000,"EToday": 1.491,"PACTot": 180.000,"MeteringWOut": 2.000,"MeteringWIn": 0.000,"MeteringWTot": -2.000,"**BatVol": 312.700,"BatAmp": 0.000,"BatChaStt": 16.000**
Thank you very much indeed.
Ulrich

sbf-, 2023-04-01:
> just curious: was it a matter of offsets or sthg different ?

SBFspot did not recognise the (relatively new) type of hybrid inverters, so battery info was ignored.
Thank you for testing!

sbf-, 2023-04-01:
Fixed in version 3.9.7

closed 2023-04-01 by sbf-