SBFspot/SBFspot #633: same issue on 2 different raspberries: CRITICAL: bthConnect() returned -1

broke23, 2023-04-11:
I had to reinstall my raspberry due to issues with pihole...

Reinstall all fine, installing SBF again, put in PVoutput id's and such, found bluetooth inverter, no problem

put in userPW (which is not the default ofcourse) and I get this error:

CRITICAL: bthConnect() returned -1

Multiple times.

Ok, other raspberry lying here, 3b+ instead of 3b, installed, updated, same thing, even when I didn't fill in the Pvoutput stuff, so it's something with the userPW...

though, if i connect with laptop and sma sunny explorer, I test the same password there, it works... It's simple password: Capital letter, small letter and numbers, no special char ...

I don't get it, why it's working for sunny portal, but not with SBF 3.9.7 nor with 3.9.6 (new config created)

I know for sure this PW is legit, cause I once corrupted the passwords and couldn't reset so I had to pay to get this special key to reset again... since I put those pw's in database not to forget

sbf-, 2023-04-11:
> CRITICAL: bthConnect() returned -1

This error in not password related...
Built-in bluetooth or external dongle?
What is the distance inverter/Rpi?

Can you post a debug log -d5 -v5

SE is logged out? It can give issues if SE is connected with inverter.

broke23, 2023-04-11:
What the.... Now it works all of a sudden?

Could it mean the connection was really busy without led lit? Waited a bit afterwards and on new pi the same?

broke23, 2023-04-11:
Thx anyway SBF

closed 2023-04-11 by broke23