SBFspot/SBFspot #635: The mystery of the sole 2.5 min interval datapoint … and it's corrupt.

wimleers, 2023-04-13:
For many months now, `sbfspot` has worked flawlessly and uploaded my data points to pvoutput.org without fail. But today, it failed.

Turns out that it's because it's been trying to upload a faulty value since yesterday:
```
[00:00:30.902] ERROR: Uploading 34 datapoints, starting with 20230412,17:35,2022612,480,,,0,243.91 Bad request 400: Power value [2147483647] too high for system size [6000]
```
… and it says it's for 2023-04-12 (yesterday)'s 17:35 value.

So I go and look for out of the ordinary values:
```
sqlite> select TimeStamp,TotalYield,Power,PVoutput from daydata ORDER by Power DESC LIMIT 5;
TimeStamp|TotalYield|Power|PVoutput
1681313850|1683627183633|8299927991|
1656067500|418354|4992|
1656068700|419658|4992|
1681381200|2030434|4980|
1681382400|2031786|4980|
```

… and sure enough, there it is. So let's see what the surrounding values are:
```
sqlite> select TimeStamp,TotalYield,Power,PVoutput from daydata WHERE TimeStamp > 1681313850 - 1000 AND TimeStamp < 1681313850 + 1000;
TimeStamp|TotalYield|Power|PVoutput
1681313100|2022521|420|1
1681313400|2022572|612|1
1681313700|2022612|480|
1681313850|1683627183633|8299927991|
1681314000|2022643|372|
1681314300|2022671|336|
1681314600|2022700|348|
```
Well … that all looks fine. So I went to my inverter's web UI and downloaded the 5 min values to figure out what the value SHOULD be:
```
12.04.2023 17:30:00,2022572,0.985,OvExt,NaN,NoneMsg,49.98,610,610,NaN,NaN,243.47,NaN,NaN,NaN,NaN,NaN,2.687,NaN,NaN,Ok,346,370,271.58,274.50,1.274,1.347,1529505,1529505,1185401,1185401,338,2887204,0.009
12.04.2023 17:35:00,2022612,0.974,OvExt,NaN,NoneMsg,49.99,488,488,NaN,NaN,243.80,NaN,NaN,NaN,NaN,NaN,2.202,NaN,NaN,Ok,281,305,270.06,272.29,1.042,1.120,1529505,1529505,1185421,1185421,338,2887204,0.009
12.04.2023 17:40:00,2022643,0.950,OvExt,NaN,NoneMsg,49.97,368,368,NaN,NaN,244.09,NaN,NaN,NaN,NaN,NaN,1.739,NaN,NaN,Ok,222,242,266.36,270.49,0.834,0.896,1529505,1529505,1185452,1185452,338,2887204,0.009
```
Well shit. All those surrounding values _are_ already there?

And that's when I noticed: the timestamps are 300 s (5 mins) apart: `1681313400`, `1681313700`, `1681314000` … but between them: `1681313850` — only 150 s apart!

The `pvoutput.org` upload functionality seems to coalesce this to the `17:35` values, which seems like a fine choice for robustness. But the question remains: **how did we suddenly log values every 2.5 mins, but only a single time?! 😳**

A simple `DELETE FROM daydata WHERE TimeStamp = 1681313850;` later it's all fine (and uploading again), but … how could this possibly happen 😮

silliewimons, 2023-04-14:
dear wimleers,  can you extract from your inverter  (webUI, -> events  ->  export)  the list of events for the month april 

kind regard  sillie

sbf-, 2023-04-14:
Thanks for reporting this issue and for the detailed investigation!

> And that's when I noticed: the timestamps are 300 s (5 mins) apart: `1681313400`, `1681313700`, `1681314000` … but between them: `1681313850` — only 150 s apart!

Never seen this before... Web UI ignores this entry, but SBFspot doesn't. I'm confident the inverter returned this faulty record, and it is possibly still in the inverter's DB.

The DB/CSV won't help to debug, only the raw inverter data.

Can you execute the following command and post the result?
```
SBFspot -v5 -d5 -finq -nocsv -nosql -startdate:20230412 -am1 -ad1 -ae1
```

`-nosql` will prevent the storage of the faulty record - if it's still there - and blocking the PVO upload once again.

wimleers, 2023-04-14:
So to be clear: that command would not cause *any* writes to either my SQLite DB not my CSV files, and just generate `stdout` output?

wimleers, 2023-05-06:
Same thing again yesterday:
```
[17:37:31.637] INFO: Uploading datapoint: 20230505,17:35,2443357,696,,,0,242.91 => OK (200)
[17:42:31.232] INFO: Uploading datapoint: 20230505,17:40,2443414,684,,,0,241.93 => OK (200)
[17:47:30.867] ERROR: Uploading 2 datapoints, starting with 20230505,06:35,2428823,0,,,0,244.34 Bad request 400: Power value [947530749] too high for system size [6000]
[17:48:31.614] ERROR: Uploading 2 datapoints, starting with 20230505,06:35,2428823,0,,,0,244.34 Bad request 400: Power value [947530749] too high for system size [6000]
[17:49:31.389] ERROR: Uploading 2 datapoints, starting with 20230505,06:35,2428823,0,,,0,244.34 Bad request 400: Power value [947530749] too high for system size [6000]
[17:50:31.155] ERROR: Uploading 2 datapoints, starting with 20230505,06:35,2428823,0,,,0,244.34 Bad request 400: Power value [947530749] too high for system size [6000]
```

👇 
```
sqlite> select TimeStamp,datetime(TimeStamp, 'unixepoch'),TotalYield,Power,PVoutput from daydata ORDER by Power DESC LIMIT 5;
1683301650|2023-05-05 15:47:30|1799591300625|947530749|
1656067500|2022-06-24 10:45:00|418354|4992|
1656068700|2022-06-24 11:05:00|419658|4992|
1681381200|2023-04-13 10:20:00|2030434|4980|1
1681382400|2023-04-13 10:40:00|2031786|4980|1
```
Sure enough, same thing:
```
sqlite> select TimeStamp,datetime(TimeStamp, 'unixepoch'),TotalYield,Power,PVoutput from daydata WHERE TimeStamp > 1683301650 - 1000 AND TimeStamp < 168330
1683300900|2023-05-05 15:35:00|2443357|696|1
1683301200|2023-05-05 15:40:00|2443414|684|1
1683301500|2023-05-05 15:45:00|2443475|732|
1683301650|2023-05-05 15:47:30|1799591300625|947530749|
1683301800|2023-05-05 15:50:00|2443538|756|
1683302100|2023-05-05 15:55:00|2443599|732|
1683302400|2023-05-05 16:00:00|2443657|696|
```
```
SBFspot -v5 -d5 -finq -nocsv -nosql -startdate:20230505 -am1 -ad1 -ae1
```
generates the attached log output.
[docker-sbfspot.log](https://github.com/SBFspot/SBFspot/files/11412116/docker-sbfspot.log)

Most fascinating subset:
```
fspot-1  | 05/05/2023 14:35:00 : 2437.622kWh - 3996.000W

sbfspot-sbfspot-1  | 05/05/2023 14:40:00 : 2437.947kWh - 3900.000W

sbfspot-sbfspot-1  | 05/05/2023 14:45:00 : 2438.211kWh - 3168.000W

sbfspot-sbfspot-1  | 05/05/2023 14:50:00 : 2438.485kWh - 3288.000W

sbfspot-sbfspot-1  | 05/05/2023 14:55:00 : 2438.710kWh - 2700.000W

sbfspot-sbfspot-1  | 05/05/2023 15:00:00 : 2439.012kWh - 3624.000W

sbfspot-sbfspot-1  | 05/05/2023 15:05:00 : 2439.203kWh - 2292.000W

sbfspot-sbfspot-1  | 05/05/2023 15:10:00 : 2439.325kWh - 1464.000W

sbfspot-sbfspot-1  | 05/05/2023 15:15:00 : 2439.481kWh - 1872.000W

sbfspot-sbfspot-1  | 05/05/2023 15:20:00 : 2439.651kWh - 2040.000W

sbfspot-sbfspot-1  | 05/05/2023 15:25:00 : 2439.872kWh - 2652.000W

sbfspot-sbfspot-1  | 05/05/2023 15:30:00 : 2440.090kWh - 2616.000W

sbfspot-sbfspot-1  | 05/05/2023 15:35:00 : 2440.292kWh - 2424.000W

sbfspot-sbfspot-1  | 05/05/2023 15:40:00 : 2440.457kWh - 1980.000W

sbfspot-sbfspot-1  | 05/05/2023 15:45:00 : 2440.628kWh - 2052.000W

sbfspot-sbfspot-1  | 05/05/2023 15:50:00 : 2440.836kWh - 2496.000W

[... 69 lines cut ...]

sbfspot-sbfspot-1  | 05/05/2023 18:50:00 : 2444.211kWh - 624.000W

sbfspot-sbfspot-1  | 05/05/2023 18:55:00 : 2444.253kWh - 504.000W

sbfspot-sbfspot-1  | 05/05/2023 19:00:00 : 2444.285kWh - 384.000W

sbfspot-sbfspot-1  | 05/05/2023 19:05:00 : 2444.313kWh - 336.000W

sbfspot-sbfspot-1  | 05/05/2023 19:10:00 : 2444.336kWh - 276.000W

sbfspot-sbfspot-1  | 05/05/2023 19:15:00 : 2444.357kWh - 252.000W

sbfspot-sbfspot-1  | 05/05/2023 19:20:00 : 2444.375kWh - 216.000W

sbfspot-sbfspot-1  | 05/05/2023 19:25:00 : 2444.391kWh - 192.000W

sbfspot-sbfspot-1  | 05/05/2023 19:30:00 : 2444.407kWh - 192.000W

sbfspot-sbfspot-1  | 05/05/2023 19:35:00 : 2444.421kWh - 168.000W
```

So … packet corruption? 🤔 

(Sadly, I can't seem to get the _full_ log output.)

sbf-, 2023-05-06:
It looks like the erroneous value in the inverter's memory is overwritten after a short time.
The best solution for now is [this workaround](https://github.com/SBFspot/SBFspot/issues/137#issuecomment-341902473)

wimleers, 2023-05-08:
I wonder if it'd be simpler to restrict `vwDayData` to omit values that are divisible by `150`? Or to put it differently: only use values divisible by `300`?

So:
```
WHERE Timestamp % 300 = 0
```
?

sbf-, 2023-05-08:
> I wonder if it'd be simpler to restrict `vwDayData` to omit values that are divisible by `150`? Or to put it differently: only use values divisible by `300`?

That will indeed fix your problem, but all the cases we are aware of have (or had) too high power values on 300 sec intervals. Your case is really an exception.

I'll leave the ticket open and will try to fix it later

wimleers, 2023-05-08:
Is there any other digging/diagnostics you would like me to do?

(I hate it when there's inexplicable edge cases, so anything I can do to reduce edge cases _you_ need to handle I'd be happy to assist you with! 🤓)

silliewimons, 2023-05-10:
dear wim-leers,  you apply sbfspot via napla,  which is a method to be "machine" independent,
       we have seen (fm github) that, for the time being, you can not migrate to v397 (actual best sbfspot-version)
?so i need first to understand what options you can execute (and what not),    ?C?<-please comment where you can give info
eg  your limit in providing a  complete log-file might be part of 'napla' / docker environment   ?C?
eg  from your info we do assume that  sbfspot is repeatedly executed  (each ~300sec.)  and that is part of  .. . 
eg  from your info - in parallel with database output   THERE ARE csv-file produced (but you did not indicate which csv-type
since  there is no complete log-file  "from start-to-bottom"  we can not see under which options sbfspot is executed

in doing the digging  remember that  sbfspot has two distinct parts  
    -A- interogate inverter ad write the results to database and/or csv and/or . . . 
    -B- starting from the database - upload data to PVoutput or . . .
within your setup one must understand "for each of these two parts" -what is executed correctly    -when/where are issues

from you description there is suddenly a mismatch  that is visible in the database   
         today we can not identify  whether  the mismatch  is  introduced by  -A-  or later on

how to progress digging:   if csv-files are available -> this can be used as second-opinion  on -A-part 'interogate-inverter     
??about the csv files (for your purpose there are two csv-types useful) 
      daily (i use the nickname 5min-in-day) which keeps records for each 5min interval (a record has timestamp, Etotal, . . . .
      spot-table - each time that sbfspot is executed a record with values for Udc, Idc, Pdc, Uac . . .  is created
      aside of having these 'RealTime-records'   "spot table"  acts as histogram on execution of sbfspot  (if Spot=RealTime is on)
IMPORTANT - all configuration-files / all sbfspot.executions  must  send  the csv files to the same directory  
           
digging on the -B-part is the next step    but when   mismatches are created  during  -A- part ?? 
kr  sillie     by the way  due to fact that we do not have a log-file, can you inform about type of inverter  and . . .

wimleers, 2023-05-11:
The problem is definitely in

> -A- interogate inverter ad write the results to database and/or csv and/or . . .

Because it's stored incorrectly in the SQLite database `sbfspot` generates.

I've already provided relevant CSV files and @sbf- has indicated he needs no more. 

> eg from your info we do assume that sbfspot is repeatedly executed (each ~300sec.) and that is part of .. .

No, it runs continuously, and decided autonomously when to upload to PVOutput.

P.S.: I'm having a very hard time to understand message due to the absence of capitalization and punctuation 😅

wimleers, 2023-06-12:
One more time:
```
sqlite> .headers on
sqlite> select TimeStamp,datetime(TimeStamp, 'unixepoch'),TotalYield,Power,PVoutput from daydata ORDER by Power DESC LIMIT 5;
TimeStamp|datetime(TimeStamp, 'unixepoch')|TotalYield|Power|PVoutput
1686404850|2023-06-10 13:47:30|1971389992465|5243961142|
1656067500|2022-06-24 10:45:00|418354|4992|
1656068700|2022-06-24 11:05:00|419658|4992|
1681381200|2023-04-13 10:20:00|2030434|4980|1
1681382400|2023-04-13 10:40:00|2031786|4980|1
```
Note that once again this is a 2.5 minute interval.

wimleers, 2023-09-06:
In hopes there is a pattern to this … reporting that it happened once again:
```
sqlite> select TimeStamp,TotalYield,Power,PVoutput from daydata ORDER by Power DESC LIMIT 5;
1690616550|2203318226449|38567093|
1656067500|418354|4992|
1656068700|419658|4992|
1681381200|2030434|4980|1
1681382400|2031786|4980|1
```

wimleers, 2023-12-18:
Hasn't happened in months now. It appears to be somehow correlated with days of high production: I have a 5 kW inverter, and you can see in the examples above it's 20 Watt or less shy of its max capacity. 🤔

closed 2024-01-23 by sbf-

wimleers, 2024-07-08:
It happened again.

```
sqlite> select TimeStamp,datetime(TimeStamp, 'unixepoch'),TotalYield,Power,PVoutput from daydata ORDER by Power DESC LIMIT 5;
TimeStamp|datetime(TimeStamp, 'unixepoch')|TotalYield|Power|PVoutput
1716060450|2024-05-18 19:27:30|3607772532241|6913584250|
1716223050|2024-05-20 16:37:30|3616362466833|5362023152|
1656067500|2022-06-24 10:45:00|418354|4992|
1656068700|2022-06-24 11:05:00|419658|4992|
1713436500|2024-04-18 10:35:00|6349589|4992|1
```

sbf-, 2024-07-09:
What version are you running? This issue has been fixed in [V3.9.8](https://github.com/SBFspot/SBFspot/wiki/Modification-History#398-23-jan-2024-bugfix-release)

commit d6c259ce (2024-01-23): Refactored ArchiveDayData()
Fixed #578 Corrupted/Future data from Inverter
Fixed #635 The mystery of the sole 2.5 min interval datapoint... and it's corrupt