SBFspot/SBFspot #637: ERROR: addBatchStatus() returned Unsupported protocol in Windows install

marnixt, 2023-04-22:
Hi

when uploading using the windows upload service I get the error below:

[20:18:39.574] INFO: Stopping Service...
[20:18:40.895] INFO: Starting ServiceWorkerThread()
[20:18:40.904] DEBUG: Retrieving new datapoints from DB...
[20:18:40.912] ERROR: addBatchStatus() returned Unsupported protocol

what does addBatchStatus mean?

ps. I have been running SBFspot on RPI for years but ran into some issue there and wanted to migrate to windows anyway

sbf-, 2023-04-23:
> what does addBatchStatus mean?

[Add Batch Status Service](https://pvoutput.org/help/api_specification.html#add-batch-status-service)

marnixt, 2023-04-25:
thanks but unfortunately it doesn't mention error unsupported protocol. Does it have something to do with SSL?

sbf-, 2023-04-29:
> Does it have something to do with SSL?

I have no idea... never seen this before

marnixt, 2023-05-18:
I have tried 3.9.6 and it uploads fine. So the issue indeed seems to be with 3.9.7

stipman, 2023-07-03:
Confirmed 3.9.6 does not throw any errors.
3.9.7 displays the following when LogLevel set to debug

[15:52:18.625] DEBUG: Retrieving new datapoints from DB...
[15:52:18.627] ERROR: addBatchStatus() returned Unsupported protocol

sbf-, 2024-01-23:
Fixed in V3.9.8

closed 2024-01-23 by sbf-