SBFspot/SBFspot #642: date multi-inverter

jeedee5, 2023-05-23:
Hi, 

I have 2 SB4000-TL20 inverters connected, but 1 of them is having issues with its date setting. I can't even set the cortrect date anymore with sunny explorer. It currently has a date of 26/01/1970. The date still progresses every day, but I cannot set it to what it should be. 

Following is reported by sbfspot:

SUSyID: 78 - SN: 2100280013
Current Inverter Time: 23/05/2023 11:29:01
Inverter Wake-Up Time: 23/05/2023 06:10:00
Inverter Sleep Time  : 23/05/2023 11:29:04
SUSyID: 78 - SN: 2100278992
Current Inverter Time: 26/01/1970 01:40:52
Inverter Wake-Up Time: 25/01/1970 20:21:57
Inverter Sleep Time  : 26/01/1970 01:40:57

Following is reported by sbfspot -settime:

Local Host Time: 23/05/2023 11:29:52
Plant Time     : 23/05/2023 11:29:52 (0 sec)
TZ offset      : 3600 sec - DST: On
Last Time Set  : 23/05/2023 10:40:12
Tue May 23 11:29:52 2023: INFO: Done.

In the logs, you can see it appear as:

23/05/2023 10:40:12 Tijd wordt ingesteld / nieuwe tijd
26/01/1970 00:52:07 Tijd wordt ingesteld / oude tijd
23/05/2023 10:20:03 Tijd wordt ingesteld / nieuwe tijd
26/01/1970 00:31:59 Tijd wordt ingesteld / oude tijd
23/05/2023 10:19:26 Tijd wordt ingesteld / nieuwe tijd
26/01/1970 00:31:22 Tijd wordt ingesteld / oude tijd
23/05/2023 09:04:49 Tijd wordt ingesteld / nieuwe tijd
25/01/1970 23:16:47 Tijd wordt ingesteld / oude tijd

While it is still doing it's job, I don't get any uploads anymore to pvoutput. It still enters data in my DB in the table SpotData, but it doesn't add any new entries for the specific inverter in Daydata (where the pvoutput variable is)... 

Can I change the code somewhere so that the second inverter (with the date issue), takes the "plant time" instead of its own time?

sbf-, 2023-05-23:
> it doesn't add any new entries for the specific inverter in Daydata

That's the biggest issue... for PVoutput the Daydata table has to have the correct date/time
SBFspot is unable to get the data. It requests the records for today (with -ad1) but there aren't any.
Even `-ad300` (get archive data of the past 300 days) will not give you any data

Does Sunnyexplorer shows any data?

silliewimons, 2023-05-24:
seemingly your inverter-2100278992  did get a clear of its date-time settings   end april
 
since your issue started end april (about 26days) - do you have any indication what happened  end april ?? 
   ??password-change / grid-breakdown / . . . 
- since this is "multi-inverter" you will need to make  "temporary" for TESTING purposes two EXTRA setup-s  one per inverter
       I mean   using   other 'directory' for each inverter
- in these setup's you must set MIS to OFF
       control that you use the address for inverter 2100278992 
       set  adjust-time  to  OFF  (test-purpose)
       if needed inspect the config-file such that csv-output is set to recuperate daily and events file 
- EXECUTE  sbfspot  ONLY FOR   your inverter  2100278992   (use the seperated setup)        
       apply option -startdate:%sta-dat%   where  %sta-dat% is replaced by 19700226 
       apply options -ad60 -am2 -ae3  -nosql -d5 -v5  1>log-file.wrong-date.txt   
       COMBINED ->   sbfspot.exe   -startdate:19700226 -ad60 -am2 -ae3 -nosql -d5 -v5    
  this should deliver in "csv" format   but also    in the log-file  the events  and  day-information 
       inspect the event-file  to understand what happened  26days ago,

you might need to keep the daily files to reconstruct the database,

the reason that you can not update the date is because MultiInverter  your two-inverters  do  react as  one  single device,
and the Date-Time difference  is  out of the range that is  handled by SunnyExplorer 

please inform on results          kr  sillie

jeedee5, 2023-05-25:
I ran the commands, but there is nothing pointing out to what happened... I have set the inverter to netid 1, so it acts as a standalone installation, but even than, I cannot set the date correctly.
I am going to do a firmware update on that inverter. It is currently running 03.20.01.R, but version 3.20.05 is the latest. Nothing has changed that can help me according to the change log, but the firmware will get gefreshed (removing any current firmware corruptions)...

silliewimons, 2023-05-25:
can you provide:    (you can include them in he issues-chapter, or private > to sbf  and/or "willie.simons@belgacom.net"
 - log-file (txt),  events file (csv),  month-file (csv),   <= result after execution of the commands,
 - a copy of your configuration file  (  'SBFspot.cfg'   see directory with executable)
    if you have more then one config-file  please a copy of each  (identify for which inverter) 

NOTE: when you have more then one inverter,  then SMA might apply concept "Plant", Plant is 'group of inverters'
    SMA's  'Plant concept'  has several 'hidden' mechanisms, 
          for example - inverters with SAME password are considered/treated as member of a "Plant",
for that reason - my  ??question   are  "user"  password's  the same for both inverters  ?
    SunnyExplorer does show 'Plant'  and  relation/dependency between inverters,  with different "icoon"
    please consult the documentation from SMA about this topic,                    kr  sillie