SBFspot/SBFspot #646: SBF

olmo1501, 2023-06-01:
Since switching my Inverter to a Sunny Tripower Smart Energy, SBFspot stopped creating the "Spot-CSV":

`root@monitoring:/home/pi/smadata/2023# ls -la *20230601*
-rw-r--r-- 1 root root  8234  1. Jun 19:29 'Am Spielplatz 9-20230601.csv'
-rw-r--r-- 1 root root 52874  1. Jun 19:29 'Am Spielplatz 9-Battery-20230601.csv'`

Before the switch:
`root@monitoring:/home/pi/smadata/2023# ls -la *20230527*
-rw-r--r-- 1 root root  3318 29. Mai 10:29 'Am Spielplatz 9-20230527.csv'
-rw-r--r-- 1 root root 34887 27. Mai 14:24 'Am Spielplatz 9-Battery-20230527.csv'
-rw-r--r-- 1 root root 44502 27. Mai 14:24 'Am Spielplatz 9-Spot-20230527.csv'`

SBFspotUploadDaemon stopped uploading data also. No hint why. Even if I start it with debug=1 it just logs something like "Checking for new data in the database". Can't remember the correct wording at the momen.

After reading through a lot of older issues I assume this is a "time-issue". My inverter doesn't report any "plant-time":
`SBFspot V3.9.7
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -v2 -finq -nocsv -nosql
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Thu Jun  1 19:38:01 2023: INFO: Starting...
sunrise: 05:15
sunset : 21:23
Connecting to Local Network...
Initialising...
SUSyID: 125 - SessionID: 949847827
Device IP address: 192.168.241.20 from config
Inverter replied: 192.168.241.20 -> 440:3015604826
Logon OK
SUSyID: 440 - SN: 3015604826
Device Name:      STP6.0-3SE-40 826
Device Class:     Hybrid-Wechselrichter
Device Type:      SUNNY TRIPOWER 6.0 SE
Software Version: 03.02.22.R
SUSyID: 440 - SN: 3015604826
Batt. Charging Status: 100%
SUSyID: 440 - SN: 3015604826
Batt. Temperature: 23.0°C
Batt. Voltage    : 321.00V
Batt. Current    : 0.000A
SUSyID: 440 - SN: 3015604826
Grid Power Out : 959W
Grid Power In  : 0W
SUSyID: 440 - SN: 3015604826
Device Status:      Ok
Packet status: 21
SUSyID: 440 - SN: 3015604826
Device Temperature: Information liegt nicht vor
SUSyID: 440 - SN: 3015604826
Energy Production:
        EToday: 42.213kWh
        ETotal: 230.619kWh
        Operation Time: 0.00h
        Feed-In Time  : 0.00h
00464B09 28 Thu Jun  1 19:38:01 2023 'Netzspannung Phase L1 gegen L2' 39680 39680 39680 39680
00464C09 28 Thu Jun  1 19:38:01 2023 'Netzspannung Phase L2 gegen L3' 39680 39680 39680 39680
00464D09 28 Thu Jun  1 19:38:01 2023 'Netzspannung Phase L3 gegen L1' 39610 39610 39610 39610
00464E09 28 Thu Jun  1 19:38:01 2023 '' 100 100 100 100
SUSyID: 440 - SN: 3015604826
DC Spot Data:
        MPPT 1 Pdc:   0.149kW - Udc: 462.80V - Idc:  0.321A
        MPPT 2 Pdc:   0.994kW - Udc: 294.90V - Idc:  3.370A
        Calculated Total Pdc:   1.143kW
AC Spot Data:
        Phase 1 Pac :   0.349kW - Uac: 229.10V - Iac:  1.500A
        Phase 2 Pac :   0.349kW - Uac: 229.10V - Iac:  1.600A
        Phase 3 Pac :   0.346kW - Uac: 228.70V - Iac:  1.500A
        Total Pac   :   1.040kW - Calculated Pac:   1.044kW
        Efficiency  :   91.34%
SUSyID: 440 - SN: 3015604826
Grid Freq. : 50.00Hz
********************
* ArchiveDayData() *
********************
startTime: 01/06/2023 00:00:00
**********************
* ArchiveMonthData() *
**********************
startTime: 01/06/2023 13:00:00
**********************
* ArchiveMonthData() *
**********************
startTime: 01/05/2023 12:00:00
Reading events: 2023-Jun-01
Thu Jun  1 19:38:02 2023: INFO: Done.
`
Any ideas what is going wrong?

Kind regards,

olmo1501, 2023-06-01:
Sorry, I somehow screwed up the Code-Tags. After some fiddling around with it and making the formatting even worse, I decided to leave it as it is.

sbf-, 2023-06-01:
> SBFspot stopped creating the "Spot-CSV"
> SBFspotUploadDaemon stopped uploading data also. No hint why

You disable csv and sql export

```
Commandline Args: -v2 -finq -nocsv -nosql
```

olmo1501, 2023-06-01:
No I didn't. It's just a bad example. I used these Commandline Arguments to not write "Test-Data" to the CSV or mySQL.

My Cronjob just calls

`/usr/local/bin/sbfspot.3/SBFspot -v2 -finq`

These CSV-Files are being written:
`root@monitoring:/home/pi/smadata/2023# ls -la *20230601*
-rw-r--r-- 1 root root  8234  1. Jun 19:29 'Am Spielplatz 9-20230601.csv'
-rw-r--r-- 1 root root 52874  1. Jun 19:29 'Am Spielplatz 9-Battery-20230601.csv'`

But there should also be something like `'Am Spielplatz 9-Spot-20230601.csv'` which is not being written.

Data is being written to the database, timestamps seem to be correct:
`MariaDB [SBFspot]> select * from DayData;
+------------+------------+------------+-------+----------+
| TimeStamp  | Serial     | TotalYield | Power | PVoutput |
+------------+------------+------------+-------+----------+

<Cut>

| 1685649600 | 3015604826 |     231614 |   144 |     NULL |
`
SBFspotUploadDaemon isn't uploading.

Sorry for not being more precise.

sbf-, 2023-06-01:
Since you changed inverter, S/N changed too. Did you update the daemon config file?
Take a look at [No uploads to PVOutput.org](https://github.com/SBFspot/SBFspot/wiki/No-uploads-to-PVOutput.org)

olmo1501, 2023-06-02:
I updated the config file. Data gets retrieved fine, it's being written to file and to the database.

What is not working anymore is the creating of the "Spot-CSV" and uploading the data to PVoutput. I guess these two issues are somehow related.

So let me try to ask differently as I am no good in reading source code: What is necessary for SBFspotUploadDaemon to detect new data from the database? Currently it only logs every minute:

`[11:04:30.665] DEBUG: Retrieving new datapoints from DB...`

Querying the table "DayData" shows data with the correct timestamp. Somewhere there must be something missing / I must be doing something wrong.

sbf-, 2023-06-03:
For battery or hybrid inverters, the spot-csv is replaced by battery-csv, so your data is in `Am Spielplatz 9-Battery-20230601.csv`
For PVOutput I have to dive into the code...

sbf-, 2023-06-03:
What is the output of `select * from vwinverters;` ?

olmo1501, 2023-06-03:
`+------------+-------------------+-----------------------+------------+---------------------+----------+--------+--------+---------------+------------+--------+-----------+-------------+
| Serial     | Name              | Type                  | SW_Version | TimeStamp           | TotalPac | EToday | ETotal | OperatingTime | FeedInTime | Status | GridRelay | Temperature |
+------------+-------------------+-----------------------+------------+---------------------+----------+--------+--------+---------------+------------+--------+-----------+-------------+
| <myserial> | STP6.0-3SE-40 826 | SUNNY TRIPOWER 6.0 SE | 03.02.22.R | 2023-06-03 19:24:02 |     1150 |  44940 | 319765 |             0 |          0 | OK     | ?         |   -21474800 |
+------------+-------------------+-----------------------+------------+---------------------+----------+--------+--------+---------------+------------+--------+-----------+-------------+
`

sbf-, 2023-06-04:
And this (MySQL/MariaDB):
``` SQL
SELECT DATE_FORMAT(Timestamp,'%Y%m%d,%H:%i') DTTM,V1,V2,V3,V4,V5,V6,V7,V8,V9,V10,V11,V12
FROM vwPvoData
WHERE TimeStamp>NOW()-INTERVAL 14 DAY
    AND PVoutput IS NULL
ORDER BY TimeStamp
LIMIT 5;

```

olmo1501, 2023-06-04:
```
+----------------+--------+------+------+------+------+------+------+------+------+------+------+------+
| DTTM           | V1     | V2   | V3   | V4   | V5   | V6   | V7   | V8   | V9   | V10  | V11  | V12  |
+----------------+--------+------+------+------+------+------+------+------+------+------+------+------+
| 20230604,00:00 | 321324 |  108 | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL |
| 20230604,00:05 | 321334 |  120 | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL |
| 20230604,00:10 | 321340 |   72 | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL |
| 20230604,00:15 | 321349 |  108 | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL |
| 20230604,00:20 | 321367 |  216 | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL | NULL |
+----------------+--------+------+------+------+------+------+------+------+------+------+------+------+
```

closed 2023-06-05 by olmo1501

sbf-, 2023-06-06:
This proves some more logging can be important ;-)