SBFspot/SBFspot #647: MQTT setup loses login name & password

hneel, 2023-06-05:
I run SBFSPOT on a raspberry pi. I use it to get data from the PV and upload that to pvout.org. That part works fine.
I also use it to get data directly to Home Assistant with mqtt. This has worked well in the past, but now I haven't seen any data for a couple of days.

When I ran the config on the rapsberry pi I noticed that the mqtt user and pw fields were empty. I hade seen this happen before in the past. Then I could enter them and save them and it would work again. This time however it is again empty when I run config again.

(The command I use for running config is:
curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash )

sbf-, 2023-06-06:
You can add username and password to the MQTT_PublisherArgs option in SBFspot.cfg

```
MQTT_PublisherArgs=-u username -P password -h {host} -t {topic} -m "{{message}}"
```

See #380

hneel, 2023-06-06:
Ah, yes. That sentence was really messed up in my config. I fixed it and it works now! 
Thanks!

closed 2023-06-07 by sbf-