SBFspot/SBFspot #649: SBFspot upload values wrong. MQTT right.

Gr1pen, 2023-06-09:
My SBFSpot has been running for year without any issues. Nsince a while back the upload failes because the values are too high.
I cleaned out these values from the database but still all new values are way to high. I also recreated the database completly.

My workaround is to send it to Node Red via MQTT and let Node Red do the upload. This works great and the values are correct.

How can this be?
Also I did not find any option in SBFspot to turn off the pvoutput uploads?

sbf-, 2023-06-10:
> My workaround is to send it to Node Red via MQTT and let Node Red do the upload. This works great and the values are correct

Correct for energy, but not for power values I assume?

> Also I did not find any option in SBFspot to turn off the pvoutput uploads?

```
sudo systemctl stop SBFspotUpload.service
sudo systemctl disable SBFspotUpload.service
```

closed 2023-06-10 by sbf-

Gr1pen, 2023-06-10:
Thanks.

Yes the total daily energy correct but the power values are not. Any clue why? This has worked so fine for years until now.