SBFspot/SBFspot #659: Invalid entry in table MonthData at month end

NelliAtHome, 2023-08-01:
I execute SBFSpot daily at 5:55 AM to write the daily yield in a sql database (mariadb). The command is

`/usr/local/bin/sbfspot.3/SBFspot -v -sp0 -ad0 -am1 -ae1 -finq >>$log`

When running at 01.08.23 5:55 AM i found the following last entries (order by TimeStamp) in table monthdata:

```
# TimeStamp, Serial, TotalYield, DayYield
'1690675200', '3011200342', '1390590', '8354'
'1690761600', '3011200342', '1390590', '8354'
'1690848000', '3011200342', '1396041', '5451'
```

or in view vwMonthData:

```
# TimeStamp, Name, Type, Serial, TotalYield, DayYield
'2023-07-30 00:00:00', 'SB4.0-1AV-41 342', 'Sunny Boy 4.0', '3011200342', '1390590', '8354'
'2023-07-31 00:00:00', 'SB4.0-1AV-41 342', 'Sunny Boy 4.0', '3011200342', '1390590', '8354'
'2023-08-01 00:00:00', 'SB4.0-1AV-41 342', 'Sunny Boy 4.0', '3011200342', '1396041', '5451'
```

- The entry for 2023-08-01 is wrong, because the last day was the 2023-07-31. I expect the last entry as of 2023-07-31.
- It seems like the entry for 2023-07-31 is duplicated from 2023-07-30. The entry for 2023-08-01 has the right values for 2023-07-31.

Here is the end of the log file:

```
Current Inverter Time: 01/08/2023 05:55:00
Inverter Wake-Up Time: 31/07/2023 20:49:45
Inverter Sleep Time  : 01/08/2023 05:54:17
**********************
* ArchiveMonthData() *
**********************
startTime: 01/08/2023 13:00:00
**********************
* ArchiveMonthData() *
**********************
startTime: 01/07/2023 12:00:00
Reading events: 2023-Aug-01
Tue Aug  1 05:55:02 2023: INFO: Done.

```

thank you for clarify this,
Peter.

NelliAtHome, 2023-08-01:
Maybe it would be better to run the cron job before midnight.

NelliAtHome, 2023-08-01:
[Carport_202308.log](https://github.com/SBFspot/SBFspot/files/12230130/Carport_202308.log)

Here is the log with the debug information. The log contains the run at 5:55 and the run from now.

sbf-, 2024-01-20:
A bit late I guess... Did you have these issues later on?
In the log I see
```
29/07/2023 : 1371.422kWh - 5.522kWh
30/07/2023 : 1382.236kWh - 10.814kWh
31/07/2023 : 1390.590kWh - 8.354kWh
01/08/2023 : 1396.041kWh - 5.451kWh
```
The log is created at `Tue Aug  1 16:09:18 2023`
Strange having already an entry for the current day

Did you check with Sunnyexplorer?

closed 2024-01-27 by sbf-