SBFspot/SBFspot #660: Sunny Island 8.0H reading power input and output

larschri15, 2023-08-15:
hello first of all great project i can now see my battery percent in home assistant but is it possible to also forward the power in and out via mqtt because i can't find what name it has also i saw in my log that it is displayed there so i am just looking for the mqtt SBFspot Alias
log:
Device Class:     Battery Inverter
Device Type:      Sunny Island 8.0H
Software Version: 03.01.10.R
SUSyID: 289 - SN: 1261015323
Batt. Charging Status: 52%
SUSyID: 289 - SN: 1261015323
Batt. Temperature: 23.0°C
Batt. Voltage    : 53.10V
Batt. Current    : -37.900A
SUSyID: 289 - SN: 1261015323
Grid Power Out : 2022W
Grid Power In  : 0W

sbf-, 2023-08-15:
You're looking for `MeteringWOut` and `MeteringWIn`

All other items are described in the [default SBFspot.cfg](https://github.com/SBFspot/SBFspot/blob/V3.9.7/SBFspot/SBFspot.cfg#L230)

larschri15, 2023-08-15:
thanks! I use the addon in home assistant there is also the possibility to sync every 30 sec instead of every 5 minutes e.g.