SBFspot/SBFspot #661: Support for SUNNY TRIPOWER x.0 SE

bodomenke, 2023-08-16:
Hi,

Just installed a SUNNY TRIPOWER 6.0 SE (Hybrid Inverter). Seems that it is not supported by SBFspot:

> Unknown Device Class. Report this issue at https://github.com/SBFspot/SBFspot/issues with following info: 0x00001F49 and Device Class=...    

Any chance to get the new SE inverters supported by SBFspot?

Thanks
Bodo

sbf-, 2023-08-16:
Hybrid inverters are supported since version 3.9.7
To see the version you have, use SBFspot -?

bodomenke, 2023-08-16:
Great. Thanks for advice! 
I am on 3.8.4. Unfortunately I do not know how to get a fresh sbfspot image, which runs on Synology, so I will contact nakla/sbfspot and ask to update the repository. 
Many thanks

sbf-, 2023-08-16:
See also #630

closed 2023-08-16 by sbf-