SBFspot/SBFspot #667: I might have missed something.. Help would be appreciated. Thx

lebeck69, 2023-09-11:
after successfully build the spfspot  … I’m now trying to run sbfspot on a raspbery pi 2.

I used this install command line :
curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash

BT works from start for my 2 inverters.
Tables are created.

MQTT works when i type the following commands and listen to the topic on HA.
./SBFspot -v5 -mqtt

But then nothing …
-     nohting is added to HA MQTT (is auto discovery working ?)
-     I dont see sbfspot running frequently on my pi
-     i don’t see daily log files in my log folder.

Any idea what I could be missing ?

(my pi provides a correct date and time… and I’ve left default config interval…)

sbf-, 2023-09-11:
What is the output of `crontab -l`

lebeck69, 2023-09-11:
I went a bit further ..

I see the CRON job running
Sep 11 17:20:01 SMApi CRON[2126]: (lebeck) CMD (/usr/local/bin/sbfspot.3/daydata)
Sep 11 17:20:01 SMApi CRON[2125]: (CRON) info (No MTA installed, discarding output)

I completely new to this .. but to me the CRON and the command './SBFspot -v5 -mqtt' seem to be executed with my user ?!

lebeck69, 2023-09-12:
Problem was a folder with space at the end ..
Took me some time because I aint a linux expert.

Still have to do the HA integration..
seem that auto discovery isn't working
if you have any tip.. I'm listening :)

lebeck69, 2023-09-12:
I have 2 inverters and MIS enable.
not sure to be correct but mqtt topic should be unique ?!

I see the MQTT send for each of my  inverters but with same topic ..

MQTT: Publishing (homeassistant/sbfspot_Inc/sbfspot_{serial})  ..
MQTT: Publishing (homeassistant/sbfspot_Inc/sbfspot_{serial})  ..

{serial} is not replace per inverter
I've also tryed{InvSerial}
Is that not the correct way to do it ?

closed 2023-10-28 by sbf-