SBFspot/SBFspot #670: Issue with sbfspotupload service on Windows

kuges, 2023-09-19:
I’m running 3.9.6 on Windows and really scratching my head. Everything appears to be configured correctly and when I run the test command as per the quick reference guide `"C:\Program Files (x86)\SBFspot.3\SBFspot" -v -finq -nocsv` - it all runs fine and I see data is uploaded to pvoutput. 

After installing and running the sbfspotupload service it doesn’t appear to generate any data to upload and I can’t work out why. I just get a standard `Retrieving new datapoints from DB ` log entry when the service runs, and when checking the SQLite database there’s no data for the period.

sbf-, 2023-09-19:
You should run sbfspot.exe every 5 minutes by means of scheduled task (like cron in linux)

kuges, 2023-10-01:
Thank you, that worked.

closed 2023-10-01 by kuges