SBFspot/SBFspot #671: MQTT message config

jeedee5, 2023-09-22:
Hi, 

I am trying to generate an MQTT message, that my receiving device can understand... 

that device expects a massage in the following way:
`{"pv": {"PAC1": 0.0} }`

In SBFSpot.cfg, I tried multiple configurations, but I can't find a way to add double quotes around "pv"
closest I can get is with the following configuration:
```
MQTT_PublisherArgs=-h {host} -t {topic} -r -m "{"\"\""pv"\"\"":{message}}"
MQTT_Data=PAC1,UAC1
```

The message ends up in the broker in the following way... replacing the double quotes arround "pv", with 2 single quotes before and after it...
`{''pv'':"PAC1": 57.000,"UAC1": 234.900}`

Anyone an idea on how to fix it?

sbf-, 2023-09-22:
> Anyone an idea on how to fix it?

In the current version it is not possible.

MD1848, 2024-03-12:
(I don't know if this fits here)
I compiled the program myself on an Intel computer with Debian 12 and got the error message with version 3.9.8
"
...
10,"BTSignal": 0.000,"InvWakeupTm": "12/03/2024 07:13:30","InvSleepTm": "12/03/2024 10:38:34"
sh: 1: -h: not found
MQTT: Failed to execute '' mosquitto client installed?
Error 32512 while publishing to MQTT Broker
********************
.."
I haven't changed anything in the CFG file;
With version 3.9.7 the error does not occur and the data is transferred

do I need to change something?

sbf-, 2024-03-12:
> MQTT: Failed to execute '' mosquitto client installed?

Can you check the `MQTT_Publisher` entry in the config file? It looks like it is missing.

MD1848, 2024-03-12:
> > MQTT: Failed to execute '' mosquitto client installed?
> 
> Can you check the `MQTT_Publisher` entry in the config file? It looks like it is missing.

the call in a shell works:

/SBFspot.3.9.9/SBFspot$ /usr/bin/mosquitto_pub -h 172.19.23.24 -t test -m xxxx -u user -P password -d
Client (null) sending CONNECT
Client (null) received CONNACK (0)
Client (null) sending PUBLISH (d0, q0, r0, m1, 'test', ... (4 bytes))
Client (null) sending DISCONNECT

sbf-, 2024-03-12:
> the call in a shell works

Yes, that's why I asked to check the config

MD1848, 2024-03-12:
> > the call in a shell works
> 
> Yes, that's why I asked to check the config

see above; I have added the log from 3.9.7 and 3.9.8

sbf-, 2024-03-12:
Ok, have to check this...