SBFspot/SBFspot #673: MQTT connection not possible with username/password

nubbler, 2023-09-28:
I tried to setup mqtt with password, but it didn't work....

MQTT: Publishing (sbfspot) Timestamp": "28/09/2023 09:20:51""SunRise": "28/09/2023 07:22:00""SunSet": "28/09/2023 19:11:00""InvSerial": 2100075453"InvName": "SN: 2100075453""InvStatus": "Ok""EToday": 2.006"ETotal": 89852.302"PACTot": 2692.000"UDC1": 355.280"UDC2": 359.730"IDC1": 3.858"IDC2": 3.965"PDC1": 1370.000"PDC2": 1426.000 **Connection Refused: not authorised**. Error: The connection was refused. MQTT: Failed te execute '/usr/bin/mosquitto_pub' mosquitto client installed? Error 1280 while publishing to MQTT Broker

I I run

mosquitto_sub -d -h [ip] -t sbfspot -u [user] -P [password]

it runs and the system connects to mqtt broker.

but with SBFSpot it didn't work.

SBFspot.cfg:

MQTT_Publisher=/usr/bin/mosquitto_pub
MQTT_Host=[ip]
MQTT_Topic=sbfspot_openhabian
MQTT_ItemFormat="{key}": {value}
MQTT_ItemDelimiter=comma
MQTT_PublisherArgs=-h {host} -t {topic} -r -m "{{message}}" -u [user] -P [password]

Because running mosquitto_sub in comand line works I think mqtt is set up and working well.
So something in SBFspot.cfg is wrong? But I don't find a mistake?

Did I anything missed?

sbf-, 2023-09-28:
Complex password? Maybe some characters of your password aren´t passed. I have to check but in the meantime try with a simple password.

nubbler, 2023-09-28:
No, there are just numbers and letter.
I tried also with new user and just letters in the password, no change....

But I recognized, the test that works was with mosquitto_sub, in SBFspot.cfg was mosquitto_pub used. WHat is the difference here?

sbf-, 2023-09-28:
sub is for the client who subscribes for a topic
pub is for the client who publishes the data
There is only one publisher (SBFspot) and one or more subscribers

nubbler, 2023-09-28:
ok, understand. That means to subscibe to the broker does work, but not to publish to the broker.
But the Connection is Refused with error  "not authorised".

So the error is somethere at the Broker. But the user/password should be the same for publishing or subscribtion, right?

Or what do I need to set up at broker side that he accepts a SBFSpot connection?

Sorry, I'm complete new at the MQTT Topic....

nubbler, 2023-09-28:
I tested now a little deeper:

on MQTT broker I subscribe:

mosquitto_sub -d -h [MQTT Broker IP] -t sbfspot_openhabian -u [user] -P [password]

On SBF-Spot raspi I publish with

mosquitto_pub -d -h [MQTT Broker IP] -t sbfspot_openhabian -u [user] -P [passwod] -m test -r

--> this works, message test was received at subscriber side.

But if I run SBFSpot wit -mqtt argument it doesent work and brings the error:

Connection Refused: not authorised.
Error: The connection was refused.
MQTT: Failed te execute '/usr/bin/mosquitto_pub' mosquitto client installed?
Error 1280 while publishing to MQTT Broker

I checked MQTT Settings in SBFspot.cfg several times and find no error.
Host, user, password, mosquitto_pub path, Topic --> all matchs with the options I used in the comand line.
I also aranged the order of the arguments in MQTT_PublisherArgs to the comand I used in comand line. Doesn't help.

Is there anything else I can/need to check?

nubbler, 2023-09-28:
ok, the problem was sitting in front of the monitor.
I editing the false SBFSpot.cfg.

Thank you for your support.

closed 2023-09-28 by nubbler