SBFspot/SBFspot #675: Can SBFspot be setup for Bookworm?

HeinvdW, 2023-10-28:
SBFspot isn't prepared for a Bookworm yet.

```
curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config | sudo bash
SBFspot Configuration Tool V1.3.4
Running on Raspberry Pi 4 Model B Rev 1.5
Checking whiptail... Installed.
architecture=arm64
Debian GNU/Linux 12 (bookworm)
username=pi
userhome=/home/pi
Error: Unable to get list of available SBFspot versions for 'bookworm'
```

sbf-, 2023-10-28:
I'm aware of this. See also [issue 67](https://github.com/SBFspot/sbfspot-config/issues/67)

sbf-, 2023-11-12:
V3.9.7 updated for bookworm

⚠️ New installation command:
`sudo bash -c "$(curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config)"`

closed 2023-11-12 by sbf-

HeinvdW, 2023-11-13:
Perfect!
Thanks a lot!