SBFspot/SBFspot #676: Unable to write to logfile

Ben1asdfas, 2023-10-29:
Hello together,

i got problems write the logfiles out of the SBFspotUpload Service.

How do i grant my user or this service rights to write logfiles in the directory /var/log/sbfspot.3 ?

[image]

SBFspot, 2023-10-29:
Recent versions of SBFspot (as from 3.9.6) and upload daemon should not write to logfiles anymore

Ben1asdfas, 2023-10-29:
I installed the latest version as described in the installation guide, where i can check or update ?

Ben1asdfas, 2023-10-31:
Hi, for cearification, i isntalled version 3.9.7 and it does write log files into /var/log/sbfspot.3

My problem was i used the wrong user to write.

closed 2023-10-31 by Ben1asdfas

sbf-, 2023-10-31:
> Recent versions of SBFspot (as from 3.9.6) and upload daemon should not write to logfiles anymore

That's true, but only if `logdir` is removed from `SBFspotUpload.cfg`