SBFspot/SBFspot #686: updarte to 3.9.8 failed

wokkeltje13, 2024-01-24:
[image]

[image]

previous updates installed OK.
Any idea how I can troubleshoot?

wokkeltje13, 2024-01-24:
Installing 3.9.7 has no issue 
sudo bash -c "$(curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config)" -s 3.9.7

sbf-, 2024-01-24:
Have you tried multiple times? I installed it on new bullseye 32bit without issues.

After installation of 3.9.8 you can run it manually to see more output:
`/usr/local/bin/sbfspot.3/SBFspot -finq -v5`

wokkeltje13, 2024-01-24:
I receive "Illegal instruction", nothing more.
Tried multiple times on a rPi Zero W
Operating System: Raspbian GNU/Linux 11 (bullseye)
Kernel: Linux 6.1.21+
Architecture: arm

sbf-, 2024-01-26:
Binaries were compiled for `arm32v7` instead of `arm32v6`
Should run on RPi zero now

closed 2024-01-26 by sbf-

wokkeltje13, 2024-01-29:
tnx, this solved the issue.