SBFspot/SBFspot #687: Logon failed. Check 'USER' Password during installation

Gonzomaster, 2024-01-25:
Hello,

I had a problem with my raspberry PI and was obliged to re-install it from scratch on a brand new SD Card... During installation, I'm getting the error "Logon failed. Check 'USER' Password during installation" and the installation stops. I've used the default password 0000 as I didn't change it. Here is the output of debug of the 3.9.7 version (same problem with the version 3.9.8). Isa there also a possibility to install the old versions as I don't remember with which version I was running before... Help !

SBFspot V3.9.7
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -finq -nocsv -d5 -v5
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'

Configuration settings:
BTAddress=00:80:25:29:1C:96
MIS_Enabled=0
BTConnectRetries=10
Password=<undisclosed>
Plantname=Gonzosolar
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=50.85
Longitude=4.35
Timezone=Europe/Brussels
Locale=en-US
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
SQL_Database=/home/pi/smadata/SBFspot.db
End of Config

Thu Jan 25 14:40:00 2024: INFO: Starting...
sunrise: 08:29
sunset : 17:19
Connecting to 00:80:25:29:1C:96 (1/10)
Initialising...
SUSyID: 125 - SessionID: 956944797
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 96 1C 29 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 04 
00000020: 70 00 02 00 00 00 00 01 00 00 
00000030: 00 
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 96 1C 29 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 04 
00000020: 70 00 02 00 00 00 00 01 00 00 
00000030: 00 
<<<=================================>>>
MAX_pcktBuf is now 31 bytes
SMA netID=02
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 00 00 00 00 00 00 
00000010: 96 1C 29 25 80 00 02 00 00 04 
00000020: 70 00 02 00 00 00 00 01 00 00 
00000030: 00 
31 Bytes sent
getPacket(5)
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 96 1C 29 25 80 00 
00000010: 00 00 00 00 00 00 0A 00 96 1C 
00000020: 29 25 80 00 01 D9 2D AA EB 27 
00000030: B8 
cmd=10
Received 18 bytes
Received 2 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 14 00 6A 96 1C 29 25 80 00 
00000010: 00 00 00 00 00 00 0C 00 02 00 
cmd=12
Received 18 bytes
Received 16 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 22 00 5C 96 1C 29 25 80 00 
00000010: 00 00 00 00 00 00 05 00 96 1C 
00000020: 29 25 80 00 01 01 D9 2D AA EB 
00000030: 27 B8 02 01 
cmd=5
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 22 00 5C 96 1C 29 25 80 00 
00000010: 00 00 00 00 00 00 05 00 96 1C 
00000020: 29 25 80 00 01 01 D9 2D AA EB 
00000030: 27 B8 02 01 
<<<=================================>>>
MAX_pcktBuf is now 34 bytes
Local BT address: B8:27:EB:AA:2D:D9
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 3F 00 41 D9 2D AA EB 27 B8 
00000010: FF FF FF FF FF FF 01 00 7E FF 
00000020: 03 60 65 09 A0 FF FF FF FF FF 
00000030: FF 00 00 7D 5D 00 9D D1 09 39 
00000040: 00 00 00 00 00 00 02 80 00 02 
00000050: 00 00 00 00 00 00 00 00 00 00 
00000060: 27 D1 7E 
63 Bytes sent
getPacket(1)
Received 18 bytes
MAX_CommBuf is now 88 bytes
Received 88 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 6A 00 14 96 1C 29 25 80 00 
00000010: D9 2D AA EB 27 B8 01 00 7E FF 
00000020: 03 60 65 7D 33 90 7D 5D 00 9D 
00000030: D1 09 39 00 00 83 00 50 C7 5F 
00000040: 7D 5E 00 00 00 00 00 00 02 80 
00000050: 01 02 00 00 00 00 00 00 00 00 
00000060: 00 00 00 03 00 00 00 FF 00 00 
00000070: D8 67 00 20 01 00 83 00 50 C7 
00000080: 5F 7D 5E 00 00 0A 00 0C 00 00 
00000090: 00 00 00 00 00 03 00 00 00 01 
00000100: 01 00 00 19 81 7E 
cmd=1
PacketLength=106
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E FF 03 60 65 13 90 7D 00 9D 
00000010: D1 09 39 00 00 83 00 50 C7 5F 
00000020: 7E 00 00 00 00 00 00 02 80 01 
00000030: 02 00 00 00 00 00 00 00 00 00 
00000040: 00 00 03 00 00 00 FF 00 00 D8 
00000050: 67 00 20 01 00 83 00 50 C7 5F 
00000060: 7E 00 00 0A 00 0C 00 00 00 00 
00000070: 00 00 00 03 00 00 00 01 01 00 
00000080: 00 19 81 7E 
<<<=================================>>>
MAX_pcktBuf is now 84 bytes
Serial Nr: 7E5FC750 (2120206160)
logoffSMAInverter()
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 3B 00 45 D9 2D AA EB 27 B8 
00000010: FF FF FF FF FF FF 01 00 7E FF 
00000020: 03 60 65 08 A0 FF FF FF FF FF 
00000030: FF 00 03 7D 5D 00 9D D1 09 39 
00000040: 00 03 00 00 00 00 03 80 0E 01 
00000050: FD FF FF FF FF FF 0F AF 7E 
59 Bytes sent
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 14 00 6A D9 2D AA EB 27 B8 
00000010: 96 1C 29 25 80 00 03 00 05 00 
20 Bytes sent
getPacket(4)
Received 18 bytes
Received 6 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 18 00 66 96 1C 29 25 80 00 
00000010: D9 2D AA EB 27 B8 04 00 05 00 
00000020: 00 00 C4 00 
cmd=4
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 18 00 66 96 1C 29 25 80 00 
00000010: D9 2D AA EB 27 B8 04 00 05 00 
00000020: 00 00 C4 00 
<<<=================================>>>
BT Signal=76.9%
logonSMAInverter()
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 53 00 2D D9 2D AA EB 27 B8 
00000010: FF FF FF FF FF FF 01 00 7E FF 
00000020: 03 60 65 0E A0 FF FF FF FF FF 
00000030: FF 00 01 7D 5D 00 9D D1 09 39 
00000040: 00 01 00 00 00 00 04 80 0C 04 
00000050: FD FF 07 00 00 00 84 03 00 00 
00000060: B2 64 B2 65 00 00 00 00 B8 B8 
00000070: B8 B8 88 88 88 88 88 88 88 88 
00000080: 8B B4 7E 
83 Bytes sent
getPacket(1)
Received 18 bytes
Received 66 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 54 00 2A 96 1C 29 25 80 00 
00000010: D9 2D AA EB 27 B8 01 00 7E FF 
00000020: 03 60 65 0E D0 7D 5D 00 9D D1 
00000030: 09 39 00 01 83 00 50 C7 5F 7D 
00000040: 5E 00 01 00 01 00 00 04 80 0D 
00000050: 04 FD FF 07 00 00 00 84 03 00 
00000060: 00 B2 64 B2 65 00 00 00 00 B8 
00000070: B8 B8 B8 88 88 88 88 88 88 88 
00000080: 88 89 2E 7E 
cmd=1
PacketLength=84
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E FF 03 60 65 0E D0 7D 00 9D 
00000010: D1 09 39 00 01 83 00 50 C7 5F 
00000020: 7E 00 01 00 01 00 00 04 80 0D 
00000030: 04 FD FF 07 00 00 00 84 03 00 
00000040: 00 B2 64 B2 65 00 00 00 00 B8 
00000050: B8 B8 B8 88 88 88 88 88 88 88 
00000060: 88 89 2E 7E 
<<<=================================>>>
Thu Jan 25 14:40:02 2024: CRITICAL: Logon failed. Check 'USER' Password
Thu Jan 25 14:40:02 2024: INFO: Done.

Thanks
Steve

sbf-, 2024-01-25:
Pairing is not the same as login.
Try with -installer -password:1111
This is the default installer password. If it is not changed it should work. 
Sunnyexplorer will fail too for user password

Gonzomaster, 2024-01-25:
I tried already via installer, same problem… I’m sure that it was working before with the default password :-(

Gonzomaster, 2024-01-26:
Hello,

Here is the debug output with installer..

Configuration settings:
BTAddress=00:80:25:29:1C:96
MIS_Enabled=0
BTConnectRetries=10
Password=<undisclosed>
Plantname=Gonzosolar
OutputPath=/home/pi/smadata/%Y
OutputPathEvents=/home/pi/smadata/%Y/Events
Latitude=50.85
Longitude=4.35
Timezone=Europe/Brussels
Locale=en-US
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
SQL_Database=/home/pi/smadata/SBFspot.db
End of Config

Fri Jan 26 07:56:34 2024: INFO: Starting...
sunrise: 08:28
sunset : 17:21
Connecting to 00:80:25:29:1C:96 (1/10)
Initialising...
SUSyID: 125 - SessionID: 955332548
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 96 1C 29 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 04 
00000020: 70 00 02 00 00 00 00 01 00 00 
00000030: 00 
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 96 1C 29 25 80 00 
00000010: 00 00 00 00 00 00 02 00 00 04 
00000020: 70 00 02 00 00 00 00 01 00 00 
00000030: 00 
<<<=================================>>>
MAX_pcktBuf is now 31 bytes
SMA netID=02
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 00 00 00 00 00 00 
00000010: 96 1C 29 25 80 00 02 00 00 04 
00000020: 70 00 02 00 00 00 00 01 00 00 
00000030: 00 
31 Bytes sent
getPacket(5)
Received 18 bytes
Received 13 bytes
[... 114 lines cut ...]
00000000: 7E 54 00 2A 96 1C 29 25 80 00 
00000010: D9 2D AA EB 27 B8 01 00 7E FF 
00000020: 03 60 65 0E D0 7D 5D 00 C4 37 
00000030: F1 38 00 01 83 00 50 C7 5F 7D 
00000040: 5E 00 01 00 01 00 00 04 80 0D 
00000050: 04 FD FF 0A 00 00 00 84 03 00 
00000060: 00 A3 57 B3 65 00 00 00 00 EC 
00000070: EC EC EC BB BB BB BB BB BB BB 
00000080: BB 89 7A 7E 
cmd=1
PacketLength=84
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E FF 03 60 65 0E D0 7D 00 C4 
00000010: 37 F1 38 00 01 83 00 50 C7 5F 
00000020: 7E 00 01 00 01 00 00 04 80 0D 
00000030: 04 FD FF 0A 00 00 00 84 03 00 
00000040: 00 A3 57 B3 65 00 00 00 00 EC 
00000050: EC EC EC BB BB BB BB BB BB BB 
00000060: BB 89 7A 7E 
<<<=================================>>>
Fri Jan 26 07:56:35 2024: CRITICAL: Logon failed. Check 'INSTALLER' Password
Fri Jan 26 07:56:35 2024: INFO: Done.

Steve

sbf-, 2024-01-26:
Sorry, can't help here... There are [plenty of similar issues](https://github.com/SBFspot/SBFspot/issues?q=Check+%27USER%27+Password) where it was indeed a wrong password
You can try with sunnyexplorer (latest version will force you to change PW when using 0000/1111 but I suppose you'll need the current PW to be able to change it.
Last option is to request a PUK code to reset the password

Gonzomaster, 2024-01-26:
OK I'll try that ! Thx for the support !

closed 2024-01-26 by Gonzomaster