SBFspot/SBFspot #688: Not able to set inverter time

polanjir, 2024-01-28:
Hi All,

I would like to ask if anyone knows where the problem might be. I have been running **SMA STP-15000TL-10** inverter for more than 10 years, but a month ago my data reading stopped working because the inverter does not keep time.
I am able to set the time successfully using either the official Sunny Explorer or `./SBFspot -v -settime`. But the next time I use SBFspot the time is back to `26/02/1908`. 🤷‍♂️
I was thinking that there might be a battery backup of some RTC chip at the end of its life. But I'm not sure if there's anything like that inside. And also it's strange that the inverter loses time very quickly even when under voltage. The RTC chip should keep time even without battery if it is properly powered.

On the other hand, I must point out that the inverter works w/o any issue. It produces energy from the solar panels and supplies the grid.

Here are my two attempts to set the time
```
pi@rpiglbs:/srv/SBFspot $ ./SBFspot -v -settime -cfg:SBFspot.glbs_ji.cfg
SBFspot V3.9.7
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -settime -cfg:SBFspot.glbs_ji.cfg
Reading config '/srv/SBFspot/SBFspot.glbs_ji.cfg'
Sun Jan 28 16:03:36 2024: INFO: Starting...
sunrise: 07:37
sunset : 16:46
Connecting to 00:80:25:AA:BB:CC (1/10)
Initialising...
SUSyID: 125 - SessionID: 806964044
SMA netID=01
Serial Nr: 7DCA88D2 (2110123456)
BT Signal=71.8%
Logon OK
Local Host Time: 28/01/2024 16:03:38
Plant Time     : 26/02/1908 00:29:16 (+636735234 sec)
TZ offset      : 3600 sec - DST: Off
Last Time Set  : 28/01/2024 16:03:26
Adjusting plant time...
New plant time is now 28/01/2024 16:03:38
Sun Jan 28 16:03:38 2024: INFO: Done.
pi@rpiglbs:/srv/SBFspot $
pi@rpiglbs:/srv/SBFspot $
pi@rpiglbs:/srv/SBFspot $ ./SBFspot -v -settime -cfg:SBFspot.glbs_ji.cfg
SBFspot V3.9.7
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -settime -cfg:SBFspot.glbs_ji.cfg
Reading config '/srv/SBFspot/SBFspot.glbs_ji.cfg'
Sun Jan 28 16:04:17 2024: INFO: Starting...
sunrise: 07:37
sunset : 16:46
Connecting to 00:80:25:AA:BB:CC (1/10)
Initialising...
SUSyID: 125 - SessionID: 809355283
SMA netID=01
Serial Nr: 7DCA88D2 (2110123456)
BT Signal=71.8%
Logon OK
Local Host Time: 28/01/2024 16:04:19
Plant Time     : 26/02/1908 00:29:57 (+636735234 sec)
TZ offset      : 3600 sec - DST: Off
Last Time Set  : 28/01/2024 16:03:38
Adjusting plant time...
New plant time is now 28/01/2024 16:04:19
Sun Jan 28 16:04:19 2024: INFO: Done.
```

sbf-, 2024-01-29:
Most likely the time isn't set, even when SBFspot (and Sunnyexplorer) shows `New plant time is now...`
The settime command is sent to the inverter but there is never a response if the time is actually set or not.

Also, a date/time of 26/02/1908 is not possible, because the time used by the inverter is Unix Epoch (32 bit) which starts at 01 jan 1970. But this behaviour is probably a bug in SBFspot (can you post a debuglog with `-settime -v5 -d5` to verify?)

> I was thinking that there might be a battery backup of some RTC chip at the end of its life. But I'm not sure if there's anything like that inside.

That's what I was thinking [some time ago](https://github.com/SBFspot/SBFspot/discussions/624#discussioncomment-5778518). I have opened my inverters and I didn't find any backup battery

polanjir, 2024-01-30:
Here is my debug log from trying to set the time.
```
poli@rpiglbs:/srv/SBFspot $ ./SBFspot -v5 -d5 -settime -cfg:SBFspot.glbs_ji.cfg
SBFspot V3.9.7
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v5 -d5 -settime -cfg:SBFspot.glbs_ji.cfg
Reading config '/srv/SBFspot/SBFspot.glbs_ji.cfg'

Configuration settings:
BTAddress=00:80:25:27:89:FA
MIS_Enabled=0
BTConnectRetries=10
Password=<undisclosed>
Plantname=glbs_ji
OutputPath=/srv/smadata/glbs_ji/%Y
OutputPathEvents=/srv/smadata/glbs_ji/%Y/Events
Latitude=50.094
Longitude=14.4299
Timezone=Europe/Prague
Locale=en-US
CalculateMissingSpotValues=1
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
SQL_Database=/srv/smadata/DB/SBFspot.db
End of Config

Tue Jan 30 13:35:39 2024: INFO: Starting...
sunrise: 07:35
sunset : 16:49
Connecting to 00:80:25:27:89:FA (1/10)
Initialising...
SUSyID: 125 - SessionID: 893827925
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 FA 89 27 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 01 00 00 00 00 01 00 00
00000030: 00
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 FA 89 27 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 01 00 00 00 00 01 00 00
00000030: 00
<<<=================================>>>
MAX_pcktBuf is now 31 bytes
SMA netID=01
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 00 00 00 00 00 00
00000010: FA 89 27 25 80 00 02 00 00 04
[... 186 lines cut ...]
Adjusting plant time...
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 5B 00 25 93 FC 02 70 F3 5C
00000010: FF FF FF FF FF FF 01 00 7E FF
00000020: 03 60 65 10 A0 FF FF FF FF FF
00000030: FF 00 00 7D 5D 00 55 BB 46 35
00000040: 00 00 00 00 00 00 06 80 0A 02
00000050: 00 F0 00 6D 23 00 00 6D 23 00
00000060: 00 6D 23 00 1D ED B8 65 1D ED
00000070: B8 65 1D ED B8 65 10 0E 00 00
00000080: 0C 00 00 00 01 00 00 00 13 CC
00000090: 7E
91 Bytes sent
New plant time is now 30/01/2024 13:35:41
logoffSMAInverter()
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 3B 00 45 93 FC 02 70 F3 5C
00000010: FF FF FF FF FF FF 01 00 7E FF
00000020: 03 60 65 08 A0 FF FF FF FF FF
00000030: FF 00 03 7D 5D 00 55 BB 46 35
00000040: 00 03 00 00 00 00 07 80 0E 01
00000050: FD FF FF FF FF FF 04 ED 7E
59 Bytes sent
Tue Jan 30 13:35:41 2024: INFO: Done.
```

sbf-, 2024-02-03:
The debug info only revealed a bug is SBFspot:
`Plant Time     : 27/02/1908 22:01:16 (+636735232 sec)`
Offset is positive (+636735232 sec is almost 20 year), but the date goes back in time. The actual time should be `04/04/2044 05:29:32`

Can you verify the events, if there are some useful entries about time setting?
`SBFspot -v5 -ae6 -am0 -ad0 -nocsv -nosql -finq`

polanjir, 2024-02-04:
You're right. Today I connected to the inverter via Sunny Explorer and the year was set to `2044` even though I set it to the correct date a week ago via this app. 🤷‍♂️ 

Here are the event logs
```
poli@rpiglbs:/srv/SBFspot $ ./SBFspot -v5 -ae6 -am0 -ad0 -nocsv -nosql -finq -cfg:SBFspot.glbs_ji.cfg
SBFspot V3.9.7
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v5 -ae6 -am0 -ad0 -nocsv -nosql -finq -cfg:SBFspot.glbs_ji.cfg
Reading config '/srv/SBFspot/SBFspot.glbs_ji.cfg'

Configuration settings:
BTAddress=00:80:25:27:89:FA
MIS_Enabled=0
BTConnectRetries=10
Password=<undisclosed>
Plantname=glbs_ji
OutputPath=/srv/smadata/glbs_ji/%Y
OutputPathEvents=/srv/smadata/glbs_ji/%Y/Events
Latitude=50.094
Longitude=14.4299
Timezone=Europe/Prague
Locale=en-US
CalculateMissingSpotValues=1
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
SQL_Database=/srv/smadata/DB/SBFspot.db
End of Config

Sun Feb  4 10:59:55 2024: INFO: Starting...
sunrise: 07:28
sunset : 16:57
Connecting to 00:80:25:27:89:FA (1/10)
Initialising...
SUSyID: 125 - SessionID: 953877316
SMA netID=01
Serial Nr: 7DCA88D2 (2110425298)
BT Signal=72.5%
Logon OK
Local Host Time: 04/02/2024 10:59:56
Plant Time     : 03/03/1908 19:25:24 (+636735224 sec)
TZ offset      : 3600 sec - DST: Off
Last Time Set  : 30/01/2024 13:35:41
The time difference between inverter/host is more than 3600 seconds.
As a precaution the plant time won't be adjusted.
To overrule this behaviour, execute SBFspot -settime
SUSyID: 128 - SN: 2110425298
Device Name:      SN: 2110425298
Device Class:     Solar Inverters
Device Type:      STP 15000TL-10
Software Version: 02.50.20.R
Packet status: 21
SUSyID: 128 - SN: 2110425298
Device Status:      Ok
SUSyID: 128 - SN: 2110425298
[... 187 lines cut ...]
06/12/2023 14:22:53 Time adjusted / old time
06/12/2023 14:15:03 Time adjusted / new time
06/12/2023 13:16:55 Time adjusted / old time
06/12/2023 13:15:02 Time adjusted / new time
06/12/2023 12:25:54 Time adjusted / old time
06/12/2023 12:20:02 Time adjusted / new time
06/12/2023 10:29:56 Time adjusted / old time
06/12/2023 10:20:02 Time adjusted / new time
06/12/2023 08:32:56 Time adjusted / old time
06/12/2023 08:20:02 Time adjusted / new time
06/12/2023 07:35:56 Time adjusted / old time
06/12/2023 07:30:03 Time adjusted / new time
05/12/2023 15:49:56 Time adjusted / old time
05/12/2023 15:40:02 Time adjusted / new time
05/12/2023 14:43:57 Time adjusted / old time
05/12/2023 14:40:03 Time adjusted / new time
05/12/2023 13:47:58 Time adjusted / old time
05/12/2023 13:40:02 Time adjusted / new time
05/12/2023 12:41:58 Time adjusted / old time
05/12/2023 12:40:03 Time adjusted / new time
05/12/2023 11:45:58 Time adjusted / old time
05/12/2023 11:45:03 Time adjusted / new time
05/12/2023 11:09:58 Time adjusted / old time
Sun Feb  4 11:00:00 2024: INFO: Done.
```

sbf-, 2024-02-05:
Maybe a reset (power off) of your inverter can do something magically?