SBFspot/SBFspot #690: Unknown Inverter Type - SMA STP10.0-3SE-40

Demel75, 2024-02-06:
SMA Sunny Tripower 10.0 Smart Energy (SMA STP10.0-3SE-40) seems to be not supported:

./SBFspot -nocsv -nosql -sp0 -finq

SBFspot V3.9.8
Yet another tool to read power production of SMA solar inverters
(c) 2012-2023, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -nocsv -nosql -sp0 -finq
**Unknown Inverter Type. Report this issue at https://github.com/SBFspot/SBFspot/issues with following info:**
ID='19051' and Type=<Fill in the exact inverter model> (e.g. SB1300TL-10)
00464B09 28 Tue Feb  6 17:25:49 2024 'Netzspannung Phase L1 gegen L2' 40490 40490 40490 40490
00464C09 28 Tue Feb  6 17:25:49 2024 'Netzspannung Phase L2 gegen L3' 40300 40300 40300 40300
00464D09 28 Tue Feb  6 17:25:49 2024 'Netzspannung Phase L3 gegen L1' 40520 40520 40520 40520
00464E09 28 Tue Feb  6 17:25:49 2024 '' 3 3 3 3

sbf-, 2024-02-06:
That's strange... it is defined in the taglist files:
`19051=STP10.0-3SE-40\0\SUNNY TRIPOWER 10.0 SE`
Can you verify `/usr/local/bin/sbfspot.3/TagListDE-DE.txt`?

Demel75, 2024-02-07:
thanks for that hint. the taglist files was not checked out proberly, so an old one was used.

Now all works fine.

closed 2024-02-07 by Demel75