SBFspot/SBFspot #692: Compile failure on Debian 12 - boost_ext.cpp: __time_64_t, time_t

JiriTywoniak, 2024-02-18:
I am not able to compile on Debian 12.

g++ boost_ext.cpp -c -Wall -O2 -Wno-unused-local-typedefs -Wno-psabi -o nosql/obj/boost_ext.o
...
/usr/include/x86_64-linux-gnu/bits/types/struct_timeval.h:11:3: error: ‘__time64_t’ does not name a type; did you mean ‘__timer_t’?
 ...
/usr/include/time.h:266:8: note: ‘time_t’ is defined in header ‘<ctime>’; did you forget to ‘#include <ctime>’?
...

$ cat /usr/include/boost/version.hpp
#define BOOST_VERSION 107400
...
#define BOOST_LIB_VERSION "1_74"

sbf-, 2024-02-19:
As a quick workaround, you can remove or comment out `#define __USE_TIME_BITS64` in `oslinux.h`

JiriTywoniak, 2024-02-19:
> Komu: SBFspot/SBFspot ***@***.***>
> github.com>
> __time_64_t, time_t (Issue #692)
> "

Thanks. It works now.
---------- Původní e-mail ----------

(https://github.com/SBFspot/SBFspot/issues/692#issuecomment-1952767038), or 
unsubscribe
"

commit aec029fd (2024-02-20): Fix #692 Compile failure on Debian 12 - 64 bit
commit 8b666d3f (2024-02-20): Fix #692 Compile failure on Debian 12 - 64 bit
commit 9e9b43c8 (2024-02-20): Fix #692 Compile failure on Debian 12 - 64 bit
closed 2024-02-20 by sbf-