SBFspot/SBFspot #694: Corrupt data

Driekes, 2024-03-04:
Hi,

I have recently seen a couple of updates that tackle corrupt data. I have seen last 2 weeks 2 issues, so I upgraded to latest version a couple of days ago. Today still an issue, so I thought to mention it here:

End of day yesterday:
```03/03/2024 17:45:00;24152,406;0,216
03/03/2024 17:50:00;24152,419;0,156
03/03/2024 17:55:00;24152,429;0,120
03/03/2024 18:00:00;24152,436;0,084
03/03/2024 18:05:00;24152,439;0,036
03/03/2024 18:10:00;24152,442;0,036
03/03/2024 18:15:00;24152,442;0,000
03/03/2024 18:20:00;24152,442;0,000
03/03/2024 18:25:00;24152,442;0,000
03/03/2024 18:30:00;24152,442;0,000
```

Data of this morning:

```
sep=;
Version CSV1|Tool SBFspot3.9.9 (Windows)|Linebreaks CR/LF|Delimiter semicolon|Decimalpoint comma|Precision 3

;STP5.0-3AV-40 157;STP5.0-3AV-40 157
;Sunny Tripower 5.0;Sunny Tripower 5.0
;3009496157;3009496157
;Total yield;Power
;Counter;Analog
dd/MM/yyyy HH:mm:ss;kWh;kW
04/03/2024 00:00:00;24046,190;0,000
04/03/2024 00:05:00;24046,190;0,000
04/03/2024 00:10:00;24046,190;0,000
04/03/2024 00:15:00;24046,190;0,000
04/03/2024 00:20:00;24046,190;0,000
04/03/2024 00:25:00;24046,190;0,000
04/03/2024 00:30:00;24046,190;0,000
04/03/2024 00:35:00;24046,190;0,000
04/03/2024 00:40:00;24046,190;0,000
04/03/2024 00:45:00;24046,190;0,000
04/03/2024 00:50:00;24046,190;0,000
04/03/2024 00:55:00;24046,190;0,000
04/03/2024 01:00:00;24046,190;0,000
04/03/2024 01:05:00;24046,190;0,000
04/03/2024 01:10:00;24046,190;0,000
04/03/2024 01:15:00;24046,190;0,000
04/03/2024 01:20:00;24046,190;0,000
04/03/2024 01:25:00;24046,190;0,000
04/03/2024 01:35:00;24152,442;637,512
04/03/2024 01:40:00;24152,442;0,000
04/03/2024 01:45:00;24152,442;0,000```

Which then continues its normal routine. However the night data is just corrupt and (assuming) from an earlier date.

sbf-, 2024-03-04:
That's indeed another issue not being handled by current codebase...

Is the data correct the next day? Can you test this tomorrow? So, if you query for the data of 04 march on the 5th of march?

Driekes, 2024-03-04:
I had the exact same thing on 26-02, and after that I upgraded SBFSpot to latest version. Before that i had long time without issues, and before that was on my old convertor which caused complete ranges of data being incorrect, so SMA replaced it.

This is end of day on 25:
```
25/02/2024 17:55:00;24046,184;0,060
25/02/2024 18:00:00;24046,188;0,048
25/02/2024 18:05:00;24046,190;0,024
25/02/2024 18:10:00;24046,190;0,000
25/02/2024 18:15:00;24046,190;0,000
25/02/2024 18:20:00;24046,190;0,000
``` 
This is start of day on 26:
```
sep=;
Version CSV1|Tool SBFspot3.9.9 (Windows)|Linebreaks CR/LF|Delimiter semicolon|Decimalpoint comma|Precision 3

;STP5.0-3AV-40 157;STP5.0-3AV-40 157
;Sunny Tripower 5.0;Sunny Tripower 5.0
xx
;Total yield;Power
;Counter;Analog
dd/MM/yyyy HH:mm:ss;kWh;kW
26/02/2024 00:00:00;23910,317;0,000
26/02/2024 00:05:00;23910,317;0,000
26/02/2024 00:10:00;23910,317;0,000
26/02/2024 00:15:00;23910,317;0,000
26/02/2024 00:20:00;23910,317;0,000
26/02/2024 00:25:00;23910,317;0,000
26/02/2024 00:30:00;23910,317;0,000
26/02/2024 00:35:00;23910,317;0,000
26/02/2024 00:40:00;23910,317;0,000
26/02/2024 00:45:00;23910,317;0,000
26/02/2024 00:50:00;23910,317;0,000
26/02/2024 00:55:00;23910,317;0,000
26/02/2024 01:00:00;23910,317;0,000
26/02/2024 01:05:00;23910,317;0,000
26/02/2024 01:10:00;23910,317;0,000
26/02/2024 01:15:00;23910,317;0,000
26/02/2024 01:20:00;23910,317;0,000
26/02/2024 01:25:00;23910,317;0,000
26/02/2024 01:30:00;23910,317;0,000
26/02/2024 01:35:00;23910,317;0,000
26/02/2024 01:40:00;23910,317;0,000
26/02/2024 01:45:00;23910,317;0,000
26/02/2024 01:50:00;23910,317;0,000
26/02/2024 01:55:00;24046,190;1630,476
26/02/2024 02:00:00;24046,190;0,000
26/02/2024 02:05:00;24046,190;0,000
26/02/2024 02:10:00;24046,190;0,000
26/02/2024 02:15:00;24046,190;0,000
26/02/2024 02:20:00;24046,190;0,000
26/02/2024 02:25:00;24046,190;0,000
26/02/2024 02:30:00;24046,190;0,000
``` 
This is start of day on 27:
```
sep=;
Version CSV1|Tool SBFspot3.9.9 (Windows)|Linebreaks CR/LF|Delimiter semicolon|Decimalpoint comma|Precision 3

;STP5.0-3AV-40 157;STP5.0-3AV-40 157
;Sunny Tripower 5.0;Sunny Tripower 5.0
;x;x
;Total yield;Power
;Counter;Analog
dd/MM/yyyy HH:mm:ss;kWh;kW
27/02/2024 00:00:00;24049,913;0,000
27/02/2024 00:05:00;24049,913;0,000
27/02/2024 00:10:00;24049,913;0,000
27/02/2024 00:15:00;24049,913;0,000
27/02/2024 00:20:00;24049,913;0,000
27/02/2024 00:25:00;24049,913;0,000
27/02/2024 00:30:00;24049,913;0,000
27/02/2024 00:35:00;24049,913;0,000
27/02/2024 00:40:00;24049,913;0,000
27/02/2024 00:45:00;24049,913;0,000
``` 
So issue corrects itself. The only real issue I have is that PVOutput gets stuck. So what I needed to do was remove the day data and remove the total day data on pvoutput and then I updated the PVOutput column in PHPMyAdmin to 'NULL' just after the faulty data, and it gets corrected.

I think it would be good that this number can only increase, and it will neglect any decreasing number. However, now it corrects itself after one instance, not sure if this can go on for days.

Driekes, 2024-03-04:
> Can you test this tomorrow? So, if you query for the data of 04 march on the 5th of march?

How would you like me to do that? The data is already in MySQL/csv, and uploaded to pvoutput already.

sbf-, 2024-03-06:
@Driekes Can you check the monthdata csv and monthdata table in mySQL? V3.9.9 has regression and it is better not to use it.
In some cases the monthdata has missing first day and all data is shifted by one day.
However, this is not related to your issue here.

Version 3.9.8 is safe to use

Driekes, 2024-03-23:
@sbf- missed the comments, sorry! I will revert to 3.9.8 then.

Didn't see your comments yet, sorry!

> My guess is the data is changed to the correct values
I have manually adjusted them so I suppose i will not see the difference anyway.

> In some cases the monthdata has missing first day and all data is shifted by one day.
I am not using SBF spot for month totals, I am using SBF spot to upload to PVOutput. I'm using a Home assistant integration atm to have live data there. So I think that is not too problematic for me.

I will revert to 3.9.8 nevertheless.

Driekes, 2024-04-08:
Still had the issue a couple of days ago. I have previously fixed it by removing everything of that day on PVOutput, and then setting PVoutput column to 'NULL' for the correct rows. Is that still the way to go or can I force update PVOutput as well with a SFB command?

commit d4a7eeb3 (2024-06-14): Fix #694 Corrupt data
closed 2024-06-14 by sbf-