SBFspot/SBFspot #696: Problem after re-installing SBFspot

roelvdh, 2024-04-27:
The SD-card of my old Raspi 2 suddenly broke and I had to re-install SBFspot on a new SD.  That seemed to work fine, but the data seem not to be archived at all and, also, there are funny Error 21 lines (from SMA) in the logs. Nothing was changed in the 2 inverters. I include 1 cfg file and 1 log file. Actually, I use 2 cfg files for 2 inverters and have modified daydata and monthdata accordingly (that worked better for me than using Multiple Inverter)

cfg1:
`# SMA Inverter's Bluetooth address
# Linux  : hcitool scan
BTAddress=00:80:25:26:xx:xx

#Multi Inverter Support: Default=0 Disabled
MIS_Enabled=0

# BTConnectRetries
# Number of Bluetooth Connection attempts (1-15; Default=10)
BTConnectRetries=10

# User password (default 0000)
Password=xxxxxxxxxxxxxxxxxxxxxxx

# Plantname
Plantname=Roel1

# OutputPath (Place to store CSV files)
# Linux  : /home/pi/smadata/%Y
# %Y %m and %d will be expanded to Year Month and Day
OutputPath=/home/pi/smadata/%Y

# OutputPathEvents (Place to store CSV files for events)
# If omitted, OutputPath is used
OutputPathEvents=/home/pi/smadata/%Y/Events

# Timezone
# Select the right timezone in date_time_zonespec.csv
Timezone=Europe/Amsterdam

# Position of pv-plant https://www.gps-coordinates.net/maps
Latitude=52.xxxxxxx
Longitude=4.yyyyyyy

# Locale
# Default en-US
Locale=en-US

# DateTimeFormat (default %d/%m/%Y %H:%M:%S)
DateTimeFormat=

# DateFormat (default %d/%m/%Y)
DateFormat=%d/%m/%Y

# TimeFormat (default %H:%M:%S)
TimeFormat=%H:%M:%S

# DecimalPoint (comma/point default comma)
DecimalPoint=comma

# Calculate Missing SpotValues
# If set to 1, values not provided by inverter will be calculated
# eg: Pdc1 = Idc1 * Udc1
CalculateMissingSpotValues=0

# SynchTime (0-30 - 0=disabled, 1=once a day (default), 7=once a week, 30=once a                                                                              month)
# If set to non-zero value, the plant time is synchronised with local host time
# Some inverters don't have a real-time clock
SynchTime=7

# SynchTimeLow (1-120 - default 1)
# SynchTimeHigh (1200-3600 - default 3600)
# Plant time is adjusted to local host time when SynchTime=1 and
# time difference is between SynchTimeLow and SynchTimeHigh limits
SynchTimeLow=60
SynchTimeHigh=3600

# SunRSOffset
# Offset to start before sunrise and end after sunset (0-3600 - default 900 seco                                                                             nds)
SunRSOffset=900

# With CSV_* settings you can define the CSV file format

# CSV_Export (default 1 = Enabled)
# Enables or disables the CSV Export functionality
CSV_Export=0

# CSV_ExtendedHeader (default 1 = On)
# Enables or disables the SMA extended header info (8 lines)
# isep=;
# Version CSV1|Tool SBFspot|Linebreaks CR/LF|Delimiter semicolon|Decimalpoint co                                                                             mma|Precision 3
# etc...
# This is usefull for manual data upload to pvoutput.org
CSV_ExtendedHeader=1

# CSV_Header (default 1 = On)
# Enables or disables the CSV data header info (1 line)
# dd/MM/yyyy HH:mm:ss;kWh;kW
# This is usefull for manual data upload to pvoutput.org
# If CSV_ExtendedHeader is enabled, CSV_Header is also enabled
CSV_Header=1

# CSV_SaveZeroPower (default 1 = On)
# When enabled, daily csv files contain all data from 00:00 to 23:55
# This is usefull for manual data upload to pvoutput.org
CSV_SaveZeroPower=1

# CSV_Delimiter (comma/semicolon default semicolon)
CSV_Delimiter=semicolon

# CSV_Spot_TimeSource (Inverter|Computer default Inverter)
CSV_Spot_TimeSource=inverter

# CSV_Spot_WebboxHeader (Default 0 = Off)
# When enabled, use Webbox style header (DcMs.Watt[A];DcMs.Watt[B]...)
CSV_Spot_WebboxHeader=0

SQL_Database=/home/pi/smadata/SBFspot.db

# End of Config
`
log1:
`SBFspot V3.9.8
Yet another tool to read power production of SMA solar inverters
(c) 2012-2023, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 32 bit with SQLite support
Commandline Args: -v -ad1 -am0 -ae1 -cfg/usr/local/bin/sbfspot.3/SBFspot1.cfg.rvdh
Reading config '/usr/local/bin/sbfspot.3/SBFspot1.cfg.rvdh'
Sat Apr 27 12:35:01 2024: INFO: Starting...
sunrise: 06:19
sunset : 20:59
Connecting to 00:80:25:26:F8:35 (1/10)
Initialising...
SUSyID: 125 - SessionID: 978806982
SMA netID=01
Serial Nr: 77570578 (2002191736)
BT Signal=73.3%
Logon OK
Failed to get current plant time (-1)
SetPlantTime returned an error: -1
SUSyID: 99 - SN: 2002191736
Device Name:      Schuin dak
Device Class:     Solar Inverters
Device Type:      SB 1600TL-10
Software Version: 12.12.206.R
Packet status: 21
SUSyID: 99 - SN: 2002191736
Device Status:      Ok
Packet status: 21
SUSyID: 99 - SN: 2002191736
Device Temperature: Information not available
Packet status: 21
getGridRelayStatus returned an error: 21
SUSyID: 99 - SN: 2002191736
Energy Production:
        EToday: 1.461kWh
        ETotal: 19484.151kWh
        Operation Time: 49422.78h
        Feed-In Time  : 40432.71h
Packet status: 21
getSpotDCPower returned an error: 21
Packet status: 21
getSpotACPower returned an error: 21
SUSyID: 99 - SN: 2002191736
DC Spot Data:
        MPPT 1 Pdc:   0.000kW - Udc: 230.00V - Idc:  1.806A
        MPPT 2 Pdc:   0.000kW - Udc:   0.00V - Idc:  0.000A
        Calculated Total Pdc:   0.000kW
AC Spot Data:
        Phase 1 Pac :   0.000kW - Uac: 230.10V - Iac:  1.684A
        Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   0.387kW - Calculated Pac:   0.000kW
        Efficiency  :    0.00%
SUSyID: 99 - SN: 2002191736
Grid Freq. : 49.98Hz
SUSyID: 99 - SN: 2002191736
Current Inverter Time: 27/04/2024 12:35:14
Inverter Wake-Up Time: 27/04/2024 12:35:14
Inverter Sleep Time  : 27/04/2024 12:35:15
********************
* ArchiveDayData() *
********************
startTime: 27/04/2024 00:00:00
Reading events: 2024-Apr-01
Sat Apr 27 12:35:16 2024: INFO: Done.
`

roelvdh, 2024-04-27:
Comment lines in the first Code are converted to Headings and I can't edit that

sbf-, 2024-04-27:
> Comment lines in the first Code and the bottom of the second Code were converted to Headings and I tried to edit that by changing #'s to >'s, creating vertical lines

Just select your text and click the `code block function <>`
[image]

sbf-, 2024-04-27:
> but the data will just not appear in /home/pi/smadata/2024

You mean no CSV files?
What is the CSV_export setting in the config file? If the data is in the DB, there should be a CSV if the export is not disabled

roelvdh, 2024-04-27:
Thank you for remembering me I need new glasses. CSV_export should be on
default=1, but it stood at 0 for some unknown reason.

roelvdh, 2024-04-27:
Code worked very funny. I used the "<>" and then part of the #-s for comment were translated into big Headings. It may be my browser though.

roelvdh, 2024-04-28:
I am still wondering about the SMA Error 21's. Certainly not a SBFspot problem but do you happen to know what they mean? I can't remember I had those errors in the past.

sbf-, 2024-04-28:
Error 21 is returned when a certain data item is not available. For example, not all inverters return internal temperature.

This is OK:
```
Packet status: 21
SUSyID: 99 - SN: 2002191736
Device Temperature: Information not available

```

This is never OK, all inverters should return power production for DC and AC
```
Packet status: 21
getSpotDCPower returned an error: 21
Packet status: 21
getSpotACPower returned an error: 21

```

Maybe a full debug log (-d5 -v5) can show something. Attach it as a file, do not copy/paste the output in this thread.

roelvdh, 2024-04-28:
[Status21_Inverter1.txt](https://github.com/SBFspot/SBFspot/files/15142097/Status21_Inverter1.txt)
[Status21_Inverter2.txt](https://github.com/SBFspot/SBFspot/files/15142098/Status21_Inverter2.txt)
I appreciate your help. Only Inverter1 has the Status 21 errors.

roelvdh, 2024-04-28:
[sbf.txt](https://github.com/SBFspot/SBFspot/files/15142172/sbf.txt)
Working with SBFspot on a remote Raspi means a lot of cd-ing in a PuTTY screen.  I made a tiny batch file (in Dutch) that exists in every directory associated with SBFspot and makes life a bit easier. SBFspot is a great program you can leave alone for many years but that causes you to forget its file structure. Put something like this in all 3 directories (~/smadata, /usr/local/bin/sbfspot.3, /var/log/sbfspot.3), chmod +x it and fly from directory to directory.

sbf-, 2024-05-01:
@roelvdh If you are willing to generate a USB trace while connecting with SunnyExplorer, please contact me via e-mail (s.b.f at skynet.be)

closed 2024-09-27 by sbf-