SBFspot/SBFspot #700: Partial ArchiveDayData after PacketID mismatch

sbf-, 2024-05-21:
### Discussed in https://github.com/SBFspot/SBFspot/discussions/698

<div type='discussions-op-text'>

<sup>Originally posted by **tannegarn** May 16, 2024</sup>
Irregularly, I do have partial archivedaydata from a Sunny Boy 5.0. The argument -D5 shows the following data:
```
Current Inverter Time: 16/05/2024 05:52:59
Inverter Wake-Up Time: 15/05/2024 21:19:46
Inverter Sleep Time  : 16/05/2024 05:52:42
********************
* ArchiveDayData() *
********************
startTime: 16/05/2024 `00:00:00`
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 26 00 10 60 65 09 E0 
00000020: 99 01 25 A2 54 B3 00 00 7D 00 
00000030: DB B6 71 32 00 00 00 00 00 00 
00000040: 13 80 00 02 00 70 34 2F 45 66 
00000050: B4 80 46 66 00 00 00 00 
58 Bytes sent to [192.168.2.7]
58 bytes received from [192.168.2.7]
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 26 00 10 60 65 09 A0 
00000020: FF FF FF FF FF FF 00 00 99 01 
00000030: 25 A2 54 B3 00 00 00 00 00 00 
00000040: F5 9F 00 02 80 51 00 48 21 00 
00000050: FF 4A 41 00 00 00 00 00 
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 00 00 10 60 65 09 A0 FF FF FF 
00000010: FF FF FF 00 00 99 01 25 A2 54 
00000020: B3 00 00 00 00 00 00 F5 9F 00 
00000030: 02 80 51 00 48 21 00 FF 4A 41 
00000040: 00 00 00 00 
<<<=================================>>>
Packet ID mismatch. Expected 19, received 8181
logoffMultigateDevices()
logoffSMAInverter()
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00 
00000010: 00 01 00 22 00 10 60 65 08 A0 
00000020: FF FF FF FF FF FF 00 03 7D 00 
00000030: DB B6 71 32 00 03 00 00 00 00 
00000040: 14 80 0E 01 FD FF FF FF FF FF 
00000050: 00 00 00 00 
54 Bytes sent to [192.168.2.7]
```
Is the above a known issue? Is there any possibility to retry the get packet function when invalid data is received?</div>

sbf-, 2024-05-21:
Confirmed as bug, introduced in [commit 90c4a5e2](https://github.com/SBFspot/SBFspot/commit/90c4a5e293d1e8d9ffd02c9c29b59fa06c5a6b3e)

tannegarn, 2024-05-21:
Many thanks for confirming this as a bug. If youwant me to test an updated version, just let me know.

commit 2166c0a3 (2024-06-15): Fix #700 Partial ArchiveDayData after PacketID mismatch
closed 2024-06-15 by sbf-