SBFspot/SBFspot #701: Unable to get historic intraday data

JHthe4, 2024-05-28:
Hi there, 

We had a connection issue that went unnoticed for a while, so we missed about 1.5 months of intraday data from our inverter. I'm trying to restore that now by using the -ad60 flag. Weirdly, I can only get data as far back as 05/22, anything before that seems inaccessible. I can get the daily data, which seems to be uploaded fine, but the intraday doesn't appear to even get transmitted. I'm using V3.9.9 (current master branch) via IP. Here's an example of what the exchange for 05/21 looks like:

```
********************
* ArchiveDayData() *
********************
startTime: 21/05/2024 00:00:00
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00
00000010: 00 01 00 26 00 10 60 65 09 E0
00000020: 7A 01 50 AA 05 B3 00 00 7D 00
00000030: C4 66 95 39 00 00 00 00 00 00
00000040: 1A 80 00 02 00 70 B4 C6 4B 66
00000050: 34 18 4D 66 00 00 00 00
58 Bytes sent to [192.168.x.x]
58 bytes received from [192.168.x.x]
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 53 4D 41 00 00 04 02 A0 00 00
00000010: 00 01 00 26 00 10 60 65 09 E0
00000020: 7D 00 C4 66 95 39 00 E0 7A 01
00000030: 50 AA 05 B3 00 00 15 00 00 00
00000040: 1A 80 01 02 00 70 B4 C6 4B 66
00000050: 34 18 4D 66 00 00 00 00
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 00 00 10 60 65 09 E0 7D 00 C4
00000010: 66 95 39 00 E0 7A 01 50 AA 05
00000020: B3 00 00 15 00 00 00 1A 80 01
00000030: 02 00 70 B4 C6 4B 66 34 18 4D
00000040: 66 00 00 00
<<<=================================>>>

```

What do you make of this? I assume it's some inverter problem, but I can't interpret the hex output.

Thanks so much for your help!

sbf-, 2024-05-30:
It looks like there is no data recorded for that period

```
********************
* ArchiveDayData() *
********************
startTime: 21/05/2024 00:00:00
58 Bytes sent to [192.168.x.x]
53 4D 41 00 00 04 02 A0 00 00 00 01 00 26 00 10 60 65 09 E0 7A 01 50 AA 05 B3 00 00 7D 00 C4 66 95 39 00 00 00 00 00 00 1A 80 00 02 00 70 B4 C6 4B 66 34 18 4D 66 00 00 00 00
00 10 60 65          SMA Header (Speedwire)
09                   # DWORDs: 9
E0                   Ctrl byte
7A 01 50 AA 05 B3    SUSyID/Serial Destination: 378/3003492944 (Unknown SID)
00 00                Ctrl
7D 00 C4 66 95 39    SUSyID/Serial Source: 125/966092484 (SBFspot)
00 00                Ctrl
00 00                ErrorCode (OK)
00 00
1A 80                PcktID: 26

00 02 00 70          Request Archive Day Data (LogTotWhDevice)
B4 C6 4B 66          From: 20/05/2024 23:55:00
34 18 4D 66          To  : 21/05/2024 23:55:00
00 00 00 00
58 bytes received from [192.168.x.x]
00 00 10 60 65 09 E0 7D 00 C4 66 95 39 00 E0 7A 01 50 AA 05 B3 00 00 15 00 00 00 1A 80 01 02 00 70 B4 C6 4B 66 34 18 4D 66 00 00 00
00 10 60 65          SMA Header (Speedwire)
09                   # DWORDs: 9
E0                   Ctrl byte
7D 00 C4 66 95 39    SUSyID/Serial Destination: 125/966092484 (SBFspot)
00 E0                Ctrl
7A 01 50 AA 05 B3    SUSyID/Serial Source: 378/3003492944 (Unknown SID)
00 00                Ctrl
15 00                ErrorCode (Requested Data Not Available)
00 00
1A 80                PcktID: 26

01 02 00 70          Receive Archive Day Data (LogTotWhDevice)
B4 C6 4B 66          From: 20/05/2024 23:55:00
34 18 4D 66          To  : 21/05/2024 23:55:00

00 00 00
```

JHthe4, 2024-05-30:
Thank you so much for the detailed analysis! Have you seen this before in your experience? It seems strange to me that the inverter wouldn't have retained such recent data. In any case, at least we have the daily values from the MonthData table, so we can use those.

However, there is a reason we don't do that yet: 
Speaking of the MonthData, and while I have your attention, we operate a second (older) inverter which we query using Bluetooth. Since we started using SBFspot in 2016, we've seen dual entries in the MonthData table, offset from each other by 5 minutes:

[image]

Things break down around the switch to DST, and the later report no longer has the correct date anymore. Also, there are some intermittent days where there's only one entry:

[image]

A similar doubling (with an offset of an hour) seems to happen for the newer (Speedwire) inverter too, but only in the months of March and October, and only until the day of conversion to/from DST:

[image]

Is this a known issue? Do you have any experience in how to handle this consistently?

sbf-, 2024-05-30:
> Have you seen this before in your experience?

No, this one I haven't seen... #694 is a bit related

Your other issue with month data is a bit strange as it is a very old problem that was fixed in [V3.0.2](https://github.com/SBFspot/SBFspot/wiki/Modification-History#302-09-nov-2014-bugfix-release) (see #79)
The project was hosted on CodePlex at that time. The (dirty?) fix was to delete the data of the month and re-inserting it.
Are you running recent SBFspot version for all your inverters? You never reported this issue?

JHthe4, 2024-05-30:
Interesting, thanks for the link to #79! That definitely could be the same thing we're seeing on our speedwire inverter (Sunny Tripower 5.0). Do you think this is also the reason for the 5 minute offset doubling for the Bluetooth inverter (SB 5000TL-20)?

In any case, we're using a Raspberry Pi 4 (released 2019), running Ubuntu 22.04 LTS with current SBFspot installed via github:

```
 /usr/local/bin/sbfspot.3/SBFspot -version
3.9.9
```

I think it's near impossible that we're still running the <3.0.1 code somehow. This might be a new bug or a regression.

We never reported the issue, no. Instead, I've been summing up the DayData to create my own MonthData, but due to the mass of data those queries are taking over 2 seconds by now, which is no longer sustainable and prompted me to look into this.

sbf-, 2024-05-31:
You're using MySQL/MariaDB, right?

For some unknown reason, this V302 fix is not in the MySQL export. But, like I said before, it's a 10yo problem and has never been reported.

sbf-, 2024-05-31:
Do the timezones of the inverters and RPi match?

JHthe4, 2024-05-31:
> For some unknown reason, this V302 fix is not in the MySQL export. But, like I said before, it's a 10yo problem and has never been reported.

Interesting. I wonder how that happens. We are using the MariaDB version, yes. Should be possible then to implement the fix in that version as well, I assume?
 

> Do the timezones of the inverters and RPi match?

RPi, database server, and speedwire inverter are all on Europe/Berlin. As for the BT inverter, this is the output we get:

```
SBFspot V3.9.9
Yet another tool to read power production of SMA solar inverters
(c) 2012-2024, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -v -finq
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri May 31 11:56:57 2024: INFO: Starting...
sunrise: 05:20
sunset : 21:25
Connecting to 00:80:25:0A:A8:8B (1/10)
Connecting to 00:80:25:0A:A8:8B (2/10)
Connecting to 00:80:25:0A:A8:8B (3/10)
Initialising...
SUSyID: 125 - SessionID: 944052762
SMA netID=01
Serial Nr: 7D2D79A5 (2100132261)
BT Signal=65.5%
Logon OK
Local Host Time: 31.05.2024 11:57:23
Plant Time     : 01.01.1970 01:00:00 (-1717149443 sec)
TZ offset      : -2004317990 sec - DST: On
Last Time Set  : 16.06.1961 10:25:01
The time difference between inverter/host is more than 3600 seconds.
As a precaution the plant time won't be adjusted.
To overrule this behaviour, execute SBFspot -settime
getSoftwareVersion returned an error: -1
SUSyID: 78 - SN: 2100132261
Device Name:      SN: 2100132261
Device Class:     Solar-Wechselrichter
Device Type:      SB 5000TL-20
Software Version:
Packet status: 21
SUSyID: 78 - SN: 2100132261
Device Status:      Ok
SUSyID: 78 - SN: 2100132261
Device Temperature: 60.0°C
SUSyID: 78 - SN: 2100132261
GridRelay Status:      Geschlossen
SUSyID: 78 - SN: 2100132261
Energy Production:
        EToday: 5.962kWh
        ETotal: 91173.154kWh
        Operation Time: 60877.04h
        Feed-In Time  : 59113.31h
SUSyID: 78 - SN: 2100132261
DC Spot Data:
        MPPT 1 Pdc:   1.815kW - Udc: 363.03V - Idc:  5.001A
        MPPT 2 Pdc:   1.774kW - Udc: 358.45V - Idc:  4.949A
        Calculated Total Pdc:   3.589kW
AC Spot Data:
        Phase 1 Pac :   3.490kW - Uac: 234.31V - Iac: 14.893A
        Phase 2 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Phase 3 Pac :   0.000kW - Uac:   0.00V - Iac:  0.000A
        Total Pac   :   3.490kW - Calculated Pac:   3.490kW
        Efficiency  :   97.24%
SUSyID: 78 - SN: 2100132261
Grid Freq. : 49.99Hz
SUSyID: 78 - SN: 2100132261
Current Inverter Time: 31.05.2024 11:53:08
Inverter Wake-Up Time: 31.05.2024 05:57:17
Inverter Sleep Time  : 31.05.2024 11:53:08
********************
* ArchiveDayData() *
********************
startTime: 31/05/2024 00:00:00
**********************
* ArchiveMonthData() *
**********************
startTime: 01/05/2024 13:00:00
Reading events: 2024-May-01
ExportEventsToCSV()
Fri May 31 11:58:17 2024: INFO: Done.
Exit request sent.
```

Seems like the "plant time" might be wrong, but the "current inverter time" is more or less correct?

JHthe4, 2024-05-31:
I have some more updates: It appears that there's a consistent 5 minute offset between the real time and the "Current Inverter Time", which might be the cause of the database issue as well.

Even after successfully executing the flag -settime2:

```
/usr/local/bin/sbfspot.3/SBFspot -settime2
SBFspot V3.9.9
Yet another tool to read power production of SMA solar inverters
(c) 2012-2024, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with MySQL support
Commandline Args: -settime2
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'
Fri May 31 16:34:42 2024: INFO: Starting...
sunrise: 05:20
sunset : 21:25
Connecting to 00:80:25:0A:A8:8B (1/10)
Connecting to 00:80:25:0A:A8:8B (2/10)
Connecting to 00:80:25:0A:A8:8B (3/10)
Connecting to 00:80:25:0A:A8:8B (4/10)
Connecting to 00:80:25:0A:A8:8B (5/10)
Initialising...
SUSyID: 125 - SessionID: 958226484
SMA netID=01
Serial Nr: 7D2D79A5 (2100132261)
BT Signal=65.1%
Logon OK
Local Time: 31.05.2024 16:35:11
TZ offset (s): 7200
Adjusting plant time...
New plant time is now 31.05.2024 16:35:11
Fri May 31 16:35:11 2024: INFO: Done.
```

the problem persists and the time does not appear to have been set/saved correctly:

```
--- shortened for brevity ---
Logon OK
Local Host Time: 31.05.2024 16:38:54
Plant Time     : 01.01.1970 01:00:00 (-1717166334 sec)
TZ offset      : -2004317990 sec - DST: On
Last Time Set  : 16.06.1961 10:25:01
The time difference between inverter/host is more than 3600 seconds.
As a precaution the plant time won't be adjusted.
To overrule this behaviour, execute SBFspot -settime
---shortened---
Current Inverter Time: 31.05.2024 16:34:38
Inverter Wake-Up Time: 31.05.2024 05:57:17
Inverter Sleep Time  : 31.05.2024 16:34:39
---shortened---
Fri May 31 16:39:47 2024: INFO: Done.
```

sbf-, 2024-06-01:
Getting/Setting inverter time by SBFspot has always been difficult...
But on your inverter it is completely lost:
```
Plant Time     : 01.01.1970 01:00:00 (-1717149443 sec)
TZ offset      : -2004317990 sec - DST: On
Last Time Set  : 16.06.1961 10:25:01
```
`01.01.1970 01:00:00` is 0 in Unix Epoch, so this seems very wrong. Would be interresting to have a -d5 output of this.

> We are using the MariaDB version, yes. Should be possible then to implement the fix in that version as well, I assume?

Sure, will fix this

JHthe4, 2024-06-04:
> Would be interresting to have a -d5 output of this.

Coming right up! These are too long for pasting here, so I've moved them into txt files:

[bt_log.txt](https://github.com/user-attachments/files/15545835/bt_log.txt)
[settime2.txt](https://github.com/user-attachments/files/15545836/settime2.txt)

Notice that the 5 minute offset with the current inverter time persists. Hope these are helpful!

> Sure, will fix this

Awesome, thanks so much! Sorry for the late report :)

closed 2024-06-14 by sbf-

JHthe4, 2024-06-17:
Hi,

thanks for the fix, I'm happy to report that it appears to be working perfectly so far! 

We still have the weird 5 minute offset with the older inverter, but I'm able to circumvent that by just subtracting a few seconds from the timestamp to move it back into the right day (it might be an idea in the future to select the time for the day timestamp as noon instead of 5 minutes to end of day).

For anyone reading this in the future, I was able to fix bad old data in the database by reimporting using the -am300 flag. This removes old values and rewrites correctly.

Thanks again @sbf- for your efforts!

commit 0ff1b754 (2024-06-14): Merge branch 'issue_701'
Fix #701
commit b640d4bc (2024-06-14): Fix #701 Double data in Monthdata table