SBFspot/SBFspot #702: after update to 3.9.10, version remains 3.9.8

wokkeltje13, 2024-06-15:
is it normal the version remains 3.9.8 after running 
`sudo bash -c "$(curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config)"`

thanks in advance

Sixtus81, 2024-06-15:
`sudo bash -c "$(curl -s https://raw.githubusercontent.com/sbfspot/sbfspot-config/master/sbfspot-config)"`
shows only version 3.9.8 to install
```
SBFspot Configuration Tool V1.3.5
Running on Raspberry Pi 3 Model B Rev 1.2
Checking whiptail... Installed.
architecture=arm64
Debian GNU/Linux 11 (bullseye)
username=dietpi
userhome=/home/dietpi
Minimum SBFspot version supported by this tool is 3.5.0
Request to install latest version (3.9.8)
Searching for existing configuration(s)...
Aborting script...
```
thanks

sbf-, 2024-06-15:
> is it normal the version remains 3.9.8 after running...

No, I'll look at this after the weekend ;-)