SBFspot/SBFspot #712: Not publishing to MQTT

ralgg, 2024-11-10:
MQTT: Publishing (sbfspot_sma2130403157),"Timestamp": "10/11/2024 10:10:02","SunRise": "10/11/2024 07:21:00","SunSet": "10/11/2024 16:19:00","InvSerial": 2130403157,"InvName": "SB 3000TL-21 157","InvTime": "10/11/2024 10:09:59","InvStatus": "Ok","InvTemperature": 15.890,"InvGridRelay": "Closed","EToday": 0.279,"ETotal": 33242.511,"PACTot": 265.000,"UDC1": 375.950,"UDC2": 0.000,"IDC1": 0.731,"IDC2": 0.000,"PDC1": 274.000,"PDC2": 0.000
**sh: 1: Syntax error: EOF in backquote substitution**
MQTT: Failed to execute '/usr/bin/mosquitto_pub' mosquitto client installed?
Error 512 while publishing to MQTT Broker

**Auszug aus SBFspot.cfg**
# Full path to mosquitto_pub executable
MQTT_Publisher=/usr/bin/mosquitto_pub

# IP address or DNS name of MQTT Broker
# Don't use test broker for production environments
#MQTT_Host=test.mosquitto.org
MQTT_Host=192.168.1.3

# Port (Optional)
#MQTT_Port=

# Topic Name
# When using public broker like test.mosquitto.org make sure to define a unique topic name
# by appending the serial number of the inverter to avoid you receive data of another SBFspot user.
MQTT_Topic=sbfspot_sma2130403157

# Format of message items to be sent
# JSON: MQTT_ItemFormat="{key}": {value}
# TEXT: MQTT_ItemFormat={key}:{value}
# XML:  MQTT_ItemFormat=<item name="{key}" value="{value}" />
MQTT_ItemFormat="{key}": {value}

# MQTT_ItemDelimiter (none|blank|comma|semicolon default comma)
# JSON: MQTT_ItemDelimiter=comma
# TEXT: MQTT_ItemDelimiter=semicolon
# XML : MQTT_ItemDelimiter=none
MQTT_ItemDelimiter=comma

# Mandatory arguments for mosquitto_pub executable
# JSON: MQTT_PublisherArgs=-h {host} -t {topic} -m "{{message}}"
# TEXT: MQTT_PublisherArgs=-h {host} -t {topic} -m "{message}"
# XML : MQTT_PublisherArgs=-h {host} -t {topic} -m "<mqtt_message>{message}</mqtt_message>"
MQTT_PublisherArgs=-u Moskito -P xxxxxxxxx -h {host} -t {topic} -m "{{message}}"

# Data to be published (comma delimited)
MQTT_Data=Timestamp,SunRise,SunSet,InvSerial,InvName,InvTime,InvStatus,InvTemperature,InvGridRelay,EToday,ETotal,PACTot,UDC1,UDC2,IDC1,IDC2,PDC1,PDC2

sbf-, 2024-11-12:
`sh: 1: Syntax error: EOF in backquote substitution` is a shell error. Is there a syntax error in the calling script? Or a backquote used in your mosquitto password?

ralgg, 2024-11-13:
Yes there was a backquote in my mosquitto password. After changing the password I now get a new error at the same position. 
I changed the code in mqtt.cpp to get the value of the string mqtt_command_line. This is the new part of the protocol.

MQTT: Publishing (sbfspot_2130403157),"InvName": "SB 3000TL-21 157"

/usr/bin/mosquitto_pub  Moskito  Password -h 192.168.1.3 -t sbfspot_2130403157 -m '{"InvName": "SB 3000TL-21 157"}'
Error: Unknown option 'Moskito'.

Moskito is the name of my mqtt user in Home Assistant, Password the password.

sbf-, 2024-11-13:
You forgot the -u

ralgg, 2024-11-14:
Thank you for the advice.
There were three errors in total:
' in the password
! in the password
missing -u and -P

Now it's working.

closed 2024-11-14 by ralgg

sbf-, 2024-11-14:
> There were three errors in total:
> ' in the password
> ! in the password
> missing -u and -P

Found a [nice example](http://www.steves-internet-guide.com/mqtt-username-password-example/) to store encrypted passwords in a file. That might help to use these characters in a password.