SBFspot/SBFspot #714: CRITICAL: Failed to initialise communication with inverter

Depechie, 2025-01-06:
Hey,

I'm trying to move my SBFspot instance to another machine. But the other machine keeps giving the exception `CRITICAL: Failed to initialise communication with inverter`.

Not sure what to do in this regard. Any pointers?
On both devices doing `hcitool scan` returns:
```
Scanning ...
	00:80:25:15:23:15	SMA002d SN: 2100274254 SN2100274254
```
So the BT module is functional  on both devices.
Only on the new one, I constantly get:

```
Connecting to 00:80:25:15:23:15 (1/10)
Initialising...
SUSyID: 125 - SessionID: 821753966
SMA netID=02
Mon Jan  6 22:20:16 2025: CRITICAL: Failed to initialise communication with inverter.
Mon Jan  6 22:20:16 2025: INFO: Done.
```

Bit out op ideas on what this could be.

sbf-, 2025-01-06:
Is the old rpi still running at the same time? Or SunnyExplorer?
Pls provide debuglog (-d5 -v5)

Depechie, 2025-01-06:
Inwas still trying further and forgot I also have a minipc in the same server rack running Proxmox. And I added a Debian VM and did Bluetooth passthrough.
Then installed SBFspot and that works great.

Did not yet dive deeper in why the nanopi does not work though.
But for me I can continue with the VM.

closed 2025-01-06 by Depechie

Depechie, 2025-01-07:
Hey @sbf- sorry I rejoiced to soo, I also get the same error on the mini pc when it runs it's regular schedule.
So I did a `/usr/local/bin/sbfspot.3/SBFspot_sqlite -d5 -v5 -finq -cfg/etc/sbfspot/SBFspot.cfg` and got following info

```
SBFspot V3.9.7
Yet another tool to read power production of SMA solar inverters
(c) 2012-2022, SBF (https://github.com/SBFspot/SBFspot)
Compiled for Linux (LE) 64 bit with SQLite support
Commandline Args: -d5 -v5 -finq -cfg/etc/sbfspot/SBFspot.cfg
Reading config '/etc/sbfspot/SBFspot.cfg'

Configuration settings:
BTAddress=00:80:25:15:23:15
MIS_Enabled=0
BTConnectRetries=10
Password=<undisclosed>
Plantname=MyPlant
OutputPath=/var/sbfspot/%Y
OutputPathEvents=/var/sbfspot/%Y/Events
Latitude=51.0908
Longitude=4.55073
Timezone=Europe/Brussels
Locale=en-US
CalculateMissingSpotValues=0
DateTimeFormat=%d/%m/%Y %H:%M:%S
DateFormat=%d/%m/%Y
TimeFormat=%H:%M:%S
SynchTime=1
SynchTimeLow=1
SynchTimeHigh=3600
SunRSOffset=900
DecimalPoint=comma
CSV_Delimiter=semicolon
Precision=3
CSV_Export=1
CSV_ExtendedHeader=1
CSV_Header=1
CSV_SaveZeroPower=1
CSV_Spot_TimeSource=0
CSV_Spot_WebboxHeader=0
SQL_Database=/var/sbfspot/sbfspot.db
End of Config

Tue Jan  7 11:33:13 2025: INFO: Starting...
sunrise: 08:44
sunset : 16:51
Connecting to 00:80:25:15:23:15 (1/10)
Initialising...
SUSyID: 125 - SessionID: 930540433
getPacket(2)
MAX_CommBuf is now 18 bytes
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 15 23 15 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 02 00 00 00 00 01 00 00
00000030: 00
cmd=2
<<<====== Content of pcktBuf =======>>>
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 15 23 15 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 02 00 00 00 00 01 00 00
00000030: 00
<<<=================================>>>
MAX_pcktBuf is now 31 bytes
SMA netID=02
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 00 00 00 00 00 00
00000010: 15 23 15 25 80 00 02 00 00 04
00000020: 70 00 02 00 00 00 00 01 00 00
00000030: 00
31 Bytes sent
getPacket(5)
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 15 23 15 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 02 00 00 00 00 01 00 00
00000030: 00
cmd=2
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 15 23 15 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 02 00 00 00 00 01 00 00
00000030: 00
cmd=2
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 15 23 15 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 02 00 00 00 00 01 00 00
00000030: 00
cmd=2
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 15 23 15 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 02 00 00 00 00 01 00 00
00000030: 00
cmd=2
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 15 23 15 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 02 00 00 00 00 01 00 00
00000030: 00
cmd=2
Received 18 bytes
Received 13 bytes
--------: 00 01 02 03 04 05 06 07 08 09
00000000: 7E 1F 00 61 15 23 15 25 80 00
00000010: 00 00 00 00 00 00 02 00 00 04
00000020: 70 00 02 00 00 00 00 01 00 00
00000030: 00
cmd=2
Timeout reading socket
No data!
Tue Jan  7 11:33:43 2025: CRITICAL: Failed to initialise communication with inverter.
Tue Jan  7 11:33:43 2025: INFO: Done.
```

sbf-, 2025-01-07:
Is the old system still running at the same interval?

Depechie, 2025-01-07:
The config is the same, but I guess your question is more of multiple devices are trying to connect to the inverter.
In that case, no. I stop all sbfspot instances and make sure only 1 is running at the same time.

I'm using docker containers on each device so I stop them all and only start the one I want to try out if needed.

Depechie, 2025-01-07:
I know see that the old machine is using a docker image that has SBFspot V3.8.4 versus the others that are running SBFspot V3.9.7

I will copy over the docker image from the old machine and try to run that on a newer and report back tomorrow ( to see if the day job works ).

Depechie, 2025-01-08:
Ok... @sbf- that did fix the problem!
So I guess there is a config change between the versions maybe? Because of course, I just used the config I already had.
Or is there maybe a change in communication towards the inverter since the version I'm using versus the current one?

sbf-, 2025-01-09:
Must be a code change since V3.8.4
I have the same inverter with same firmware and running SBFspot V3.9.4 so maybe I can reproduce the issue.
But indeed strange it happens only when running with cron
Thanks for the effort!

cromefire, 2025-07-05:
I have similar issue it seems where it always has checksum error:
<details>
<summary>logs</summary>

```
sbf-spot-1  | Configuration settings:
sbf-spot-1  | BTAddress=<censored>
sbf-spot-1  | MIS_Enabled=0
sbf-spot-1  | BTConnectRetries=10
sbf-spot-1  | Password=<undisclosed>
sbf-spot-1  | Plantname=Home
sbf-spot-1  | OutputPath=/home/pi/smadata/%Y
sbf-spot-1  | OutputPathEvents=/home/pi/smadata/%Y/Events
sbf-spot-1  | Latitude=<censored>
sbf-spot-1  | Longitude=<censored>
sbf-spot-1  | Timezone=Europe/Berlin
sbf-spot-1  | Locale=en-US
sbf-spot-1  | CalculateMissingSpotValues=1
sbf-spot-1  | DateTimeFormat=%d/%m/%Y %H:%M:%S
sbf-spot-1  | DateFormat=%d/%m/%Y
sbf-spot-1  | TimeFormat=%H:%M:%S
sbf-spot-1  | SynchTime=1
sbf-spot-1  | SynchTimeLow=1
sbf-spot-1  | SynchTimeHigh=3600
sbf-spot-1  | SunRSOffset=1200
sbf-spot-1  | DecimalPoint=comma
sbf-spot-1  | CSV_Delimiter=semicolon
sbf-spot-1  | Precision=3
sbf-spot-1  | CSV_Export=0
sbf-spot-1  | CSV_ExtendedHeader=1
sbf-spot-1  | CSV_Header=1
sbf-spot-1  | CSV_SaveZeroPower=1
sbf-spot-1  | CSV_Spot_TimeSource=0
sbf-spot-1  | CSV_Spot_WebboxHeader=0
sbf-spot-1  | MQTT_Host=<censored>
sbf-spot-1  | MQTT_Port=1883
sbf-spot-1  | MQTT_Topic=devices/sbfspot/energy/{serial}/unfiltered
sbf-spot-1  | MQTT_Publisher=/usr/bin/mosquitto_pub
sbf-spot-1  | MQTT_PublisherArgs=-h {host} -p {port} -t {topic} -m "{{message}}" -V mqttv5 -u sma -P "<censored>"
sbf-spot-1  | MQTT_Data=Timestamp,SunRise,SunSet,EToday,ETotal,PACTot,PDCTot,InvStatus,InvTemperature,GridFreq,UAC1,UDC1,UDC2
sbf-spot-1  | MQTT_ItemFormat="{key}": {value}
sbf-spot-1  | End of Config
sbf-spot-1  | 
sbf-spot-1  | Sat Jul  5 14:57:34 2025: INFO: Starting...
sbf-spot-1  | sunrise: 05:16
sbf-spot-1  | sunset : 21:45
sbf-spot-1  | Connecting to <censored> (1/10)
sbf-spot-1  | Initialising...
sbf-spot-1  | SUSyID: 125 - SessionID: 816731848
sbf-spot-1  | getPacket(2)
sbf-spot-1  | MAX_CommBuf is now 18 bytes
sbf-spot-1  | Received 18 bytes
sbf-spot-1  | Received 13 bytes
sbf-spot-1  | cmd=2
sbf-spot-1  | <<<====== Content of pcktBuf =======>>>
sbf-spot-1  | --------: 00 01 02 03 04 05 06 07 08 09
sbf-spot-1  | 00000000: 7E 1F 00 61 52 56 08 25 80 00 
sbf-spot-1  | 00000010: 00 00 00 00 00 00 02 00 00 04 
sbf-spot-1  | 00000020: 70 00 02 00 00 00 00 01 00 00 
sbf-spot-1  | 00000030: 00 
sbf-spot-1  | <<<=================================>>>
sbf-spot-1  | MAX_pcktBuf is now 31 bytes
sbf-spot-1  | SMA netID=02
sbf-spot-1  | --------: 00 01 02 03 04 05 06 07 08 09
sbf-spot-1  | 00000000: 7E 1F 00 61 00 00 00 00 00 00 
sbf-spot-1  | 00000010: 52 56 08 25 80 00 02 00 00 04 
sbf-spot-1  | 00000020: 70 00 02 00 00 00 00 01 00 00 
sbf-spot-1  | 00000030: 00 
sbf-spot-1  | 31 Bytes sent
sbf-spot-1  | getPacket(5)
sbf-spot-1  | Received 18 bytes
sbf-spot-1  | Received 13 bytes
sbf-spot-1  | cmd=10
sbf-spot-1  | Received 18 bytes
sbf-spot-1  | Received 2 bytes
sbf-spot-1  | cmd=12
sbf-spot-1  | Received 18 bytes
sbf-spot-1  | Received 16 bytes
sbf-spot-1  | cmd=5
sbf-spot-1  | <<<====== Content of pcktBuf =======>>>
sbf-spot-1  | --------: 00 01 02 03 04 05 06 07 08 09
sbf-spot-1  | 00000000: 7E 22 00 5C 52 56 08 25 80 00 
sbf-spot-1  | 00000010: 00 00 00 00 00 00 05 00 52 56 
sbf-spot-1  | 00000020: 08 25 80 00 01 01 44 88 81 2F 
sbf-spot-1  | 00000030: B3 D8 02 01 
sbf-spot-1  | <<<=================================>>>
sbf-spot-1  | MAX_pcktBuf is now 34 bytes
sbf-spot-1  | Local BT address: <censored>
sbf-spot-1  | --------: 00 01 02 03 04 05 06 07 08 09
sbf-spot-1  | 00000000: 7E 3F 00 41 44 88 81 2F B3 D8 
sbf-spot-1  | 00000010: FF FF FF FF FF FF 01 00 7E FF 
sbf-spot-1  | 00000020: 03 60 65 09 A0 FF FF FF FF FF 
sbf-spot-1  | 00000030: FF 00 00 7D 5D 00 C8 56 AE 30 
sbf-spot-1  | 00000040: 00 00 00 00 00 00 02 80 00 02 
sbf-spot-1  | 00000050: 00 00 00 00 00 00 00 00 00 00 
sbf-spot-1  | 00000060: 05 3D 7E 
sbf-spot-1  | 63 Bytes sent
sbf-spot-1  | getPacket(1)
sbf-spot-1  | Received 18 bytes
sbf-spot-1  | MAX_CommBuf is now 25 bytes
sbf-spot-1  | Received 25 bytes
sbf-spot-1  | cmd=1
sbf-spot-1  | PacketLength=106
sbf-spot-1  | <<<====== Content of pcktBuf =======>>>
sbf-spot-1  | --------: 00 01 02 03 04 05 06 07 08 09
sbf-spot-1  | 00000000: 7E FF 03 60 65 13 90 7D 00 C8 
sbf-spot-1  | 00000010: 56 AE 30 00 00 4E 00 61 79 2B 
sbf-spot-1  | 00000020: 7D 00 00 00 00 00 00 00 00 00 
sbf-spot-1  | 00000030: 00 00 00 00 00 00 00 00 00 00 
sbf-spot-1  | 00000040: 00 00 00 00 00 00 00 00 00 00 
sbf-spot-1  | 00000050: 00 00 00 00 00 00 00 00 00 00 
sbf-spot-1  | 00000060: 00 00 00 00 00 00 00 00 00 00 
sbf-spot-1  | 00000070: 00 00 00 00 00 00 00 00 00 00 
sbf-spot-1  | 00000080: 00 00 00 00 00 
sbf-spot-1  | <<<=================================>>>
sbf-spot-1  | MAX_pcktBuf is now 85 bytes
sbf-spot-1  | Invalid chk 0x578D - Found 0x0000
sbf-spot-1  | Sat Jul  5 14:57:36 2025: CRITICAL: Failed to initialise communication with inverter.
sbf-spot-1  | Sat Jul  5 14:57:36 2025: INFO: Done.
sbf-spot-1  | Sat Jul  5 14:57:36 CEST 2025
sbf-spot-1  | Sleeping 60 seconds.
```

</details>

I tested that device and it worked, but then stopped sbfspot on that device again until today, where I was trying to finally migrate to the new device and now it just doesn't seem to work anymore for some reason...