SBFspot/SBFspot #717: MQTT bug std::out_of_range

WolfgangFahl, 2025-01-18:
I kept getting 
```
terminate called after throwing an instance of 'std::out_of_range'
  what():  basic_string::substr: __pos (which is 1) > this->size() (which is 0)
./check: line 14: 210501 Aborted                 (core dumped) 
```

and decided to write my own publishing bash function. The script had been in use with Volkszähler in the past and i am now using it with HomeAssistant
```
# WF 2018-07-07
# SMASpot version
# WF 2018-11-03
# SBFSpot version
# WF 2025-01-13
# on for

sbf_bin=/usr/local/bin/sbfspot.3
sbf_config=/home/wf/sbfspot
#
# getresult from SMA Inverters
# 
getresult() {
  local l_watt="$1"
  local l_file="$2"
  $sbf_bin/SBFspot -v -finq -cfg${sbf_config}/${l_watt}Watt.cfg > $l_file
}

#
# get the value
# param #1: file
# param #2: name 
#
getvalue() {
  local l_file="$1"
  local l_name="$2"
  cat "$l_file" | grep -i "$l_name"
}

# publish values via mqtt
# param #1: watt    - e.g 6000
# param #2: invtime - inverter time
# param #3: temp    - temperature
# param #4: etoday  - energy today
# param #5: etotal  - total energy
mqtt_publish() {
  local l_watt="$1"
  local l_invtime="$2"
  # Extract just the number before °C
  local l_temp=$(echo "$3" | grep -o '[0-9.]*' | head -1)
  # Extract just the number before kWh
  local l_etoday=$(echo "$4" | grep -o '[0-9.]*' | head -1)
  local l_etotal=$(echo "$5" | grep -o '[0-9.]*' | head -1)
  local topic="sbfspot_${l_watt}W"
  local msg="{\"Timestamp\":\"$timestamp\",\"InvTime\":\"$l_invtime\",\"Temperature\":$l_temp,\"EToday\":$l_etoday,\"ETotal\":$l_etotal}"
  mosquitto_pub -h mqtt.bitplan.com -u mqtt_user -P "**password**" -t "$topic" -m "$msg"
  [ $? -eq 0 ] && echo "MQTT ✓" || echo "MQTT ✗"
}

# get the current time
timestamp=$(date "+%Y-%m-%d %H:%M:%S")
echo "SMA converter state at $timestamp"
for watt in 6000 4000 
do
  wfile=/tmp/${watt}.watt
  getresult $watt $wfile 
  serial=$(getvalue $wfile "serial number")
  device=$(getvalue $wfile "device type")
  temp=$(getvalue $wfile "Device Temperature")
  today=$(getvalue $wfile "EToday")
  total=$(getvalue $wfile "ETotal")
  itime=$(getvalue $wfile "Current Inverter Time")

  echo "${watt} Watt $device $serial"
  echo "  $itime"
  echo "  $today $total $temp"
  mqtt_publish "$watt" "$itime" "$temp" "$today" "$total"
 # filter result
  cat $wfile | grep MPPT
done
```

sbf-, 2025-01-30:
`basic_string::substr: __pos (which is 1) > this->size() (which is 0)`
might be caused by [this line of code](https://github.com/SBFspot/SBFspot/blob/master/SBFspot/mqtt.cpp#L186)
`boost::replace_first(mqtt_command_line, "{message}", mqtt_message.str().substr(1));`
when `mqtt_message` is an empty string, but that would be strange, no?

Is there any verbose logging available?
What are the `MQTT_` items in the config file?

WolfgangFahl, 2025-02-02:
@sbf- sorry that was only an intermediate state of affairs and i did not save anything especially since the work around was easy. I have now moved to a raspberry PI 3B and could use the config tool and apparently MQTT works as advertised in that environment on bookworm. The non working machine was an Ubuntu 22.04 LTS.

sbf-, 2025-02-05:
OK, I'll close it for now.

closed 2025-02-05 by sbf-