SBFspot/SBFspot #722: Data for Sunny Tripower 8.0 SE is not exported to 123Solar

sbf-, 2025-02-12:
### Discussed in https://github.com/SBFspot/SBFspot/discussions/721

<div type='discussions-op-text'>

<sup>Originally posted by **AKlemm88** February  9, 2025</sup>
Hello,

I would like to retrieve the data from my Sunny Tripower 8.0 SE using SBFspot.
I am already using this tool with three other inverters, where it works perfectly (1x SB-4000TL, 1x SB1.5-TL, and 1x STP-12X).

I copied the configuration of the two Speedwire devices and changed the IP address. Then, I run the SBFspot command but receive an “empty” response.

I execute the following command:
`/usr/local/bin/sbfspot.3/SBFspot -finq -q -cfgSBFspot_1.cfg`

As a response, I get the following values:
```
00464B09 28 Sun Feb 9 20:50:25 2025 'Grid Voltage Phase L1 to L2' 41180 41180 41180 41180  
00464C09 28 Sun Feb 9 20:50:25 2025 'Grid Voltage Phase L2 to L3' 41480 41480 41480 41480  
00464D09 28 Sun Feb 9 20:50:25 2025 'Grid Voltage Phase L3 to L1' 41270 41270 41270 41270  
00464E09 28 Sun Feb 9 20:50:25 2025 '' 3 3 3 3  
```
Do I need to consider anything special when retrieving data from an SE inverter?

Thank you very much.

Best regards,
Alex</div>

sbf-, 2025-02-12:
I had a look at the code...
123solar export is only done for solar inverters. Your Sunny Tripower 8.0 SE is a battery inverter and ignored.
I'll try to fix this in the next release.

AKlemm88, 2025-02-15:
I tried to fix the issue for testing purposes. Now I get the data into 123solar.

I changed line 695 in Inverter.cpp to:
if ((m_inverters[0]->DevClass == SolarInverter) || (m_inverters[0]->DevClass == HybridInverter) || (m_inverters[0]->DevClass == BatteryInverter))

Now the problem is that 123solar is displaying the sum of the production power and the battery discharge power.

Example: The PV system generates 100W from sunlight and additionally feeds 100W from the battery into the house grid. In this case, 123solar displays 200W as production.

I think there must be a second change in CSVexport.cpp to correct the 123s values when a hybrid inverter is used.

If I find the solution, I will post it here again, to help you fix this problem.

AKlemm88, 2025-02-15:
with -v2 option there is the following output:

Energy Production:
	EToday: 1.411kWh
	ETotal: 741.236kWh
	Operation Time: 0.00h
	Feed-In Time  : 0.00h
00464B09 28 Sat Feb 15 11:36:10 2025 'Netzspannung Phase L1 gegen L2' 40650 40650 40650 40650
00464C09 28 Sat Feb 15 11:36:10 2025 'Netzspannung Phase L2 gegen L3' 40960 40960 40960 40960
00464D09 28 Sat Feb 15 11:36:10 2025 'Netzspannung Phase L3 gegen L1' 40780 40780 40780 40780
00464E09 28 Sat Feb 15 11:36:11 2025 '' 2 2 2 2
DC Spot Data:
	MPPT 1 Pdc:   0.054kW - Udc: 277.40V - Idc:  0.194A
	MPPT 2 Pdc:   0.055kW - Udc: 301.60V - Idc:  0.182A
	Calculated Total Pdc:   0.109kW
AC Spot Data:
	Phase 1 Pac :   0.038kW - Uac: 234.70V - Iac:  0.161A
	Phase 2 Pac :   0.036kW - Uac: 236.50V - Iac:  0.152A
	Phase 3 Pac :   0.036kW - Uac: 235.50V - Iac:  0.152A
	Total Pac   :   0.113kW - Calculated Pac:   0.110kW
	Efficiency  :  100.92%
Grid Freq. : 49.98Hz

SunnyExplorer is showing the same values. 

I think the correct values could only be calculated.
overal dc power - battery discharge power = sunlight power.

sbf-, 2025-02-15:
> overal dc power - battery discharge power = sunlight power.

Ok, but battery discharge power is not available in SBFspot ATM
And maybe 123solar doesn't have battery support?

AKlemm88, 2025-02-15:
Okay, then I might look into how to retrieve this from the inverter. However, I don’t know how to get it yet. I’m not familiar with the SMA data protocol, but I can read up on how it works. Is there a simple way to get all data at once for testing purposes?

I think 123solar doesn’t support batteries, but I can ask about it.

Maybe I can create the graphs with Grafana.

commit 1902db3e (2025-02-22): Fix #722 Data for Sunny Tripower 8.0 SE is not exported to 123Solar
closed 2025-02-22 by sbf-