SBFspot/SBFspot #730: CSV Export: Header broken

mahadi, 2025-05-07:
As described in this discussion https://github.com/SBFspot/SBFspot/discussions/729 the CSV export seems to be broken, as the generated header line does not fit to the data provided by the solar inverter.
@sbf- Can you confirm this classification?

sbf-, 2025-05-07:
I'm unable to reproduce this issue with V3.9.12 (Windows version, but this should not make any difference)

What was your previous version where it was OK? You can always go back till V3.9.2
Do you have a sample?

mahadi, 2025-05-09:
In my previous CSV export I see the following layout:

``` csv
sep=;
Version CSV1|Tool SBFspot3.5.2 (Linux)|Linebreaks LF|Delimiter semicolon|Decimalpoint comma|Precision 3

;;;;Watt;Watt;Amp;Amp;Volt;Volt;Watt;Watt;Watt;Amp;Amp;Amp;Volt;Volt;Volt;Watt;Watt;%;kWh;kWh;Hz;Hours;Hours;%;Status;Status;degC
dd/MM/yyyy HH:mm:ss;DeviceName;DeviceType;Serial;Pdc1;Pdc2;Idc1;Idc2;Udc1;Udc2;Pac1;Pac2;Pac3;Iac1;Iac2;Iac3;Uac1;Uac2;Uac3;PdcTot;PacTot;Efficiency;EToday;ETotal;Frequency;OperatingTime;FeedInTime;BT_Signal;Condition;GridRelay;Temperature
29/04/2025 03:00:00;SN: 305123831;STP 10000TL-20;305123831;0,000;0,000;0,000;0,000;0,000;0,000;0,000;0,000;0,000;0,000;0,000;0,000;0,000;0,000;0,000;0,000;0,000;0,000;0,000;75084,403;0,000;28532,460;28123,821;0,000;Ok;Information not available;0,000
```

So the version was 3.5.2.
Did anything change in the CSV export since then?