SBFspot/SBFspot #733: Strange behavior of UploadDeamon

freekdk, 2025-05-18:
On May 13 I have the following in the UploadDeamon log:
[12:41:30.381] INFO: Uploading datapoint: 20250513,12:40,23464196,1764,,,54.28,228.78 => OK (200)
[12:46:32.046] ERROR: Uploading datapoint: 20250513,12:45,23464490,3528,,,54.51,244.52 Bad request 400: Power value [3528] too high
 for system size [2250]
The value for the power in the table SpotData at that time is however: 
:~> date --date='2025-05-13 12:45' +%s
1747133100
:~> sqlite3 smadata/SBFspot.db 'select * from spotdata where TimeStamp>=1747133100 and timestamp<1747133105'
1747133104|2120214953|1861|0|7.614|0.0|244.52|0.0|1772|0|0|7.299|0.0|0.0|242.77|0.0|0.0|5286|23464492|49.99|48775.0|45594.0|79.2157|OK|Closed|54.51
So 1861, which is quite different from what the UploadDaemon uses: 3528.

This continues till the following output:
[16:21:29.666] ERROR: Uploading 30 datapoints, starting with 20250513,12:45,23464490,3528,,,54.51,244.52 Bad request 400: Power val
ue [3528] too high for system size [2250]
[16:26:31.684] ERROR: Uploading 30 datapoints, starting with 20250507,21:15,23391971,0,,,29.51,227.17 Bad request 400: Power value 
[3528] too high for system size [2250]

Note the change in date from May 13 to May 07

The last line continues till today.

So for some strange reason, at first the wrong value for the power is taken and at a later moment that value remains the same, but the date/time is getting back to a value that does not even exist in the table SpotData. The last date/time on that day, May 7, is 5 minutes earlier.

From the data in the table SpotData I can repair the data on pvoutput.org, but apparently that does not change the behavior of the UploadDaemon.

sbf-, 2025-05-18:
Power values are taken from `daydata` table...
Take a look at #137 for workaround

sbf-, 2025-05-19:
What version of SBFspot are you running? There was a fix for this issue some time ago.

freekdk, 2025-05-19:
I don't think this workaround is applicable to my problem.

Suddenly, without an obvious cause, the UploadDaemon uses a strange value as the power at a certain moment. That value is indeed on May 13 12:45 in the table `daydata`, but it is not the value at that time in the table `spotdata`. Nor are other values around the same time the same in both tables. So how do these values compare.

The other strange thing is that later that day, May 13, UploadDaemon is using that wrong value but with a different data/time May 7 21:15.  In the table `daydata` on May 7 the highest value of the time is 21:15

The table `daydata` has a field called PVoutput int(1); is that the indicator for UploadDaemon that that value has been uploaded?

freekdk, 2025-05-19:
I am using 3.9.12

sbf-, 2025-05-19:
> I don't think this workaround is applicable to my problem.

Sure it is!

sbf-, 2025-05-20:
> the UploadDaemon is uploading again

closed 2025-05-20 by sbf-